ghdl/ghdl #2461: GHDL Bug occurred while trying to synthesize working VHDL code

luis705, 2023-06-27:
**Description**
When trying to shynthesize VHDL 2008 ghdl raises internal error exception

**Expected behaviour**
I expected the synthesis tool to work, but an exception is raised.

**How to reproduce?**
I could not find a way to reproduce the error bu removing any of the files, so a minimum example is not possible. But all of the source code is available in [here](https://github.com/FPG-I-A/k-NN/tree/master/modulos), the code files are inside the folder with its name, for example `argmin.vhdl` is in `argmin/argmin.vhdl`. The command i am running is:

```
ghdl --synth --std=08 moda.vhdl knn_top.vhdl pacote_aux.vhdl knn.vhdl argmin.vhdl insere.vhdl norm.vhdl distancias.vhdl knn_tb.vhdl -e knn_tb
```

**Context**
Please, provide the following information:

- OS: Windows 10
- Origin: available below

```

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0 (2.0.0.r1417.g7de967c51) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: i686-w64-mingw32
C:\Users\luiso\Documents\GitHub\k-NN\build\
Command line:
C:\Program Files (x86)\Ghdl\Bin\ghdl.exe --synth --std=08 moda.vhdl knn_top.vhdl pacote_aux.vhdl knn.vhdl argmin.vhdl insere.vhdl norm.vhdl distancias.vhdl knn_tb.vhdl -e knn_tb
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : synth-vhdl_static_proc.adb:111
Load address: 0xaf0000
Call stack traceback locations:
0xd66f8b 0xd71f63 0xd72fac 0xd7470c 0xd74934 0xd5b66f 0xd5b88a 0xd1d109 0xdf4660 0xdd304a 0xe96ed1 0xaf2dea 0xaf12ee 0x761d00c7 0x772c7b4c 0x772c7b1c
******************************************************************
```

**Additional context**
I am sure the code works, and it is well tested. I wanted to synthesize it just to remove the fixed_package dependancy so i could upload it to my intel FPGA without having to buy a Quartus license.

tgingold, 2023-06-28:
Your repository is not public, so I cannot get the sources.

luis705, 2023-06-28:
Oh, I'm sorry, just changed it to public.

tgingold, 2023-06-29:
The file pacote_aux.vhdl is not in the repository.  I suppose it can be generated by the pacote.py script, but the latter needs data files outside the repo.

Also, it appear you are trying to synthesize a testbench, which is not expected.

luis705, 2023-07-24:
Okay, I added all the needed files to [this gist](https://gist.github.com/luis705/c380b3753409d861cbc592fbeeeb59ec). The command, without the testbench is:

`ghdl --synth --out=vhdl --std=08 knn.vhdl argmin.vhdl distancias.vhdl insere.vhdl moda.vhdl norm.vhdl pacote_aux.vhdl top.vhdl -e top`

And it still crashes:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0 (2.0.0.r1417.g7de967c51) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: i686-w64-mingw32
C:\Users\luiso\Documents\GitHub\k-NN\build\
Command line:
C:\Program Files (x86)\Ghdl\Bin\ghdl.exe --synth --out=vhdl --std=08 knn.vhdl argmin.vhdl distancias.vhdl insere.vhdl moda.vhdl norm.vhdl pacote_aux.vhdl top.vhdl -e top
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : synth-vhdl_oper.adb:924
Load address: 0x210000
Call stack traceback locations:
0x486df8 0x491f63 0x492fac 0x49470c 0x494934 0x47b66f 0x47b88a 0x43d109 0x514660 0x4f304a 0x5b6ed1 0x212dea 0x2112ee 0x766200c7 0x777b7b1c 0x777b7aec
******************************************************************
```

tgingold, 2023-08-26:
Yes, I can reproduce the crash.  Investigating.

closed 2023-08-29 by tgingold