ghdl/ghdl #2463: 2008: Subtype not visible in inner package

bpadalino, 2023-07-04:
**Description**
An error of `subtype is not visible here` is reported when trying to reference it from an inner package.

**Expected behaviour**
The type is accessible to the scope of the inner package.

**How to reproduce?**
```vhd :file: test.vhdl
package test_pkg is

    type test_t is record
        data : bit_vector ;
    end record ;

    type config_t is record
        width : positive ;
    end record ;

    constant DEFAULT_CONFIG : config_t := config_t'(width => 10) ;

    package make is
      generic (
        CONFIG : config_t := DEFAULT_CONFIG
      ) ;

        subtype test_t is test_t(data(CONFIG.WIDTH-1 downto 0)) ;

    end package ;

end package ;

library test ;

package newtest is new test.test_pkg.make ;

entity test_e is
end entity ;

architecture arch of test_e is

    signal x : work.newtest.test_t ;

begin

end architecture ;
```

```sh :image: ghdl/ghdl:bullseye-mcode
$ ghdl -a --std=08 --work=test test.vhdl
test.vhdl:18:17:warning: declaration of "test_t" hides type "test_t" [-Whide]
        subtype test_t is test_t(data(CONFIG.WIDTH-1 downto 0)) ;
                ^
test.vhdl:18:27:error: subtype "test_t" is not visible here
        subtype test_t is test_t(data(CONFIG.WIDTH-1 downto 0)) ;
                          ^
test.vhdl:26:29:error: unit "test_pkg" not found in library "test"
package newtest is new test.test_pkg.make ;
                            ^
test.vhdl:31:22:error: entity 'test_e' was not analysed
architecture arch of test_e is
                     ^

$ ghdl --version
GHDL 4.0.0-dev (3.0.0.r1.gc68528225) [Dunoon edition]
 Compiled with GNAT Version: 10.4.0
 mcode code generator

```

tgingold, 2023-07-04:
Sorry, but the error message is correct.

In the declaration of `subtype test_t`, `test_t` refers to the declaration but is not visible (so that you cannot reference it).

bpadalino, 2023-07-04:
Why wouldn't the `subtype test_t` become a new type in the inner package which references `test_t` from the outer package?

The visibility of the inner `subtype test_t` isn't visible yet, but `test_t` from the outer package is, right?  For example, if I change the name to `subtype subtest_t`, ghdl doesn't complain.

Note that Riviera-PRO is fine with the analysis.

tgingold, 2023-07-04:
No, as soon as you define test_t, it hides the outer one.

Riviera-Pro is not correct.

bpadalino, 2023-07-04:
By that logic, I should be able to use the fully qualified name of `work.test_pkg.test_t` to disambiguate, right?

With that, I get the following errors:

```
$ ghdl -a --std=08 --work=test test.vhdl 
test.vhdl:18:17:warning: declaration of "test_t" hides type "test_t" [-Whide]
        subtype test_t is work.test_pkg.test_t(data(CONFIG.WIDTH-1 downto 0)) ;
                ^
test.vhdl:18:32:error: circular reference of package "test_pkg"
        subtype test_t is work.test_pkg.test_t(data(CONFIG.WIDTH-1 downto 0)) ;
                               ^
test.vhdl:18:41:error: no declaration for "test_t" in package "test_pkg"
        subtype test_t is work.test_pkg.test_t(data(CONFIG.WIDTH-1 downto 0)) ;
                                        ^
```

I am trying to follow along LRM-2008 section 12.3 on Visibility to see where the original is illegal.  It seems maybe NOTE 2 is the key:

>NOTE 2—The rules defining immediate scope, hiding, and visibility imply that a reference to an identifier, character literal, or operator symbol within its own declaration is illegal (except for design units). The identifier, character literal, or operator symbol hides outer homographs within its immediate scope—that is, from the start of the declaration. On the other hand, the identifier, character literal, or operator symbol is visible only after the end of the declaration (again, except for design units). 

Is that the rule you're following?

If I did want to reuse the name, the answer, it seems, might be to:

```vhdl
    alias outer_test_t is test_t ;
    sutype test_t is outer_test_t(data(CONFIG.WIDTH-1 downto 0));
```

Does this seem right to you?

closed 2023-07-04 by bpadalino

tgingold, 2023-07-04:
You can use expanded names, like this:
  subtype test_t is test_pkg.test_t(data(CONFIG.WIDTH-1 downto 0)) ;

I don't think you can use work.test_pkg, because the design units are added to the libraries after the analysis.

Aliases could be another solution.

Yes, the note re-explain why you cannot use the identifier within the declaration.

diogratia, 2023-07-06:
> Is that the rule you're following?

1.3.4 Front matter, examples, notes, references, and annexes 

>Some clauses of this standard contain examples, notes, and cross-references to other clauses of the standard;
these parts always appear at the end of a clause. Examples are meant to illustrate the possible forms of the
construct described. Illegal examples are italicized. Notes are meant to emphasize consequences of the rules
described in the clause or elsewhere. In order to distinguish notes from the other narrative portions of this
standard, notes are set as enumerated paragraphs in a font smaller than the rest of the text. Cross-references
are meant to guide the user to other relevant clauses of the standard. Examples, notes, and cross-references
are not part of the definition of the language.

Notes are not sufficient to prove adherence or lack thereof to semantic requirements by themselves not being part of the language definition.

12.2 Scope of declarations

>The scope of a declaration, except for an architecture body, extends from the beginning of the declaration to
the end of the immediately closing declarative region; the scope of an architecture body extends from the
beginning to the end of the architecture body. In either case, this part of the scope of a declaration is called
the *immediate scope*. ...

12.3 Visibility - definitions of *visible*, *hidden* and *homograph*, also where three paragraphs define what's going on:

>A declaration is visible only within a certain part of its scope; this part starts at the end of the declaration
except in the declaration of a design unit other than a PSL verification unit, a package declaration, or a
protected type declaration, in which case it starts immediately after the reserved word is occurring after the
identifier of the design unit, a package declaration, or protected type declaration. This rule applies to both
explicit and implicit declarations.

>Where it is not visible by selection, a visible declaration is said to be *directly visible*. A declaration is said to
be directly visible within a certain part of its immediate scope; this part extends to the end of the immediate
scope of the declaration but excludes places where the declaration is hidden as explained in the following paragraphs. ...
>  
>A declaration is said to be *hidden* within (part of) an inner declarative region if the inner region contains a
homograph of this declaration; the outer declaration is then hidden within the immediate scope of the inner
homograph. Each of two declarations is said to be a *homograph* of the other if and only if both declarations
have the same designator, and they denote different named entities, and either overloading is allowed for at
most one of the two, or overloading is allowed for both declarations and they have the same parameter and
result type profile (see 4.5.1).

The immediate scope in the subtype declaration blocks visibility of the type name and the subtype declaration is not until after the declaration. No declaration decorated with that name is visible within the subtype declaration. You can see NOTE 2 emphasizes this, the consequence of several semantic requirements alluded to elsewhere in the clause.

Your alias works because it is not a homograph with the type or subtype identifier and isn't hidden.

Tristan's use of a selected name with the prefix of the package name

```vhdl
 subtype test_t is test_pkg.test_t(data(CONFIG.WIDTH-1 downto 0)) ;
```
works because of a different semantic rule.

>Visibility is either by selection or direct. A declaration is visible *by selection* at places that are defined as
follows:  
>>...  
>>f)  For a declaration given in a package declaration, other than in a package declaration that defines an
uninstantiated package: at the place of the suffix in a selected name whose prefix denotes the
package.

and here you might expect the language designers anticipated a need to supplement visibility. The same rule is found in IEEE Std 1076-1987 10.3 Visibility. In fact none of the visibility issues encompassed here vary from the first release of the standard.