ghdl/ghdl #2467: GHDL bug occured on synthetising package containing shared variable

miklos-abdul, 2023-07-11:
**Description**
I was synthetising a design using the GHDL plugin in Yosys, when the GHDL bug block appeared, even though simulation (also with GHDL) worked just fine. I've narrowed the issue down to importing package that contains shared variables.

**How to reproduce?**
Since the design simulated succesfully, AFAIK it can only be reproduced using the Yosys CAD suite.

```vhd :file: package.vhd 
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

package rvapo_ice40_mem_init is
    subtype Line_t is std_logic_vector(255 downto 0);
    type memory_array_t is array (0 to 1) of Line_t;
	type memory_all_t is array (0 to 7) of memory_array_t;

	shared variable MEMORY_INIT_lo:memory_all_t:=(
		0=>(
0=>X"000000000000000000000000000000008F800000000000000000000000000000",
1=>X"000000000000000000000000000000008F800000000000000000000000000000"
			),
		others=>(others=>(others=>'0'))	
		);
end package rvapo_ice40_mem_init;
```
```vhd :file: cm.vhd 
library IEEE;
use WORK.rvapo_ice40_mem_init.all;
entity CombinedMemory_Delayed is
end;
architecture rtl of CombinedMemory_Delayed is
begin
end;
```

```
# Makefile for icestorm tools + yosys + nextpnr
# Modified from examples in icestorm distribution
# 01-16-18 E. Brombaugh

NUMBERS=package.vhd cm.vhd 

PROJ = bitstream
PIN_DEF = ../src/bitstream.pcf
SDC = ../src/bitstream.sdc
DEVICE = up5k
PACKAGE = sg48

TOOLS =/home/achmet/Documents/libs/oss-cad-suite
YOSYS = $(TOOLS)/bin/yosys
NEXTPNR = $(TOOLS)/bin/nextpnr-ice40
NEXTPNR_ARGS = --pre-pack $(SDC) --package $(PACKAGE) --$(DEVICE) --timing-allow-fail
ICEPACK = $(TOOLS)/bin/icepack
ICETIME = $(TOOLS)/bin/icetime
ICEPROG = $(TOOLS)/bin/iceprog
ICEBRAM = $(TOOLS)/bin/icebram
SENDBIT = ../../../python/send_c3usb.py
VERILATOR = verilator
TECH_LIB = $(TOOLS)/share/yosys/ice40/cells_sim.v
MODULE = -m ghdl

all: $(PROJ).bin

analyse:
	$(foreach var,$(NUMBERS),ghdl -a --std=93c --ieee=synopsys -fexplicit -g ../rvapo/core/$(var);)

%.json: analyse 										
	$(YOSYS) $(MODULE) -p 'ghdl CombinedMemory_Delayed; synth_ice40 -top CombinedMemory_Delayed -json $@'

%.asc: %.json $(PIN_DEF) 
	$(NEXTPNR) $(NEXTPNR_ARGS) --json $< --pcf $(PIN_DEF) --asc $@

%.bin: %.asc
	$(ICEPACK) $< $@

prog: $(PROJ).bin
	python3.11 $(SENDBIT) $(PROJ).bin

flash: $(PROJ).bin
	$(SENDBIT) -f $(PROJ).bin

%.rpt: %.asc
	$(ICETIME) -d $(DEVICE) -mtr $@ $<
	
lint: $(SRC)
	$(VERILATOR) --lint-only -Wall --top-module $(PROJ) $(TECH_LIB) $(SRC)

clean:
	rm -f *.json *.asc *.rpt *.bin *.hex *.o *.cf

.SECONDARY:
.PHONY: all prog clean
```

**Context**
Please, provide the following information:

- OS: Ubuntu 22.04.2
- Origin:
GHDL 4.0.0-dev (3.0.0.r169.g586f2ef5b.dirty) [Dunoon edition]
 Compiled with GNAT Version: 9.3.0
 llvm 10.0.0 code generator
Distributed as part of Yosys CAD suite.

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r169.g586f2ef5b.dirty) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-linux-gnu
/home/achmet/Documents/cvut/other/ICE-V-Wireless/Gateware/rvapo_creep/icestorm/
Command line:

Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : areapools.adb:141 access check failed
******************************************************************
```

umarcor, 2023-07-12:
I can produce a crash using `ghdl synth`:

```
# ghdl synth *.vhd -e CombinedMemory_Delayed      

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (v3.0.0-321-g472cf6443) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/w/
Command line:
ghdl synth ent.vhd pkg.vhd -e CombinedMemory_Delayed
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists.adb:416
Call stack traceback locations:
0x7f32c0b41ab2 0x5629c55ecfc4 0x5629c55ed8e4 0x5629c55fbbd6 0x5629c59190dc 0x5629c5919460 0x5629c591966f 0x5629c5919d73 0x5629c5914c7a 0x5629c5916461 0x5629c591796d 0x5629c5917a4c 0x5629c5915a32 0x5629c58e1ca4 0x5629c58e1ee8 0x5629c5820226 0x5629c5948654 0x5629c5920c7b 0x5629c59fdeac 0x5629c54b2223 0x7f32c070dd08 0x5629c54b02b8 0xfffffffffffffffe
******************************************************************
```

That's on a gcr.io/hdl-containers/ghdl (Debian Bullseye).

Same crash on MSYS2 with an older version of GHDL:

```
# ghdl synth *.vhd -e CombinedMemory_Delayed 

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r147.g6c56631a7) [Dunoon edition]
Compiled with GNAT Version: 13.1.0
Target: x86_64-w64-mingw32
F:\hdl\containers\tmp-ghdl\
Command line:
C:\msys64\mingw64\bin\ghdl.exe synth ent.vhd pkg.vhd -e CombinedMemory_Delayed
Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
Exception information:
raised ADA.ASSERTIONS.ASSERTION_ERROR : netlists.adb:416
Load address: 0x7ff750240000
Call stack traceback locations:
0x7ff750510f76 0x7ff75029424f 0x7ff750294bcd 0x7ff7502a0aa2 0x7ff7503ce6d9 0x7ff7503ce9dd 0x7ff7503ceb73 0x7ff7503cf317 0x7ff7503d16e3 0x7ff7503d2e3d 0x7ff7503d4326 0x7ff7503d43ec 0x7ff7503d2422 0x7ff750420a0a 0x7ff750420c24 0x7ff7503dde85 0x7ff7504e1a90 0x7ff75049e3ce 0x7ff7504ea100 0x7ff750242e78 0x7ff7502412ec 0x7ff750241404 0x7ffbdd7a7612 0x7ffbdefa26ef
******************************************************************
```

tgingold, 2023-07-14:
I will investigate, but I don't think that shared variables declared in a package is supported.
Just move it to the architecture.

commit 5945a006 (2023-08-26): synth-vhdl_decls: reject top-level shared variables

as they don't belong to an instance
Fix #2467
closed 2023-08-26 by tgingold