ghdl/ghdl #2470: unconstrained std_logic_vector in package instantiation (GHDL 2.0.0)

alphanumericnonsense, 2023-07-18:
**Description**
Problem instantiating package generics with unconstrained std_logic_vector.  Might be fixed in v3, I don't know.

**Two examples**
This works (-a, -e, -r, --std=08):
vhdl```
package min_gen_pkg is
    generic(
        size_g : positive;
        type my_type_t
    );
    type my_type_array_t is array (0 to size_g - 1) of my_type_t;
end package min_gen_pkg;
--------------------------------------------------
library ieee;
use ieee.std_logic_1164.all;
package slv_64_8_array_pkg is new work.min_gen_pkg
    generic map(
        size_g => 64,
        my_type_t => std_logic_vector(8-1 downto 0)
    );
--------------------------------------------------
-- library ieee;
-- use ieee.std_logic_1164.all;
-- package slv_64_array_pkg is new work.min_gen_pkg
--     generic map(
--         size_g => 64,
--         my_type_t => std_logic_vector
--     );
---------------------------------------------------
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use work.slv_64_8_array_pkg.all;
-- use work.slv_64_array_pkg.all;
entity min_gen_pkg_tb_e is

end entity min_gen_pkg_tb_e;

architecture test of min_gen_pkg_tb_e is
    signal array_s : my_type_array_t := (others => "00000000");
begin
    process is
    begin
        for i in 0 to 64-1 loop
            array_s(i) <= std_logic_vector(unsigned(array_s(i)) + i);
            wait for 1 ns;
            report to_string(array_s(i));
        end loop;
        std.env.stop;
    end process;
end architecture test;
```
But swapping out the package instantiation does not.  Analysis and elaboration OK, but an error during the run:
vhdl```
package min_gen_pkg is
    generic(
        size_g : positive;
        type my_type_t
    );
    type my_type_array_t is array (0 to size_g - 1) of my_type_t;
end package min_gen_pkg;
--------------------------------------------------
-- library ieee;
-- use ieee.std_logic_1164.all;
-- package slv_64_8_array_pkg is new work.min_gen_pkg
--     generic map(
--         size_g => 64,
--         my_type_t => std_logic_vector(8-1 downto 0)
--     );
--------------------------------------------------
library ieee;
use ieee.std_logic_1164.all;
package slv_64_array_pkg is new work.min_gen_pkg
    generic map(
        size_g => 64,
        my_type_t => std_logic_vector
    );
---------------------------------------------------
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
-- use work.slv_64_8_array_pkg.all;
use work.slv_64_array_pkg.all;
entity min_gen_pkg_tb_e is

end entity min_gen_pkg_tb_e;

architecture test of min_gen_pkg_tb_e is
    signal array_s : my_type_array_t := (others => "00000000");
begin
    process is
    begin
        for i in 0 to 64-1 loop
            array_s(i) <= std_logic_vector(unsigned(array_s(i)) + i);
            wait for 1 ns;
            report to_string(array_s(i));
        end loop;
        std.env.stop;
    end process;
end architecture test;
```
There was no "GHDL Bug" info, but a GCC bug?
sh```
ghdl -a --std=08  min_gen.vhdl
./min_gen.vhdl: In function ‘work__min_gen_pkg_tb_e__ARCH__test__DECL_ELAB’:
./min_gen.vhdl:35: internal compiler error: Segmentation fault
   35 |     signal array_s : my_type_array_t := (others => "00000000");
      | 
0x159415e diagnostic_impl
        ../../../../gcc-10.3.0/gcc/../libcpp/include/line-map.h:1309
0x1594537 internal_error(char const*, ...)
        ../../../../gcc-10.3.0/gcc/../libcpp/include/line-map.h:1711
0xc2ee8f crash_signal
        ../../../../gcc-10.3.0/gcc/toplev.c:328
0x66ebb4 new_dyadic_op
        ../../../../gcc-10.3.0/gcc/vhdl/ortho-lang.c:935
0x78cbc6 trans__chap3__index_base
        /opt/gcc-10.3.0/gcc/vhdl/trans-chap3.adb:3130
0x7a8654 trans__chap7__translate_array_aggregate_gen__do_assign_el
        /opt/gcc-10.3.0/gcc/vhdl/trans-chap7.adb:3061
0x7a8654 trans__chap7__translate_array_aggregate_gen__do_assign
        /opt/gcc-10.3.0/gcc/vhdl/trans-chap7.adb:3123
0x7a8c75 trans__chap7__translate_array_aggregate_gen__translate_array_aggregate_gen_positional
        /opt/gcc-10.3.0/gcc/vhdl/trans-chap7.adb:3191
0x7a8c75 trans__chap7__translate_array_aggregate_gen
        /opt/gcc-10.3.0/gcc/vhdl/trans-chap7.adb:3369
0x7a7ec5 trans__chap7__translate_array_aggregate
        /opt/gcc-10.3.0/gcc/vhdl/trans-chap7.adb:3628
0x7a222d trans__chap7__translate_aggregate_expression
        /opt/gcc-10.3.0/gcc/vhdl/trans-chap7.adb:4426
0x7a222d trans__chap7__translate_expression
        /opt/gcc-10.3.0/gcc/vhdl/trans-chap7.adb:4498
0x7a429d trans__chap7__translate_expression__2
        /opt/gcc-10.3.0/gcc/vhdl/trans-chap7.adb:4447
0x7c23fe trans__chap4__elab_signal_declaration_object
        /opt/gcc-10.3.0/gcc/vhdl/trans-chap4.adb:1310
0x7c6f28 trans__chap4__elab_signal_declaration
        /opt/gcc-10.3.0/gcc/vhdl/trans-chap4.adb:1336
0x7c6f28 trans__chap4__elab_declaration_chain
        /opt/gcc-10.3.0/gcc/vhdl/trans-chap4.adb:2682
0x7e1843 trans__chap9__elab_block_declarations
        /opt/gcc-10.3.0/gcc/vhdl/trans-chap9.adb:2932
0x7b5290 trans__chap1__translate_architecture_body
        /opt/gcc-10.3.0/gcc/vhdl/trans-chap1.adb:362
0x7b15a9 translation__translate
        /opt/gcc-10.3.0/gcc/vhdl/translation.adb:315
0x7e5ceb ortho_front__do_compile
        /opt/gcc-10.3.0/gcc/vhdl/ortho_front.adb:420
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
ghdl:error: exec error
```

**Context**
- OS:  Arch Linux, 6.1.38-1-lts
- Origin:
I built this from source at some point.  GHDL version info:
> ghdl --version
GHDL 2.0.0 (tarball) [Dunoon edition]
 Compiled with GNAT Version: 12.1.0
 GCC back-end code generator

alphanumericnonsense, 2023-07-18:
Need to think about what I'm doing, use newer version, see if there really is an issue....

tgingold, 2023-07-19:
Yes, I can reproduce the crash with the gcc and llvm backends.
Thank you for the small testcase!

commit 5f9840a5 (2023-08-26): vhdl-sem_assocs: propagate has_signal_flag to actual

in type associations.
Fix #2470
closed 2023-08-26 by tgingold