ghdl/ghdl #2471: Force on signal in package

avelure, 2023-07-24:
**Description**
Forcing a signal in a package from a procedure in another package gives a `signal "glob_sig" is not a formal parameter` error.

**Expected behaviour**
Expect the force to complete. It works for NVC and Modelsim.

**How to reproduce?**

```vhd :file: test.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package test_pkg is
  signal glob_sig : std_logic := '1';
end package;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use work.test_pkg.all;

package test2_pkg is
  procedure test_proc;
end package;
package body test2_pkg is
  procedure test_proc is
  begin
    glob_sig <= force '0';
    wait for 0 ps;
    assert glob_sig = '0' severity error;
    glob_sig <= release;
    wait for 0 ps;
    assert glob_sig = '1' severity error;
  end procedure;
end package body;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use work.test2_pkg.all;

entity test is
end entity;

architecture rtl of test is
begin
  p_proc : process
  begin
    test_proc;
    wait;
  end process;
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 test.vhd
ghdl --elab-run --std=08 test
```

**Context**
Please, provide the following information:

- OS: Win10
- Origin: nightly 3.0.0.r339.g0a34430cd

diogratia, 2023-08-05:
This represents an incompletely implemented -2008 feature simple force assignment representing finer granularity for signal assignment syntax and semantics.

The semantic rule found in -1993 8.4 Signal assignment statement:

>If a given procedure is declared by a declarative item that is not contained within a process statement, and if a signal assignment statement appears in that procedure, then the target of the assignment statement must be a formal parameter of the given procedure or of a parent of that procedure, or an aggregate of such formal parameters. Similarly, if a given procedure is declared by a declarative item that is not contained within a process statement, and if a signal is associated with an **inout** or **out** mode signal parameter in a subprogram call within that procedure, then the signal so associated must be a formal parameter of the given procedure or of a parent of that procedure.

The corresponding rule in -2008 10.5.2.2 Executing a simple assignment statement:

>If a given procedure is declared by a declarative item that is not contained within a process statement, and if
a simple waveform assignment statement appears in that procedure, then the target of the simple waveform
assignment shall be a formal parameter of the given procedure or of a parent of that procedure, or an
aggregate of such formal parameters. Similarly, if a given procedure is declared by a declarative item that is
not contained within a process statement, and if a signal is associated with an **inout** or **out** mode signal
parameter in a subprogram call within that procedure, then the signal so associated shall be a formal
parameter of the given procedure or of a parent of that procedure.

while the syntax in 10.5.2 Simple signal assignments, 10.5.2.1 General:

>simple_signal_assignment ::=  
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;simple_waveform_assignment  
&nbsp;&nbsp;&nbsp;&nbsp;| simple_force_assignment  
&nbsp;&nbsp;&nbsp;&nbsp;| simple_release_assignment  

distinguishes a simple waveform assignment from a simple force assignment. The exclusion for a simple force assignment to the semantic rule isn't implemented in ghdl judging from the error message.

tgingold, 2023-08-24:
Thank you for the issue, I am testing a fix for it.

commit 9990e6eb (2023-08-25): vhdl-sem_stmts: do not add drivers for force/release

Fix #2471
closed 2023-08-25 by tgingold