ghdl/ghdl #2474: raised CONSTRAINT_ERROR : elab-vhdl_context.adb:453 discriminant check failed

sensille, 2023-07-29:
**Description**
When trying to convert a set of vhdl-files to verilog I receive the error pasted below.

GHDL from oss-cad-suite-20221018 does not show this error, but fails silently (exit code 1).

**How to reproduce?**
The attached archive contains all sources and the script ghdl.sh which contains the invocation that failed.
[ghdlbug.zip](https://github.com/ghdl/ghdl/files/12206745/ghdlbug.zip)

**Context**
- OS: Arch linux on x64
- Origin: https://github.com/YosysHQ/oss-cad-suite-build/releases/download/2023-07-29/oss-cad-suite-linux-x64-20230729.tgz

```
******************* GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r340.g664bf403b.dirty) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/arne/ghdlbug/
Command line:
/home/arne/conan/oss-cad-suite-20230729/libexec/ghdl --synth --std=08 --out=verilog --work=unisim -fsynopsys -frelaxed log2.vhd IDROMConst.vhd idrom.vhd dpram.vhd InputPinsPerModule.vhd irqlogics.vhd OutputInteg.vhd outm.vhd MaxOutputPinsPerModule.vhd hostmotid.vhd decodedstrobe.vhd decodedstrobe2.vhd oneofndecode.vhd adpram.vhd lutsrl16.vhd lutsrl32.vhd d8o8sqwsb.vhd d8o8sqws.vhd twidrom.vhd twiddle.vhd pwmpdmgenh.vhd inmuxm.vhd simplessi.vhd srl16delay.vhd qcounterated.vhd fanucabs.vhd dpainter.vhd uartr8blutsrl.vhd waveram.vhd GetModuleHint.vhd ubrategen.vhd syncwavegen.vhd qcountersfpd.vhd timestampd.vhd NumberOfModules.vhd sslbprom43.vhd sslbprom46.vhd kubstepgenz.vhd sslbpram.vhd uartx.vhd uartr.vhd uartx8lutsrl.vhd uartr8lutsrl.vhd uartx8blutsrl.vhd sserialwa.vhd timestamp.vhd ubrategend.vhd rcpwmgen.vhd resrom.vhd parity.vhd pktuartx.vhd scalercounter.vhd qcounterateskd.vhd fixicap.vhd qcountersf.vhd rcpwmrate.vhd binosc.vhd kubstepgenzd.vhd OutputPinsPerModule.vhd ibound.vhd oneshot.vhd qcounterate.vhd resolverdaq2.vhd scalertimer.vhd qcounteratesk.vhd pktuartr.vhd kubstepgenzi.vhd threephasepwm.vhd biss_loop.vhd qcountersfd.vhd cpdrive.vhd wordpr.vhd sserialwab.vhd b32qcondmac2ws.vhd MaxInputPinsPerModule.vhd sine16.vhd ModuleExists.vhd PinExists.vhd inm.vhd CountPinsInRange.vhd hmtimers.vhd MaxIOPinsPerModule.vhd kubstepgenzid.vhd wordrb.vhd qcountersfp.vhd simplespix.vhd daqfifo16.vhd resolver.vhd pwmrefh.vhd bufferedspi.vhd MaxPinsPerModule.vhd xy2mod.vhd boutreg.vhd wavegen.vhd xfrmrout.vhd watchdog.vhd drqlogic.vhd hostmot2.vhd consts_gen.vhd hostmot2_top.vhd -e TopHostMot2
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : elab-vhdl_context.adb:453 discriminant check failed
Call stack traceback locations:
0x7ff2efa9e884 0x7ff2efa9e8d7 0x7ff2efc31314 0x7ff2efbfd628 0x7ff2efbfd57d 0x7ff2efbfe2ad 0x7ff2efbfe5ea 0x7ff2efb3bc93 0x7ff2efc7dcd8 0x7ff2efc3c593 0x7ff2efc8766f 0x7ff2ef8c175f 0x7ff2ef1fb0b1 0x7ff2ef8bfeec 0xfffffffffffffffe
```

tgingold, 2023-08-21:
I can reproduce the crash.

tgingold, 2023-08-22:
I was able to fix the crash, but the design is using tri-state internally, which is not yet fully supported.

sensille, 2023-08-22:
Great news, thanks :)
Which part of the internal tri-state is not supported? Conversion to muxes would be sufficient. The design is another open source project which I'm trying to port, so I want to make as few modifications as possible.

tgingold, 2023-08-22:
In the top-level, many sub-entities are assigning obus.

Unfortunately, it is not quick to change.  But I am investigating what I 
could do.

tgingold, 2023-08-24:
Now fixed.  The design is now handled without errors.
Thank you for the issue.

closed 2023-08-24 by tgingold

sensille, 2023-08-28:
Design translates now and yosys synthesizes it without complaints. Thanks!

commit 91f8e792 (2023-08-24): synth-environment: handle partial resolved assignment.

Fix #2474