ghdl/ghdl #2475: GHDL LSP: Symbol errors

JJ-Gaisler, 2023-08-02:
**Description**
GHDL LSP crashes after updating GHDL from `7de967c51f352fe2d724dbec549b71a392e5ebae`.

**How to reproduce?**
I have not had time to find a minimal reproducible example but in the latest version of grlib the LSP crashes immediately in some files and in some files it works until I do a go to definition.

It is possible to reproduce this issue using the GPL version of grlib this time. This can be found here: [grlib](https://www.gaisler.com/products/grlib/grlib-gpl-2023.2-b4283.tar.gz)

1.  Once downloaded, go to the root directory of GRLIB.
2. Use this [script](https://pastebin.com/pXv8kH7r) to generate the hdl-prj.json file. Before running the script the GRLIB environment variable has to be set to the root of GRLIB. Enter the design directory "designs/noelv-generic" when prompted. Enter 93 as the vhdl version. 
3. Open the file $GRLIB/lib/gaisler/noelv/core/nvsupport.vhd, the LSP should now crash.
4. Open the file $GRLIB/lib/gaisler/noelv/core/iunv.vhd, the LSP starts and inline diagnostics is displayed. If i try to go to definition on any function, for example is_rv then the LSP also crashes. 

**Context**
Please, provide the following information:

- OS: OpenSuse leap 15.4
- Origin: 
  - [ ] Built from sources: `6c56631a711591a26cc3a7e1b2d12417b8d17c66`
  - [ ] PyGHDL ghdl-ls 4.0.0-dev (re-installed with the ghdl commit hash)

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
[ERROR][2023-08-02 11:09:24] .../vim/lsp/rpc.lua:734	"rpc"	"ghdl-ls"	"stderr"	"2023-08-02 11:09:24,358 [ERROR] Caught exception while handling textDocument/documentSymbol with params 
{
  'textDocument': {'uri': 'file:///home/jonathan/Downloads/grlib-gpl-2023.2-b4283/lib/gaisler/noelv/core/nvsupport.vhd'}
}:
Traceback (most recent call last):
  File \"/home/jonathan/.local/lib/python3.10/site-packages/pyGHDL/lsp/lsp.py\", line 137, in handle
    response = fmethod(**params)
  File \"/home/jonathan/.local/lib/python3.10/site-packages/pyGHDL/lsp/vhdl_ls.py\", line 127, in textDocument_documentSymbol
    return doc.document_symbols()
  File \"/home/jonathan/.local/lib/python3.10/site-packages/pyGHDL/lsp/document.py\", line 200, in document_symbols
    syms = symbols.get_symbols_chain(self._fe, nodes.Get_First_Design_Unit(self._tree))
  File \"/home/jonathan/.local/lib/python3.10/site-packages/pyGHDL/lsp/symbols.py\", line 84, in get_symbols_chain
    res = [get_symbols(fe, el) for el in pyutils.chain_iter(n)]
  File \"/home/jonathan/.local/lib/python3.10/site-packages/pyGHDL/lsp/symbols.py\", line 84, in <listcomp>
    res = [get_symbols(fe, el) for el in pyutils.chain_iter(n)]
  File \"/home/jonathan/.local/lib/python3.10/site-packages/pyGHDL/lsp/symbols.py\", line 93, in get_symbols
    return get_symbols(fe, nodes.Get_Library_Unit(n))
  File \"/home/jonathan/.local/lib/python3.10/site-packages/pyGHDL/lsp/symbols.py\", line 156, in get_symbols
    children.extend(get_symbols_chain(fe, nodes.Get_Declaration_Chain(n)))
  File \"/home/jonathan/.local/lib/python3.10/site-packages/pyGHDL/lsp/symbols.py\", line 84, in get_symbols_chain
    res = [get_symbols(fe, el) for el in pyutils.chain_iter(n)]
  File \"/home/jonathan/.local/lib/python3.10/site-packages/pyGHDL/lsp/symbols.py\", line 84, in <listcomp>
    res = [get_symbols(fe, el) for el in pyutils.chain_iter(n)]
  File \"/home/jonathan/.local/lib/python3.10/site-packages/pyGHDL/lsp/symbols.py\", line 96, in get_symbols
    raise AssertionError(f\"get_symbol: unhandled {pyutils.kind_image(k)}\")
AssertionError: get_symbol: unhandled Library_Declaration

[ERROR][2023-08-02 11:27:03] .../vim/lsp/rpc.lua:734	"rpc"	"ghdl-ls"	"stderr"	"\nraised ADA.ASSERTIONS.ASSERTION_ERROR : no field Identifier\n"

```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2023-08-16:
I thought I replied by email, but I cannot see it here.

I simply was not able to reproduce the crash. 
First obvious question: did you rebuild both libghdl and pyGHDL ?  If they are not in sync it could be a problem.

Second, as this is using gpl-grlib, use:
  export GHDL_LS_TRACE=ghdl-ls
before starting the editor.  Then ghdl-ls will create rotating log files (ghdl-ls.in and ghdl-ls.out).
Send me the .1 version of the .in file, I will try to replay it.

JJ-Gaisler, 2023-08-17:
Hi and thank you for the reply.

It seems like I'm the one at fault this time. I was convinced I had rebuilt everything but restarting from scratch with a new clone of ghdl seems to have resolved the issue. I suppose something must have gone wrong when i rebuilt last time. Pardon the inconvenience.

closed 2023-08-17 by JJ-Gaisler