ghdl/ghdl #2476: Crash on (-r) Launch Using Subtypes

Elkniwt, 2023-08-10:
**Description**
Upon trying to launch a simulation, I get this:

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r266.g3ce6e75) [Dunoon edition]
Compiled with GNAT Version: 4.9.3
Target: x86_64-linux-gnu
/home/jspeter/work/src/git/vhdl-ws-xv/.xv-aa-shared/src/shared/internal/sim_tests/axi_package/ensemble1/ghdl.run/
Command line:
ghdl -r --ieee=synopsys -fexplicit sim_top --wave=tb.ghw --ieee-asserts=disable
Exception ORTHO_CODE.SYNTAX_ERROR raised
Exception information:
Exception name: ORTHO_CODE.SYNTAX_ERROR
Message: ortho_code-exprs.adb:462
Call stack traceback locations:
0x4eb033 0x4ecbb9 0x4cc439 0x5b3ce9 0x5b4355 0x5b4738 0x6cb785 0x6cc0dd 0x6cc8bc 0x6ce3cb 0x6ce52d 0x6aa023 0x6af8b1 0x6af6ec 0x6af9f7 0x7312f8 0x68b732 0x72f6cf 0x696fbc 0x625314 0x762b3b 0x575838 0x598ed8 0x40a7f1 0x7f255aaa583e 0x408df7 0xfffffffffffffffe
******************************************************************

The wave file (tb.ghw) was not yet created.

**Expected behaviour**
On a different machine, simulating the same VHDL runs fine with this version:

GHDL 4.0.0-dev (3.0.0.r15.gc0975bc8a) [Dunoon edition]
 Compiled with GNAT Version: 7.5.0

**How to reproduce?**
This was a largish collection of VHDL files I'm not inclined to share.  strace doesn't show much useful info.  Maybe the answer is just to make sure everything is up to date (though it looks like the crashing version is more recent?).

It has something to do with this code:

```
	wbypass : if TRUE generate
		subtype ID_RANGE   is natural range ID_WIDTH-1 downto 0;
		subtype RESP_RANGE is natural range ID_WIDTH+1 downto ID_WIDTH;
		constant LAST_IDX : natural := ID_WIDTH+2;
		signal data_in  : std_logic_vector(LAST_IDX downto 0);
		signal data_out : std_logic_vector(LAST_IDX downto 0);
	begin
		data_in(ID_RANGE)   <= wbyp_in.id;
		data_in(RESP_RANGE) <= wbyp_in.resp;
		data_in(LAST_IDX)   <= wbyp_in.last;

		fifo : general_fifo
			generic map (
				DATA_WIDTH => data_in'length,
				LOG2_DEPTH => BYP_FIFO_DEPTH_LOG2 -- this must be >= log2(LATENCY)
			)
			port map (
				reset => reset,
				clock => aclk,

				push  => wbyp_push,
				din   => data_in,
				full  => wbyp_full,

				empty => wbyp_empty,
				dout  => data_out,
				pull  => wbyp_pull
			);

		wbyp_out.id   <= data_out(ID_RANGE);
		wbyp_out.resp <= data_out(RESP_RANGE);
		wbyp_out.last <= data_out(LAST_IDX);
	end generate;
```

In particular, the way in which RESP_RANGE is used to pack data into and out of a std_logic_vector.  Commenting out the lines that define/use RESP_RANGE allow simulation to proceed.

Strangely, swapping the two ranges like so:
```
		subtype ID_RANGE   is natural range ID_WIDTH+1 downto 2;
		subtype RESP_RANGE is natural range 1 downto 0;
```

also allows it to proceed.

**Context**

- OS:
  crashes on Ubuntu 16.04.3 LTS
  succeeds on Ubuntu 18.04.6 LTS
  
  ^ these versions are irrelevant.  I can "git pull" and make it crash on 18.04 v

- Origin:
  build from commit 3ce6e7562f7d43ecb911ebbaa47961017002ae3c crashes
  build from older commit c0975bc8a3e2ba9058a2236d01c79aaec3dc4169 succeeds

  These hashes are in linear order, newest first (but not necessarily adjacent):
  
  664bf403b75d3747f8df84e7614ca96ab0802618 crashes
  3ce6e7562f7d43ecb911ebbaa47961017002ae3c crashes
  d2c1b7a5c1a1f68e7b8f99f29c46d1f050452963 crashes
  2d34c88b80275b5812ac62a3bf83354c41494265 crashes
  b72b92ad29ec7b60a3abd8b344c6bddca0c68ee8 crashes  *** these are adjacent
  aff11ca0c49a8c0c6917bace81976a92ff7dc6b9 succeeds *** these are adjacent
  cefc7ebafa0869dc9f67c675969e36a4041643c8 succeeds
  f721c20920414f86d97e025aea8587e116368471 succeeds
  10d6f0c3520c2ebb93091f635014c6cd4c35631f succeeds
  a211cdcf7d5eb5f058d2e38c93092f4548283451 succeeds
  3495e979361d3464073a2cf4acf5a249943d5f23 succeeds
  c0975bc8a3e2ba9058a2236d01c79aaec3dc4169 succeeds
  
  This seems to be the culprit (it's a smallish change):
  
commit b72b92ad29ec7b60a3abd8b344c6bddca0c68ee8 (HEAD)
Author: Tristan Gingold <tgingold@free.fr>
Date:   Thu Apr 20 18:26:01 2023 +0200

    trans-chap7: optimize is_a_derived_type for #2417
    
    Avoid useless subtype conversions

  Coincidentally, the VHDL involved uses subtypes.  Let me know if I can help debug further.
  
If a `GHDL Bug occurred` block is shown in the log, please paste it here:
[see above]

diogratia, 2023-08-11:
Any non-file object declaration involves a subtype, your comment  "Coincidentally, the VHDL involved uses subtypes. Let me know if I can help debug further." doesn't convey your intent clearly.

A good faith attempt to provide a reproducer from your question:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity general_fifo is
    generic (
        DATA_WIDTH:     positive;
        LOG2_DEPTH:     natural
    );
    port (
        reset:          in  std_logic;
        clock:          in  std_logic;
        push:           in  std_logic;
        din:            in  std_logic_vector (DATA_WIDTH - 1 downto 0);
        full:           out std_logic;
        empty:          out std_logic;
        dout:           out std_logic_vector (DATA_WIDTH - 1 downto 0);
        pull:           in  std_logic
    );
end entity;

architecture foo of general_fifo is
begin
end architecture;

library ieee;
use ieee.std_logic_1164.all;

entity bug_from_2417_fix is
end entity;

architecture good_faith of bug_from_2417_fix is
    
    constant ID_WIDTH:  natural := 4;
    type wbyp_type  is record
        id:     std_logic_vector (ID_WIDTH - 1 downto 0);
        resp:   std_logic_vector (ID_WIDTH + 1 downto ID_WIDTH);
        last:   std_logic;
    end record;
    constant DATA_WIDTH:    natural := ID_WIDTH + 3;
    
    constant  BYP_FIFO_DEPTH_LOG2:  natural := 16;

    signal reset:       std_logic;
    signal aclk:        std_logic;
    signal push:        std_logic;
    signal din:         std_logic_vector (DATA_WIDTH - 1 downto 0);
    signal full:        std_logic;
    signal empty:       std_logic;
    signal dout:        std_logic_vector (DATA_WIDTH - 1 downto 0);
    signal pull:        std_logic;
    
    signal wbyp_in:     wbyp_type;
    signal wbyp_out:    wbyp_type;
    signal wbyp_push:   std_logic;
    signal wbyp_full:   std_logic;
    signal wbyp_empty:  std_logic;
    signal wbyp_pull:   std_logic;
    
    component general_fifo is
        generic (
            DATA_WIDTH:     positive;
            LOG2_DEPTH:     natural
        );
        port (
            reset:          in  std_logic;
            clock:          in  std_logic;
            push:           in  std_logic;
            din:            in  std_logic_vector (DATA_WIDTH - 1 downto 0);
            full:           out std_logic;
            empty:          out std_logic;
            dout:           out std_logic_vector (DATA_WIDTH - 1 downto 0);
            pull:           in  std_logic
        );
    end component;

begin
    
    
wbypass: 
    if TRUE generate
        subtype ID_RANGE   is natural range ID_WIDTH-1 downto 0;
    	subtype RESP_RANGE is natural range ID_WIDTH+1 downto ID_WIDTH;
    	constant LAST_IDX : natural := ID_WIDTH+2;
    	signal data_in  : std_logic_vector(LAST_IDX downto 0);
    	signal data_out : std_logic_vector(LAST_IDX downto 0);
    begin
    	data_in(ID_RANGE)   <= wbyp_in.id;
    	data_in(RESP_RANGE) <= wbyp_in.resp;
    	data_in(LAST_IDX)   <= wbyp_in.last;

fifo:
        general_fifo
		    generic map (
    			DATA_WIDTH => data_in'length,
    			LOG2_DEPTH => BYP_FIFO_DEPTH_LOG2 -- this must be >= log2(LATENCY)
		    )
    		port map (
    			reset => reset,
    			clock => aclk,

    			push  => wbyp_push,
    			din   => data_in,
    			full  => wbyp_full,

    			empty => wbyp_empty,
    			dout  => data_out,
    			pull  => wbyp_pull
    		);
    	wbyp_out.id   <= data_out(ID_RANGE);
    	wbyp_out.resp <= data_out(RESP_RANGE);
    	wbyp_out.last <= data_out(LAST_IDX);
    end generate;
end architecture;
```
Doesn't crash, it analyzes (-a), elaborates (-e) and simulates (-r) without doing anything useful. That suggests there may be something relating to how DATA_WIDTH is calculated causing a crash or there's a need for a reproducer.

```none
%: ghdl --version
GHDL 4.0.0-dev (3.0.0.r339.g0a34430cd) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 llvm 15.0.7 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
%: 
```

(OR it's since been fixed, or it's platform specific, ...)

Elkniwt, 2023-08-11:
You're right.  I haven't tried hard enough.  That code you provide does not
cause the bug for me, either, but this code does:

library IEEE;
use IEEE.std_logic_1164.all;

entity bug_from_2417_fix is
 generic (
   ID_WIDTH : integer := 16
 );
 port (
   aclk     : in    std_logic
 );
end entity;

architecture rtl of bug_from_2417_fix is

 signal reset : std_logic;

 -- Bypass-data FIFO signals:
 type bypass_data_type is record
   resp : std_logic_vector(1 downto 0);
   id   : std_logic_vector(ID_WIDTH-1 downto 0);
   last : std_logic;
 end record;

 signal wbyp_in     : bypass_data_type;
 signal wbyp_out    : bypass_data_type;

begin

 wbypass : if TRUE generate
   subtype ID_RANGE   is natural range ID_WIDTH-1 downto 0;
   subtype RESP_RANGE is natural range ID_WIDTH+1 downto ID_WIDTH;
   constant LAST_IDX : natural := ID_WIDTH+2;
   signal data_in  : std_logic_vector(LAST_IDX downto 0);
   signal data_out : std_logic_vector(LAST_IDX downto 0);
 begin
   data_in(ID_RANGE)   <= wbyp_in.id;
   data_in(RESP_RANGE) <= wbyp_in.resp;
   data_in(LAST_IDX)   <= wbyp_in.last;

   wbyp_out.id   <= data_out(ID_RANGE);
   wbyp_out.resp <= data_out(RESP_RANGE);
   wbyp_out.last <= data_out(LAST_IDX);
 end generate;

end architecture;

When I save this to 'test.vhd' and run:

ghdl -a --ieee=synopsys -fexplicit test.vhd
ghdl -r --ieee=synopsys -fexplicit bug_from_2417_fix --wave=tb.ghw
--ieee-asserts=disable

I see the error

--Jim

Elkniwt, 2023-08-11:
Wow... apparently e-mail replies are terrible.  Here it is, better formatted:

```
library IEEE;
use IEEE.std_logic_1164.all;

entity bug_from_2417_fix is
  generic (
    ID_WIDTH : integer := 16
  );
  port (
    aclk     : in    std_logic
  );
end entity;

architecture rtl of bug_from_2417_fix is

  signal reset : std_logic;

  -- Bypass-data FIFO signals:
  type bypass_data_type is record
    resp : std_logic_vector(1 downto 0);
    id   : std_logic_vector(ID_WIDTH-1 downto 0);
    last : std_logic;
  end record;

  signal wbyp_in     : bypass_data_type;
  signal wbyp_out    : bypass_data_type;

begin

  wbypass : if TRUE generate
    subtype ID_RANGE   is natural range ID_WIDTH-1 downto 0;
    subtype RESP_RANGE is natural range ID_WIDTH+1 downto ID_WIDTH;
    constant LAST_IDX : natural := ID_WIDTH+2;
    signal data_in  : std_logic_vector(LAST_IDX downto 0);
    signal data_out : std_logic_vector(LAST_IDX downto 0);
  begin
    data_in(ID_RANGE)   <= wbyp_in.id;
    data_in(RESP_RANGE) <= wbyp_in.resp;
    data_in(LAST_IDX)   <= wbyp_in.last;

    wbyp_out.id   <= data_out(ID_RANGE);
    wbyp_out.resp <= data_out(RESP_RANGE);
    wbyp_out.last <= data_out(LAST_IDX);
  end generate;

end architecture;

```

diogratia, 2023-08-11:
And the code you just provided works for me:

```none
%: ghdl -a bug_from_2417_fix.vhdl
%: ghdl -e bug_from_2417_fix
%: ghdl -r bug_from_2417_fix
%: ghdl --version
GHDL 4.0.0-dev (3.0.0.r339.g0a34430cd) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 llvm 15.0.7 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
%:  
```

The mcode version with the same commit runs successfully as well.

Elkniwt, 2023-08-11:
That's interesting.  On a much more up-to-date machine, the code does not crash.  Perhaps it's something to do with a bug in gnat?  This issue should probably be closed.

0[sparkle:~]1003: ghdl -a bug_from_2417_fix.vhd
0[sparkle:~]1004: ghdl -e bug_from_2417_fix
0[sparkle:~]1005: ghdl -r bug_from_2417_fix
0[sparkle:~]1006: ghdl -v
GHDL 4.0.0-dev (3.0.0.r32.g863df0ba7) [Dunoon edition]
 Compiled with GNAT Version: 12.1.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
0[sparkle:~]1007: cat /etc/issue
Ubuntu 22.04.3 LTS \n \l

diogratia, 2023-08-14:
I convinced Tristan it'd be useful to identify the llvm back end version which got added to the ghdl revision command output. Your first post on this issue might reflect a new enough commit to provide that with a revision report.

The reason it's useful is ABI incompatibility for dynamically linked libraries between llvm releases where ghdl doesn't specify exactly which version is required (nor are the necessary dylibs decorated with the revision). In this case for a Mac it's necessary to either use Homebrew to install a particular version of llvm or otherwise install the two dylibs in a specific location in contrived file hierarchy absent setting up a dylib  search path.

It's not inconceivable you could encounter similar issues under Linux where historically distribution releases and readily installable packages are closely tied to gcc releases.

The mcode back end versions of ghdl tend to be immune from development tool versions not containing a lot of dylib dependencies (although IIRC there's a relatively recent but benign zlib dependency).

Altering ghdl for platform dependency lib dependency checking is an open ended problem and not addressed for prebuilt binary distributions while otherwise only showing up as build failures. Building from source is generally the answer to unsupported platforms or distributions.

In any event for a large enough user base for a particular target platform not addressed by a prebuilt binary capturing the actual issue can be valuable for other end users of the same platform and would imply debugging, including exploring the backtrace to distinguish between silent incompatibilities and ghdl coding errors.

You could also imagine using statically linked libraries but that may carry distribution license issues as well as storage size on a download hub.

tgingold, 2023-08-16:
I can reproduce the crash from @Elkniwt on a machine, so I can investigate.
Also thanks to @diogratia for the initial reproducer!

commit 199a6026 (2023-08-19): trans-chap7(is_a_derived_type): adjust

fix the check for new index constraints
Fix #2476
closed 2023-08-19 by tgingold