ghdl/ghdl #2478: Force command is applied successfully but fails writing waveform files.

sweeneyal, 2023-08-16:
**Description**
Using the force command like in the below example causes failure in logging waveform for GTKWave or VCD.

**Expected behaviour**
Expecting the waveform file to continue on til the end of the scenario, but instead stops halfway. I can verify that the scenario continues after the force command, as the report statement will print out the value of `sigOut`, and it is correctly forced, and the `finish` command will indicate that it finished at 40 ns, but the waveform only logs up until 20 ns, when the force command is applied.

**How to reproduce?**
Reproduce as following:

```vhd :file: DoubleInverter.vhd
library ieee;
    use ieee.std_logic_1164.all;
    use ieee.numeric_std.all;

entity DoubleInverter is
    port (
        SigIn  : in  std_logic;
        SigOut : out std_logic
    );
end entity DoubleInverter;

architecture rtl of DoubleInverter is
    signal SigBtw : std_logic;
begin
    
    SigBtw <= not SigIn after 5 ns;
    SigOut <= not SigBtw after 5 ns;

    
end architecture rtl;
```

```testbench: TbForce.vhd
library ieee;
    use ieee.std_logic_1164.all;
    use ieee.numeric_std.all;

entity TbForce is
end entity TbForce;

architecture tb of TbForce is
    signal sigIn  : std_logic;
    signal sigOut : std_logic;
begin

    Inv : entity work.DoubleInverter port map(SigIn => sigIn, sigOut => sigOut);
    
    TestForce: process
    begin
        sigIn <= '0';
        wait for 20 ns;
        sigOut <= force '1';
        wait for 20 ns;
        report "sigOut is " & std_logic'image(sigOut);
        std.env.finish;
    end process TestForce;
    
end architecture tb;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 DoubleInverter.vhd
ghdl -a --std=08 TbForce.vhd
ghdl -e --std=08 TbForce
ghdl -r --std=08 TbForce --wave=test.ghw
ghdl -r --std=08 TbForce --vcd=test.vcd
```

**Context**
- OS: Windows 11
- Origin:
  - [ ] Package manager: `version`
  - [ X ] Released binaries: `[v3.0.0](https://github.com/ghdl/ghdl/releases/tag/v3.0.0)`
  - [ ] Built from sources: `commit SHA`

tgingold, 2023-08-24:
I am not sure to understand the issue: nothing happens after 20ns.  Can you clarify what do you expect ?

sweeneyal, 2023-08-24:
The testbench runs for 40 ns, as is seen in the `TestForce` process. The first 20 ns are nominal, the second 20 ns are after the force command. When the test is run via the provided shell script, the result is the following:
```
TbForce.vhd:21:9:@40ns:(report note): sigOut is '1'
simulation finished @40ns
TbForce.vhd:21:9:@40ns:(report note): sigOut is '1'
simulation finished @40ns
```
Looking at the VCD file in a text editor, we can see the last time stamp is `#2000000`, and opening the output GHW file in Gtkwave shows that there's no more signals logged after 20 ns.

I'm expecting the last 20 ns to be logged for the waveform viewers, but they are not.

tgingold, 2023-08-24:
Yes, but the signal values don't change after 20ns.
If I understand correctly you'd like to see the waveform until 40ns.  Is it correct ?

sweeneyal, 2023-08-24:
Yes that is correct. My understanding was that even if the signals do not change, it would still log the signal for the remaining time?

closed 2023-08-24 by sweeneyal

tgingold, 2023-08-25:
I am not sure this is a mistake.  Probably the waveform could be improved.

umarcor, 2023-08-28:
@sweeneyal as a trick, I typically set a signal to true at the beginning of the main testbench process and set it to false right before finishing/stopping/terminating the execution.

Find a discussion about this same issue in https://github.com/ghdl/ghdl/issues/1187.