ghdl/ghdl #2480: Improving VDC output: add traces for State Machines

josyb, 2023-08-27:
**Is your feature request related to a problem? Please describe.**
I am debugging a system which obviously contains  state-machines as *A State Machine is worth a thousand Equations*
GHDL does not emit any traces of these in the `.vcd` output.
Instead we are told to use the `,ghw` output.
Unfortunately the Impulse plugin for Eclipse doesn't support GHW.
So we try GTKWave
But the latest available binary for Windows (I could find) 3.3.100 silently ignores GHW files.
Ok we will behave and compile the latets version: 3.3.117
Installing MSYS2 takes over an hour and eats away 5 GB disk space.
Then the fun starts: 
Download an unzip GTKWave 3.3.117.tar.gz
./autogen.sh doesn't run
Search on the internet: install more MSYS2 stuff
then ./configure fails: can't find some Tcl script
Again search on the internet but no solution found
Giving up. Live is too short ...

**Describe the solution you'd like**
Simply add the state-machine signals in the .vcd output like this:
```
$var string 2 ) smn $end
$var string 2 * smp $end

...
0%
0&
0'
0(
sIDLE )
sIDLE *
0+
0,
0-
0.
0/
```

**Describe alternatives you've considered**
I tried `ghwdump` to see whether I could do something with that output; but it doesn't seem to generate an output if no [OPTIONS] are specified

**Additional context**
[image]

diogratia, 2023-08-27:
I originally looked at Issue 2480 to see what VDC stands for. It is as yet not clear there's a ghdl issue you're addressing here.

For a small example:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity foo is
end entity;

architecture fum of foo is
    type some_enum is (a, b, c, d, e, f, g);
    signal state:   some_enum;
begin
    process
    begin
        for i in some_enum'LEFT to some_enum'RIGHT loop
            state <= i;
            wait for 10 ns;
            report LF & HT & "state = " & some_enum'image(state);
        end loop;
        wait for 10 ns;
        wait;
    end process;
end architecture;
```
```none
ghdl -r foo --vcd=foo.vcd
```
produced the vcd file
```none
%: more *.vcd
$date
  Mon Aug 28 01:00:04 2023
$end
$version
  GHDL v0
$end
$timescale
  1 fs
$end
$scope module standard $end
$upscope $end
$scope module std_logic_1164 $end
$upscope $end
$scope module foo $end
$comment state is not handled $end
$upscope $end
$enddefinitions $end
#0
#10000000
#20000000
#30000000
#40000000
#50000000
#60000000
#70000000
#80000000
foo.vcd (END)
```
which didn't do anything useful, GTKWave shows nothing (there's nothing else beside the enumerate type signal).

While 
```none
ghdl -r --wave=foo.ghw
```
demonstrates the GHW format supporting enumerated types other than well known types (character, bit, std_logic, ...):

[image]

If this doesn't match your experience, please provide a small reproducer.

GTKWave originally provides a side channel method of passing enumeration values in a file. **enum** was introduced in SystemVerilog, wasn't it? Doesn't show up in IEEE Std 1364-2005. In the mean time Tristan had experimented around with GHW.

Support custom types in VCD waveforms #647  

Add option to export only standards compliant VCD [#675](https://github.com/amaranth-lang/amaranth/issues/675) 

Shouldn't the currrent issue contain VCD and VDC?

josyb, 2023-08-27:
Sorry, I didn't catch my typo; just corrected that.
My question is about the generation of .vcd traces of the simulation.
There is no issue with them except that they don't show the state machines.
And outputting .ghw files is not a solution as none of the other tools (on Windows) can read them.

diogratia, 2023-08-28:
GTKWave has the ability use translation filter files which contain two columns, the first for the original name and the second for the substitute name.

For the example here a second signal is added, here named count which holds the position value of the enumerated type value:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity foo is
end entity;

architecture fum of foo is
    type some_enum is (a, b, c, d, e, f, g);
    signal state:   some_enum;
    signal count:   natural;  
begin
    process
    begin
        for i in some_enum'LEFT to some_enum'RIGHT loop
            state <= i;
            count <= some_enum'pos(i);
            wait for 10 ns;
            report LF & HT & "state = " & some_enum'image(state);
        end loop;
        wait for 10 ns;
        wait;
    end process;
end architecture;
```

The simulation is run
```none
%: ghdl -r foo --vcd=foo.vcd
foo.vhdl:18:13:@10ns:(report note):
	state = a
foo.vhdl:18:13:@20ns:(report note):
	state = b
foo.vhdl:18:13:@30ns:(report note):
	state = c
foo.vhdl:18:13:@40ns:(report note):
	state = d
foo.vhdl:18:13:@50ns:(report note):
	state = e
foo.vhdl:18:13:@60ns:(report note):
	state = f
foo.vhdl:18:13:@70ns:(report note):
	state = g
%: 
```
producing the foo_vcd.vcd file which can be opened with gtkwave.

A gtkwave translation filter file is generated for the enumeration type (here translate.txt):
```none
# enumerated type some_enum translate filter file
0 a
1 b
2 c
3 d
4 e
5 f
6 g
```
and yes, that's a comment.

Loading the vcd file in GTKWave and adding the integer signal representing the position value of state allows you to attach the translation filter to the integer count:

[image]

You can select the filter file then will find the integer position values are substituted by enumeration values:

[image]

If you then write the GTKWave savefile the Waves signal setup will be stored in the save file (should have a .gtkw suffix). You can open the save file in the future with GTKWave.

foo_vcd.gtkw:
```none
%: more foo_vcd.gtkw
[*]
[*] GTKWave Analyzer v3.3.107 (w)1999-2020 BSI
[*] Sun Aug 27 23:03:52 2023
[*]
[dumpfile] "/Users/dio_gratia/Desktop/foob/foo.vcd"
[dumpfile_mtime] "Sun Aug 27 22:23:29 2023"
[dumpfile_size] 423
[savefile] "/Users/dio_gratia/Desktop/foob/foo_vcd.gtkw"
[timestart] 0
[size] 1000 600
[pos] 289 67
*-24.505035 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1
[treeopen] foo.
[sst_width] 198
[signals_width] 78
[sst_expanded] 1
[sst_vpaned_height] 164
@2421
^2 /Users/dio_gratia/Desktop/foob/translate.txt
foo.count
[pattern_trace] 1
[pattern_trace] 0
```

I had a hard time under MacOS Ventura doing this because of poor GTK3 compatibility, the pre-built gtkwave.app is built for OS X 10.6.  I had to paste an absolute path into the filter list then finangled the interface to return me to the regular GTKWave window. How to do this is found in the GTKWave manual (a PDF) under **Alias Files and Attaching External Disassemblers**.

While I have no say in the matter I'm not in favor of adding syntactical and semantic extensions to VCD dump files in ghdl. The reason being [they aren't universally supported](https://electronics.stackexchange.com/questions/530123/viewing-verilog-enums-in-iverilog-gtkwave). VCD and most other common dump file formats suffer from a lack of standardization. 

That being said it's up to Tristan to decide to extend the syntax. You can see from the last link that the various flavors of translation filters GTKWave supports are a useful compromise if all you worry about is GTKWave. 

If you can identify a common set of syntax extensions to a wide enough set of specified tools it'd go a long way as an argument for changing ghdl, although it would also likely be useful to have a way to disable such extensions.

Another method comes to mind. The use of an external filter to edit a VCD file.

I found I could translate foo_vcd.vcd:
```none
$date
  Mon Aug 28 10:23:29 2023
$end
$version
  GHDL v0
$end
$timescale
  1 fs
$end
$scope module standard $end
$upscope $end
$scope module std_logic_1164 $end
$upscope $end
$scope module foo $end
$comment state is not handled $end
$var integer 32 ! count $end
$upscope $end
$enddefinitions $end
#0
b0 !
#10000000
b1 !
#20000000
b10 !
#30000000
b11 !
#40000000
b100 !
#50000000
b101 !
#60000000
b110 !
#70000000
#80000000
```
by hand changing the $var declaration to string substituting the name for the signal of the enumerated type which was dropped and switching enumeration values from integer values to strings:
```none
$date
	Mon Aug 28 12:55:18 2023
$end
$version
	GTKWave Analyzer v3.3.107 (w)1999-2020 BSI
$end
$timescale
	1fs
$end
$scope module foo $end
$var string 7 " state $end
$upscope $end
$enddefinitions $end
#0
$dumpvars
Sa "
$end
#10000000
Sb "
#20000000
Sc "
#30000000
Sd "
#40000000
Se "
#50000000
Sf "
#60000000
Sg "
#80000000
```
An inline or post processing filter would need to be enumeration type and target variable (signal) name aware. 

GTKWave successfully opens the modified VCD file.

[image]

This closely resembles the method GTKWave uses for filter translation. It'd just be moved outside to allow dump file use by other tools.  

 It'd be more efficient to implement it in ghdl.  An external filter might imply multiple passes for multiple variables of enumeration types because VCD isn't type extensible.

Paebbels, 2023-08-28:
GTKwave also exists for Windows.

See assets of https://github.com/gtkwave/gtkwave/releases/tag/nightly

umarcor, 2023-08-28:
FTR, gtkwave is prebuilt in MSYS2 and available through the default package manager: https://packages.msys2.org/base/mingw-w64-gtkwave. It is not necessary to built it from sources.

OTOH, @josyb do you use signals or variables to hold the state of the FSMs? I believe the issue might be that variables are not logged to waveforms.

josyb, 2023-08-28:
@all
I'd rather use the Impulse extension inside Eclipse - I have used/tried GTKWave before and didn't find its UI that great.
I do all my FPGA work in Eclipse: MyHDL and Python development with PyDev, VHDL-Verilog with Sigasi (also using GHDL toolchain) - Impulse to view simulation results (from MyHDL which handles state machine signals as `string`). It would be nice if I could use Impulse view GHDL simulation output too.

@Paebbels 
Thanks for the link to the `nighlty` build - I might try that next time; I found a workaround and don't want to spend/loose more time
I also couldn't resist myself here and downloaded  [gtkwave_gtk2_mingw64_standalone.tgz](https://github.com/gtkwave/gtkwave/releases/download/nightly/gtkwave_gtk2_mingw64_standalone.tgz) 
and decompressed it; same effect as the MSYS2 installed variant recommended by @umarcor : it quietly exits when I try to load the .ghw file. The file size is 24 MB but if that is a restriction ...

@umarcor
FSM always use signals; certainly the two- or three- process variants I use; you might get away with using a variable for a single-process variant but I have never seen that.
Anyway variables ought to be traced too ... probably are
Likewise I will check the MSYS2 - I couldn't resist myself and gave it a try: the application silently exits when I load the .ghw file; so that doesn't help :(

@diogratia 
I was not really looking for lessons in GTKWave. The work-around for the state machine signal is not a workable or acceptable solution.

@all
The question I wanted to raise was : _"Why are state machine signals not visible in the .vcd output by GHDL?"_
1) They could have been output as `reg` - to at least show a numerical value - there really is no good excuse why this isn't the case
2) better is to output them as `string` - as I show in the OP - I understand this a non-standard extension but we use that in MyHDL so I would think it is acceptable:
[image]

Adding a command line option to enable this would keep things standard for the less adventurous

tgingold, 2023-08-29:
I am testing a change to support enum in vcd.

diogratia, 2023-08-29:
*"Why are state machine signals not visible in the .vcd output by GHDL?"*  

There is no VHDL language construct 'state machine'.  The two examples you give appear to refer to signals of an enumerated type. You provide neither declaration of an enumerated type nor of the two signals. 

The declaration of an enumerated type also declares the static names of positional values (enumeration literals, enumeration_literal ::= identifier | character_literal).

The VHDL standard doesn't support VCD which has a syntactical and semantic definition in the Verilog standards (IEEE Std 1364 and IEEE Std 1800 SystemVerilog). The VCD format doesn't specify a var type of string nor a vector value change format prefix of S or s as accepted by GTKWave. Var type is specified as one of event | integer | parameter | real | realtime | reg | supply0 | supply1 | time |  tri | triand | trior | trireg | tri0 | tri1 | wand | wire | wor.

There are no formal mapping mechanisms defined for other types. 

Why are enumeration values not visible in VCD? Barring non-standard extensions to the VCD format enumeration types can't be represented in a VCD.

A corollary question might be 'Why do you expect your non-standard state values to be present in a ghdl VCD dumpfile? Holding up a proprietary tool isn't inspiring, although that it can be done in GTKWave is.

Looking at [Impulse HDL](https://toem.de/index.php/products/impulse-hdl) it also accepts FST using a reader based on GTKWave code. The ghdl FST output is currently broken for at least integer  types but does support enumerated types that show up with GTKWave. It's listed as a good first ghdl contribution project (and hasn't had any takers yet). Everyone looks to GTKWave source code as the reference on FST. 

A check with fst2vcd (which accompanies GTKWave) it appears count declared as an integer in the second example is seen as a bit by GTKWwave (FST is a compressed binary format we can't look at directly although ghwdump --help will also provide information on how to use that tool).

```none
%: more foo_fst.vcd
$date
        Mon Aug 28 03:56:50 2023

$end
$version
        GHDL FST v0
$end
$timescale
        1fs
$end
$var string 0 ! state $end
$var integer 1 " count $end
$enddefinitions $end
#0
$dumpvars
0"
sa !
$end
#10000000
sb !
1"
#20000000
0"
sc !
#30000000
sd !
1"
#40000000
0"
se !
#50000000
sf !
1"
#60000000
0"
sg !
#70000000
#80000000
```

With foo_fst.vcd  we see the integer var type with a size of 1 and the value given as a single decimal digit.

Looking to the original foo_vcd.vcd the difference is the count size is 32  and count values are  binary radix values (b0, b1, b01, b11, b100, b101, and b110). That tells us there should be a simple ghdl fix for the FST issue with integer output. See IEEE Std 1800-2017 6.11  Integer data types, Table 6-8, var type integer is a vector comprised of 4-state type elements representing bits). Changing the radix values also demonstrated the need for a space between the value and identifier code (") in the .vcd file,  specified in the standard:

21.7.2.2 Formats of variable values 

>Variables can be either scalars or vectors. Each type is dumped in its own format. Dumps of value changes to scalar variables shall not have any white space between the value and the identifier code.
>  
>Dumps of value changes to vectors shall not have any white space between the base letter and the value digits, but they shall have one white space between the value digits and the identifier code.

As far as changing ghdl to output a position value as an integer for an enumerated value, that certainly doesn't conflict with  21.7.2.1 Syntax of 4-state VCD file. That would be an easier fix than passing the enumeration value and would require the equivalent of GTKWaves translation filters to see the enumeration values.  

The FST fix needs to be made anyway (eventually).

We can look forward to examining the change Tristan is testing to support enum.

josyb, 2023-08-29:
@diogratia 
> There is no VHDL language construct 'state machine'. The two examples you give appear to refer to signals of an enumerated type. You provide neither declaration of an enumerated type nor of the two signals.

You can be as pedantic as me; clearly I intended the output of VHDL `enumeration` as used in a state-machine. Although there is also use outside state-machines. I don't think that an example was really necessary here. 

> The VCD format doesn't specify a var type of string nor a vector value change format prefix of S or s as accepted by GTKWave. 

Yes, I have read that spec too; I specifically mentioned that a `string` is not in the VCD spec. But at least two applications, GTKWave and Impulse HDL, support them. I guess that ModelSim will also output these in a .vcd dump.

> Why are enumeration values not visible in VCD? Barring non-standard extensions to the VCD format enumeration types can't be represented in a VCD.
A corollary question might be 'Why do you expect your non-standard state values to be present in a ghdl VCD dumpfile? Holding up a proprietary tool isn't inspiring, although that it can be done in GTKWave is.

It would be a non-standard extension that the user has to formally specify as an option in the invocation command

> Looking at [Impulse HDL](https://toem.de/index.php/products/impulse-hdl) it also accepts FST using a reader based on GTKWave code. The ghdl FST output is currently broken for at least integer types but does support enumerated types that show up with GTKWave.

I missed that GHDL can output FST; it would be good idea to use this for large simulation projects; but as the GHDL output is broken .... 
Furthermore FST is also a moving target as there is no formal specification.

tgingold, 2023-08-29:
You need to use option `--vcd-enums` to enable enums dump.

closed 2023-08-29 by tgingold

denwarenjii, 2025-01-18:
> `--vcd-enums`

Sorry for replying to a year old thread but I cannot get this to work on GHDL 3.0.0. There is also nothing about it in the documentation.