ghdl/ghdl #2487: GHDL crashes when running UVVM v3 testbench

mariuselv, 2023-09-05:
**Description**
GHDL crash when trying to run testbench on [UVVM BETA 3](https://github.com/UVVM/UVVM_3_BETA).

**How to reproduce?**
Run the testbench in [bitvis_vip_axi/tb/maintenance_tb/axi_vvc_tb.vhd](https://github.com/UVVM/UVVM_3_BETA/blob/main/bitvis_vip_axi/tb/maintenance_tb/axi_vvc_tb.vhd)

MacOS Ventura 13.5.1
GHDL 3.0.0 (2.0.0.r1417.g7de967c51) [Dunoon edition]

```
GHDL release: 3.0.0 (2.0.0.r1417.g7de967c51) [Dunoon edition]
Compiled with GNAT Version: Community 2019 (20190517-83)
Target: x86_64-apple-darwin17.7.0
/dev/bitvis/uvvm/bitvis_vip_axi/sim/hdlregression/test/axi_vvc_tb/43712842/
Command line:
ghdl --elab-run --std=08 --ieee=standard -frelaxed-rules --warn-no-shared --warn-no-hide --workdir=/dev/bitvis/uvvm/bitvis_vip_axi/sim/hdlregression/library/bitvis_vip_axi/v08 --work=bitvis_vip_axi -P/dev/bitvis/uvvm/bitvis_vip_axi/sim/hdlregression/library/ axi_vvc_tb sim
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap6.adb:1212
Load address: 0x102ce4000
Call stack traceback locations:
0x10309320a 0x10309352c 0x103102624 0x1030ffc40 0x103104098 0x103108fb5 0x10310901d 0x10310c3e2 0x1031119ba 0x1030d4178 0x1030c7231 0x1030d8b1e 0x1030d2d1c 0x10311c627 0x103055c2e 0x10304593a 0x10311d790 0x102ce6f7c
******************************************************************
Error: Program ended with exit code 2

```

tgingold, 2023-09-07:
I haven't tried to reproduce, but your version of ghdl is a little bit old.

tgingold, 2023-09-23:
But in the test 1238 I cannot see the failures anymore (they are present in test 1236).
It is already fixed ?

tgingold, 2023-09-24:
I tried to reproduce the crash, without success.  Probably it has been fixed.

Please, reopen if not (it would also be useful to file all the commands used to analyze).

closed 2023-09-24 by tgingold