ghdl/ghdl #2488: port width mismatch causes an exception

stfarley, 2023-09-07:
**Description**
When the width of a std_logic vector mismatches that of the port it is connected to an exception is raised.

**Expected behaviour**
An error should be reported indicating where the issue is..

**How to reproduce?**
Attempt to run a simulation with a mismatched port assignement

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r424.g6a50056cb.dirty) [Dunoon edition]
Compiled with GNAT Version: Community 2020 (20200818-93)
Target: x86_64-pc-linux-gnu
/projects/systemrdl/gsim/
Command line:
ghdl --elab-run --std=08 -frelaxed-rules -Plibs -Puvvm -Posvvm --workdir=libs --work=testbench tb --stop-time=1ms --wave=gsim.ghw
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : grt-signals.adb:286 access check failed
*

tgingold, 2023-09-10:
Do you have a reproducer ?

stfarley, 2023-09-11:
[tc1.zip](https://github.com/ghdl/ghdl/files/12576164/tc1.zip)

I crafted this one to reproduce the issue. If sigb.b is sized 1 bit smaller in compa is is also not detected but right shifts the data onto the port.

diogratia, 2023-09-11:
This has different behavior in ghdl-macos-llvm (3.0.0.r472.g43f5fa48c, a recent daily) 

With -e and -r 
```none
/usr/local/bin/ghdl -a  --std=08 compa_pkg.vhdl
/usr/local/bin/ghdl -a  --std=08 compa.vhdl
/usr/local/bin/ghdl -a  --std=08 top.vhdl
/usr/local/bin/ghdl -a  --std=08 tb.vhdl
/usr/local/bin/ghdl -e  --std=08 tb
/usr/local/bin/ghdl -r --std=08 tb
make: *** [run] Error 255
%: 
```

That's all the error we get.

With 

```none
%: /usr/local/bin/ghdl --elab-run --std=08 tb --wave=tb.ghw --stop-time=1ms
%: 
```

it's slent and the GHW file has a size of 0. The difference appears to be something in exception handling. (And I miss the days of OSX/MacOS doing it's own exception dumps).

The mcode version gives the error:

```none
~/opt/share/3.0.0.r472.g43f5fa48c/ghdl-macos-11-mcode/bin/ghdl -a  --std=08 compa_pkg.vhdl
~/opt/share/3.0.0.r472.g43f5fa48c/ghdl-macos-11-mcode/bin/ghdl -a  --std=08 compa.vhdl
~/opt/share/3.0.0.r472.g43f5fa48c/ghdl-macos-11-mcode/bin/ghdl -a  --std=08 top.vhdl
~/opt/share/3.0.0.r472.g43f5fa48c/ghdl-macos-11-mcode/bin/ghdl -a  --std=08 tb.vhdl
~/opt/share/3.0.0.r472.g43f5fa48c/ghdl-macos-11-mcode/bin/ghdl -e  --std=08 tb
~/opt/share/3.0.0.r472.g43f5fa48c/ghdl-macos-11-mcode/bin/ghdl -r --std=08 tb

raised CONSTRAINT_ERROR : grt-signals.adb:286 access check failed
make: *** [run] Error 1

%: 
```
--elab-run gives the same error:

```none
%: ~/opt/share/3.0.0.r472.g43f5fa48c/ghdl-macos-11-mcode/bin/ghdl --elab-run --std=08 tb --wave=tb.ghw --stop-time=1ms

raised CONSTRAINT_ERROR : grt-signals.adb:286 access check failed
%: 
```

closed 2023-09-12 by tgingold

stfarley, 2023-09-13:
Great work Guys.
I can confirm that it works properly now.

commit f72fa80d (2023-09-12): vhdl: add missing bounds check for record assocs

Fix #2488