ghdl/ghdl #2489: Difference in calling to_signed with literal vs variable - both of same value

JimLewis, 2023-09-08:
**Description**
When called with a literal, to_signed does not produce a vector truncated where as when it is called with a variable with the same value, it correctly produces the vector truncated warning.   

**Expected behaviour**
I thought I would get two messages of the form:   
../../src/ieee2008/numeric_std-body.vhdl:3117:7:@1ns:(assertion warning): NUMERIC_STD.TO_SIGNED: vector truncated

Instead, I am only getting one:   
C:/SynthWorks/Dev/_osvvm/OsvvmLibraries/Common/TbParamsPt/TbGhdl.vhd:75:5:@1ns:(report note): test 1;  to_signed(11, 4)
C:/SynthWorks/Dev/_osvvm/OsvvmLibraries/Common/TbParamsPt/TbGhdl.vhd:79:5:@1ns:(report note): test 1;  D := 11 to_signed(D, 4)
../../src/ieee2008/numeric_std-body.vhdl:3117:7:@1ns:(assertion warning): NUMERIC_STD.TO_SIGNED: vector truncated

I suspect that the first expression is being evaluated during analysis, so no message.   A curiosity.    I just was not expecting.   It made it harder to chase the bug I was trying to track down.   

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: ent.vhd
library ieee ;
  use ieee.std_logic_1164.all ;
  use ieee.numeric_std.all ;
entity ent is
end entity;

architecture a of ent is
begin
  TestProc : process
    variable A_sv4 : signed(3 downto 0) ; 
    variable D : integer ; 
  begin
  
    report "test 1;  to_signed(11, 4)" ;
    A_sv4 := to_signed(11, 4);
 
    D := 11 ; 
    report "test 1;  D := " & to_string(D) & " to_signed(D, 4)" ;
    A_sv4 := to_signed(D, 4);
    
    std.env.stop;
end;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

diogratia, 2023-09-09:
There are  syntax errors for the lack of an `end process;` as well as a semantic errors for the lack of the passing --std=08 from your example invocation.

Avoiding a diatribe and fixing those shows the single warning:

```none
%: ghdl -a --std=08 ent.vhdl
%: ghdl -e --std=08 ent
%: ghdl -r ent
ent.vhdl:14:5:@0ms:(report note): test 1;  to_signed(11, 4)
ent.vhdl:18:5:@0ms:(report note): test 1;  D := 11 to_signed(D, 4)
../../src/ieee2008/numeric_std-body.vhdl:3117:7:@0ms:(assertion warning): NUMERIC_STD.TO_SIGNED: vector truncated
%: 
```
where the warning that shows up is for the to_signed call with the variable D as a parameter.

/usr/local/lib/ghdl/src/ieee2008/numeric_std-body.vhdl:3096:

```vhdl
  -- Id: D.4
  function TO_SIGNED (ARG : INTEGER; SIZE : NATURAL) return UNRESOLVED_SIGNED is
    variable RESULT : UNRESOLVED_SIGNED(SIZE-1 downto 0);
    variable B_VAL  : STD_LOGIC := '0';
    variable I_VAL  : INTEGER   := ARG;
  begin
    if (SIZE < 1) then return NAS;
    end if;
    if (ARG < 0) then
      B_VAL := '1';
      I_VAL := -(ARG+1);
    end if;
    for I in 0 to RESULT'left loop
      if (I_VAL mod 2) = 0 then
        RESULT(I) := B_VAL;
      else
        RESULT(I) := not B_VAL;
      end if;
      I_VAL := I_VAL/2;
    end loop;
    if ((I_VAL /= 0) or (B_VAL /= RESULT(RESULT'left))) then
      assert NO_WARNING
        report "NUMERIC_STD.TO_SIGNED: vector truncated"
        severity warning;
    end if;
    return RESULT;
  end function TO_SIGNED;
```
and line 3117 is the report statement.

So why does `to_signed(D, 4)` produce the warning but the literal 11 does not?  Both invocations have the same value for I_VAL, B_VAL and RESULT. Proving that involves making a local copy of the to_signed function along with with a couple of declarations (NAS, a variable declared of type line) and adding the impure caveat to the function definition. When you do that both warnings show up.
 
With the default --std=93c:

If I substitute a wait statement with no clauses for the env.stop call and gen up a to_string[integer return string]:

```vhdl
library ieee ;
  use ieee.std_logic_1164.all ;
  use ieee.numeric_std.all ;
entity ent1 is
end entity;

architecture a of ent1 is
begin
  TestProc : process
    variable A_sv4 : signed(3 downto 0) ; 
    variable D : integer ; 
    function to_string (inp: integer) return string is
    begin
        return integer'image (inp);
    end function;
  begin
  
    report "test 1;  to_signed(11, 4)" ;
    A_sv4 := to_signed(11, 4);
 
    D := 11 ; 
    report "test 1;  D := " & to_string(D) & " to_signed(D, 4)" ;
    A_sv4 := to_signed(D, 4);
    
    wait; -- std.env.stop;
  end process;
end;
```
the result is 

```none
%: ghdl -a ent1.vhdl
%: ghdl -e ent1
%: ghdl -r ent1
ent1.vhdl:18:5:@0ms:(report note): test 1;  to_signed(11, 4)
../../src/ieee/v93/numeric_std-body.vhdl:2179:7:@0ms:(assertion warning): NUMERIC_STD.TO_SIGNED: vector truncated
ent1.vhdl:22:5:@0ms:(report note): test 1;  D := 11 to_signed(D, 4)
../../src/ieee/v93/numeric_std-body.vhdl:2179:7:@0ms:(assertion warning): NUMERIC_STD.TO_SIGNED: vector truncated
%: 
```
both warnings (and no env.stop simulation stopped message).

It looks to me like there's an assumption for the -2008 case (IEEE Std 1076-2008 9.4.2 Locally static primaries):

>An expression is said to be locally static if and only if every operator in the expression denotes an implicitly
defined operator or an operator defined in one of the packages STD_LOGIC_1164, NUMERIC_BIT,
NUMERIC_STD, NUMERIC_BIT_UNSIGNED, or NUMERIC_STD_UNSIGNED in library IEEE, and if
every primary in the expression is a locally static primary, where a locally static primary is defined to be one
of the following:  
>>  
>> a) A literal of any type other than type TIME  
>>...  
>> e) A function call whose function name denotes an implicitly defined operation or an operation defined
in one of the packages STD_LOGIC_1164, NUMERIC_BIT, NUMERIC_STD, NUMERIC_BIT_UNSIGNED, or NUMERIC_STD_UNSIGNED in library IEEE and whose actual parameters are each locally static expressions

that prevents the report statement from being streamed for the expression with the literal 11.

Basically, the report statement output for the to_signed call is falling on the floor. This is due to the 9.4 Static expressions changes that assume the to_signed function call with a locally static expression (the literal 11) as a parameter is locally static.

I'm somewhat uncomfortable with requiring the function call be evaluated during elaboration if it's a locally static expression. I'd imagine there could be other cases besides this one, dependent on locally static parameter value.

JimLewis, 2023-09-09:
Is it evaluated during elaboration or analysis?    If it is during analysis, it would be nice to output the message.   If it is elaboration, I would not want to slow simulation startup.

OTOH, simulators tend to bother to give warnings about NULL ranges during analysis, so why not this?

tgingold, 2023-09-09:
As said by @diogratia, after lrm08 the expression is static so evaluated before simulation.

diogratia, 2023-09-09:
You can always pass the value as a generic default:

```vhdl
library ieee ;
  use ieee.std_logic_1164.all ;
  use ieee.numeric_std.all ;
entity ent2 is
    generic (intval: integer := 11);
end entity;

architecture a of ent2 is
begin
  TestProc : process
    variable A_sv4 : signed(3 downto 0) ; 
    variable D : integer ; 
  begin
  
    report "test 1;  to_signed(intval, 4)" & " fintval = " &  to_string(intval);
    A_sv4 := to_signed(intval, 4);
 
    D := 11 ; 
    report "test 1;  D := " & to_string(D) & " to_signed(D, 4)" ;
    A_sv4 := to_signed(D, 4);
    
    std.env.stop;
  end process;
end;
```
keeping in mind that the part of elaboration expressed as loading occurs under the run command:

```none
%: ghdl -a --std=08 ent2.vhdl
%: ghdl -e --std=08 ent2
%: ghdl -r ent2
ent2.vhdl:15:5:@0ms:(report note): test 1;  to_signed(intval, 4) fintval = 11
../../src/ieee2008/numeric_std-body.vhdl:3117:7:@0ms:(assertion warning): NUMERIC_STD.TO_SIGNED: vector truncated
ent2.vhdl:19:5:@0ms:(report note): test 1;  D := 11 to_signed(D, 4)
../../src/ieee2008/numeric_std-body.vhdl:3117:7:@0ms:(assertion warning): NUMERIC_STD.TO_SIGNED: vector truncated
simulation stopped @0ms
%: 
```
Analysis is about placing design units in design libraries. Above, the simulation kernel is present when the expression is evaluated. The VHDL language understanding of warning is limited to severity level for assertions. It literally doesn't care what is warned about in a design specification.

For analysis you'd want a static analysis tools (a lint) that was very much VHDL aware (based on libghdl?) and could treat warnings as errors. That could also be a smart editor pointing out potentially consequences during design entry. [Sigasi doesn't do this](https://insights.sigasi.com/manual/eclipse/linting-vhdl/).

The need is probably proportional to overall design model complexity and  level of obscurity through abstraction, or can be inversely proportional to the user's experience or abilities and project design process formalism.

commit 01f248ef (2023-09-10): synth-vhdl_eval: emit warning for truncated values

for to_signed/to_unsigned.
Fix #2489
closed 2023-09-10 by tgingold