ghdl/ghdl #2490: Dual port RAM ghdl-yosys-plugin yosys assertion error

JulianKemmerer, 2023-09-11:
**Description**
I am hitting this assertion and erroring out
https://github.com/YosysHQ/yosys/blob/41b34a19353dbbe00aa08f3561e25e0bfa4c84d2/kernel/mem.cc#L473C5-L473C33
```
if ((port.transparency_mask[j] || port.collision_x_mask[j]) && !wport.removed) {
                log_assert(port.clk_enable);
```

This is when trying to convert some VHDL->Verilog (using ghdl plugin front end)
I think it has to do with how the HDL describes a dual port ram? (does successfully map to Xilinx BRAMs)

**Expected behaviour**
No assertion error.

**How to reproduce?**
``` 
cd ram_error
bash top/convert_to_verilog.sh
```

convert_to_verilog.sh:
```
/usr/local/bin/ghdl -i --std=08 -frelaxed `cat vhdl_files.txt`
/usr/local/bin/ghdl -m --std=08 -frelaxed top 
/usr/local/bin/yosys -g -m ghdl  -p "ghdl --std=08 -frelaxed top;  write_verilog top.v"
```

output snippet:
```
./main.c/the_ram/the_ram_0CLK_b45f1687.vhd:31:19:warning: type of a shared variable must be a protected type [-Wshared]
  shared variable the_ram : ram_t := (others => (others => '0')); 
                  ^
...

-- Running command `ghdl --std=08 -frelaxed top;  write_verilog top.v' --

1. Executing GHDL.
./main.c/the_ram/the_ram_0CLK_b45f1687.vhd:31:19:note: found RAM "the_ram", width: 8 bits, depth: 19200
  shared variable the_ram : ram_t := (others => (others => '0')); 
                  ^
Importing module top.
Importing module main_0clk_88c0218f.
Importing module the_ram_0clk_b45f1687.

2. Executing Verilog backend.

2.1. Executing BMUXMAP pass.

2.2. Executing DEMUXMAP pass.
ERROR: Assert `port.clk_enable' failed in kernel/mem.cc:473.
```

**Context**
GHDL 4.0.0-dev (3.0.0.r448.g2245a7fd1) [Dunoon edition]

Yosys 0.32+76 (git sha1 73cb4977b, clang 10.0.0-4ubuntu1 -fPIC -Os)

Yosys dev said this is an issue with how GHDL is emitting the RAM... :shrug: 

Does this have to do with the relaxed-to-warning `shared variable must be a protected type` error?

Thanks for your time

[ram_error.zip](https://github.com/ghdl/ghdl/files/12571909/ram_error.zip)

tgingold, 2023-09-11:
I will investigate, but if you simply need to convert from vhdl to 
verilog, you don't need yosys.  You can simply do:

ghdl synth --out=verilog FILES -e

JulianKemmerer, 2023-09-11:
Thank you for considering looking into this.

I did indeed forget that GHDL alone can convert to Verilog! I will absolutely try this :fist_raised: 
However, I feel like I once heard that the path through Yosys is generally 'better' for producing verilog - without details on what that means? I suppose we'll see :nerd_face: