ghdl/ghdl #2491: Duplicate wires and assign-to-self in Verilog output

JulianKemmerer, 2023-09-12:
**Description**
`synth --out=verilog` produces duplicate wire definitions. This is a compile error in tools like Verilator. Also code has `assign` of a wire to itself emitted - probably also an error.

**Expected behaviour**
Only one copy of wire declared instead of multiple. No assignment of wire to self.

**How to reproduce?**
``` 
cd dup_wires
/usr/local/bin/ghdl synth --std=08 -frelaxed --out=verilog `cat vhdl_files.txt` -e top > top.v
cat top.v | grep const_sr_1_main_c_l114_c10_7dba_return_output 
```

See duplicate definition of `const_sr_1_...` wire in output:
```
$ cat top.v | grep const_sr_1_main_c_l114_c10_7dba_return_output 
  wire [31:0] const_sr_1_main_c_l114_c10_7dba_return_output;
  wire [31:0] const_sr_1_main_c_l114_c10_7dba_return_output;
  assign return_output = const_sr_1_main_c_l114_c10_7dba_return_output;
  assign const_sr_1_main_c_l114_c10_7dba_return_output = const_sr_1_main_c_l114_c10_7dba_return_output; // (signal)
    .return_output(const_sr_1_main_c_l114_c10_7dba_return_output));

```

**Context**
```
GHDL 4.0.0-dev (3.0.0.r448.g2245a7fd1) [Dunoon edition]

Verilator 5.015 devel rev v5.014-50-g5b7e32581
```

[dup_wires.zip](https://github.com/ghdl/ghdl/files/12582030/dup_wires.zip)

tgingold, 2023-09-21:
Yes, I think this is a known issue.  I need to write a pass to distinguish those duplicated names.

markus-zzz, 2024-06-29:
This hit me as well with
```
GHDL 4.1.0 (4.0.0.r39.g7188e92cf) [Dunoon edition]
```
trying to convert https://github.com/mist-devel/T65.git as follows
```
$ ghdl -i --std=08 --work=t65 --workdir=build -Pbuild core/myc64-rtl/T65/*.vhd
$ ghdl -m -fsynopsys --std=08 --work=t65 --workdir=build T65
$ ghdl synth -fsynopsys --std=08 --work=t65 --workdir=build -Pbuild --out=verilog T65 > T65.v
```
results in duplicate wire `alu_q` and Verilator is unhappy.

zerkman, 2024-07-06:
Hi,
I also got the bug with GHDL 4.1.0. I wrote a minimalistic VHDL example source code:

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library work;
use work.all;

entity sub is
	port (
		clk : in std_logic;
		reset : in std_logic;
		x : out std_logic
	);
end entity;

architecture hdl of sub is
	signal z : std_logic;
begin
x <= z;
process(clk,reset)
begin
	if reset = '1' then
		z <= '0';
	elsif rising_edge(clk) then
		z <= not z;
	end if;
end process;
end architecture;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity main is
	port (
		clk : in std_logic;
		reset : in std_logic;
		sub_1_x : out std_logic
	);
end entity;

architecture hdl of main is
begin
	sub_1: entity work.sub port map (
		clk => clk,
		reset => reset,
		x => sub_1_x
	);
end architecture;
```

The resulting verilog synth output for the `main` module is:
```
module main
  (input  clk,
   input  reset,
   output sub_1_x);
  wire sub_1_x;
  assign sub_1_x = sub_1_x;
  /* test.vhd:45:9  */
  sub sub_1 (
    .clk(clk),
    .reset(reset),
    .x(sub_1_x));
endmodule
```
Here a `sub_1_x` is instantiated with the same name as an output signal.
Just removing the wire instantiation solves the problem. Not sure this would work in all cases of this error.

Xiretza, 2024-07-06:
See also #1154.