ghdl/ghdl #2493: Missing package use clause causes crash

stfarley, 2023-09-14:
**Description**
Missing package use clause causes a crash.
The error is reported but then it crashes.
This is pretty minor as the issue is identified.

**Expected behaviour**
Issue an Error for the undefined type

**How to reproduce?**
ghdl -a -v -g --std=08 -frelaxed-rules --ieee=standard --workdir=libs --work=work -Plibs -Puvvm -Posvvm src/work/tb.vhd
```vhd :file: ent.vhd

ghdl -a -v -g --std=08 -frelaxed-rules --ieee=standard  --workdir=libs --work=work -Plibs  -Puvvm -Posvvm src/work/tb.vhd
src/work/tb.vhd:33:32:error: no declaration for "test_rec"
        <<signal.tb.top_i.sigb:test_rec(a(11 downto 0), b(11 downto 0))>>.a <= force (others => '1');
                               ^
sem_selected_name(2): cannot handle IIR_KIND_EXTERNAL_SIGNAL_NAME (src/work/tb.vhd:33:9)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r493.gfeae724a0.dirty) [Dunoon edition]
Compiled with GNAT Version: Community 2020 (20200818-93)
Target: x86_64-pc-linux-gnu
/projects/ghdl_issues/new/gsim/
Command line:
ghdl -a -v -g --std=08 -frelaxed-rules --ieee=standard --workdir=libs --work=work -Plibs -Puvvm -Posvvm src/work/tb.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
******************************************************************

tgingold, 2023-09-16:
It would be nice to have a reproducer too.

stfarley, 2023-09-16:
[tc.zip](https://github.com/ghdl/ghdl/files/12640560/tc.zip)

Sorry about that the upload must have failed initially.

tgingold, 2023-09-17:
Now fixed.

closed 2023-09-17 by tgingold