ghdl/ghdl #2494: Crash when calling a procedure with hierarchical signal reference

stfarley, 2023-09-15:
**Description**
Calling a procedure with a hierarchical signal reference causes a crash at compile time. I believe that this is valid code for VHDL 2008.

**Expected behaviour**
Successful compile

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r493.gfeae724a0.dirty) [Dunoon edition]
Compiled with GNAT Version: Community 2020 (20200818-93)
Target: x86_64-pc-linux-gnu
/projects/ghdl_issues/new/gsim/
Command line:
ghdl -a -v -g --std=08 -frelaxed-rules --ieee=standard --workdir=libs --work=work -Plibs -Puvvm -Posvvm src/work/tb.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
******************************************************************
[tc.zip](https://github.com/ghdl/ghdl/files/12618165/tc.zip)

diogratia, 2023-09-15:
Note you left out the 'error' message preceding the Internal Error exception:

```none
/usr/local/bin/ghdl -a  --std=08 tb.vhdl
sem_selected_name(2): cannot handle IIR_KIND_EXTERNAL_SIGNAL_NAME (tb.vhdl:33:26)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r493.gfeae724a0) [Dunoon edition]
Compiled with GNAT Version: Community 2019 (20190517-83)
Target: x86_64-apple-darwin17.7.0
/Users/diogratia/Desktop/tc/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -o tb.o tb.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Load address: 0x100701000
Call stack traceback locations:
0x1008de36a 0x100a797df 0x100a8004f 0x100a70710 0x100a50f86 0x100a64e79 0x100a87a45 0x100a87e1c 0x100a87ecd 0x100a88cc8 0x100a88db8 0x100a896a0 0x100a89918 0x100a89a77 0x100a487cb 0x100a5088f 0x100a215cb 0x100b4be07 0x100b4d7f8 0x100b4ead6 0x1007078a0
******************************************************************
/usr/local/bin/ghdl:error: compilation error
%: 
```

iIIR_KIND_EXTERNAL_SIGNAL_NAME tells us there's an external name use not supported

```vhdl
test_record_proc(<<signal.tb.top_i.sigb:test_rec(a(11 downto 0), b(11 downto 0))>>.a);
```
Here as an actual to mode out parameter test_sig in procedure test_record_proc found in package compa_pkg.

Your previous issue #2493 asserted the lack of a use clause caused the INTERNAL_ERROR while preceded by the same error message:

```none
sem_selected_name(2): cannot handle IIR_KIND_EXTERNAL_SIGNAL_NAME (src/work/tb.vhd:33:9)
```
You appear to be encountering some unfinished support for external names. The INTERNAL_ERROR appears to be a consequence of continuing despite not handling the external signal name.

stfarley, 2023-09-16:
Ah Thanks, I figured they were related but the cause is significantly different so I figured two test cases were better.

tgingold, 2023-09-17:
Now support (but only with the mcode backend).

closed 2023-09-17 by tgingold