ghdl/ghdl #2495: PSL seems to be restricted to "Simple subset"

Topi-ab, 2023-09-18:
**Description**
PSL Reference Manual (Accellera Version 1.1. June 9, 2004) chapter 4.4.4 defines "Simple subset".

In this simple subset operator rules are more stringent.

E.g. left operand of -> needs to be boolean.
E.g. operand of negation operator needs to be boolean.

On the other hand, in the book "a practical introduction to PSL (Cindy Eisner, Dana Fisman)" Figure 13.1 has an example of:
assert always {req;ack;gnt} -> {data[*4] ; dataend};

The simple subset is meant to relax the implementation burden on simulators.

Formal checkers are implied to have full fledged version.

PSL line:

assert always {a;b} -> {c;d}; -- this fails the compiler.

Compiler used:

> GHDL 4.0.0-dev (v3.0.0-472-g43f5fa48c) [Dunoon edition]
>  Compiled with GNAT Version: 9.3.0
>  mcode code generator
> Written by Tristan Gingold.

Latest pull of hdlc/formal:all docker image.

**Expected behaviour**
Compiler pass without errors.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: test.vhdl
library ieee;
use ieee.std_logic_1164.all;

entity test is
	port(
		clk: in std_logic;
		a: in std_logic;
		b: in std_logic;
		c: in std_logic;
		d: in std_logic
	);
end;

architecture formal of test is
begin
	default clock is rising_edge(clk);
	
	assert always {a;b} -> {c;d};
end;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 test.vhdl

test.vhdl:18:29:error: left-hand side operand of logical '->' must be a boolean
        assert always {a;b} -> {c;d};
                            ^
```

**Context**
Please, provide the following information:

- OS: WLS
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2023-09-18:
No, it's the opposite.  It is very unlikely that non-simple subset can be formally handled by formal checkers.

Topi-ab, 2023-09-18:
How come?

Assuming not {SERE} was implemented. Then (from spec 4.4.4.)

never
Here the sepc says never operand is Boolean or Sequence. What else can it be (FL Property?)

eventually!
Here the sepc says eventually! operand is Boolean or Sequence. What else can it be (FL Property?)

{a;b} and {c;d}
if operands are SEREs, this _should_ be equivalent to
{a;b] & {c;d}
If operands are FL Properties, what is the defintion anyway?
Same with or -operator.

{a;b} -> {c;d} 
{not {a;b} | {c;d}}
are equivalent - no problem

{a;b} <-> {c;d} 
{{a;b} & {c;d}} | {not {a;b} & not {c;d}}
are equivalent - no problem

{a;b} until {c;d}
{{a;b}[*]; {c;d}}
_should_ be equivalent.

{a;b} until_ {c;d}
{{a;b}[*1 to inf]: {c;d}}
_should_ be equivalent.

before -operator is tougher to convert to SERE, also the simple subset variant.

For me it looks that the restricting / missing feature is not {SERE}. 
I do understand that negating sequential expression can be tough, but is it really that hard?

diogratia, 2023-09-19:
IEEE Std 1076-2008 1.3.5 Incorporation of Property Specification Language

>VHDL incorporates the simple subset of the Property Specification Language (PSL) as an embedded language for formal specification of the behavior of a VHDL description. PSL is defined by IEEE Std 1850-2005. All PSL constructs that appear in a VHDL description shall conform to the VHDL flavor of PSL. Within this standard, reference is made to syntactic rules of PSL. Each such reference has the italicized prefix PSL_ and corresponds to the syntax rule in IEEE Std 1850-2005 with the same name but without the prefix.

Note 'simple subset' as well as '...shall conform to the VHDL flavor of PSL'.

IEEE Std 1850-2005 4.1.2.3 VHDL flavor

>In the VHDL flavor, all expressions of the Boolean layer, as well as modeling layer code, are written in VHDL syntax (see IEEE Std 1076). The VHDL flavor also has some influence on the syntax of the temporal layer. For example, ranges of the temporal layer are specified using the VHDL-style syntax i to j.

Topi-ab, 2023-09-19:
Does this mean that it is against VHDL standard, if other than simple subset is supported?

diogratia, 2023-09-20:
[Semantics](https://www.doulos.com/knowhow/psl/semantics/)

>The semantics of PSL are declarative rather than imperative, meaning that PSL allows you to describe and reason about an abstract succession of states in such a way as to be equally applicable for generating simulation checkers, for formal proof and for documentation. Unlike some other current property languages, the definition of PSL is not tied to the idea of a checker implemented as a finite state machine executing in monotonically increasing simulation time. Indeed, it is possible to write PSL properties that cannot be implemented efficiently as simulation checkers. Those properties that *can* be implemented as efficient finite state machine automata are said to belong to the simple subset of PSL. Most practical properties fall in the *simple subset*.

IEEE Std 1850-2005  
4.4.4 Simple subset  

>PSL can express properties that cannot be easily evaluated in simulation, although such properties can be addressed by formal verification methods.  
>  
>In particular, PSL can express properties that involve branching or parallel behavior, which tend to be more difficult to evaluate in simulation, where time advances monotonically along a single path. The simple subset of PSL is a subset that conforms to the notion of monotonic advancement of time, left to right through the property, which in turn ensures that properties within the subset can be simulated easily. ...

Sure you can do just about anything as long as you have enough memory and don't care how long simulation takes. But just having the capability would be a drag on performance and without anyone else doing the same there'd be no portability and no upside.

closed 2023-09-30 by tgingold