ghdl/ghdl #2497: MCODE version of GHDL failing during OSVVM regression runs

JimLewis, 2023-09-22:
**Description**
3 days ago, all daily scheduled OSVVM regression runs were passing.   Two days ago, we now have 15 test failures under MCODE version of GHDL, but not on other versions.

Regressions are run by actions at:
https://github.com/OSVVM/OsvvmLibraries/actions

Examples of errors:

Error in TbAxi4_MemoryBurstPattern1.vhd
```
simulate TbAxi4_MemoryBurstPattern1 
ghdl --elab-run --std=08 --syn-binding --work=osvvm_TbAddressBusTransactionPkg --workdir=/home/runner/work/OsvvmLibraries/OsvvmLibraries/VHDL_LIBS/GHDL-4.0.0-dev/osvvm_tbaddressbustransactionpkg/v08 -P/home/runner/work/OsvvmLibraries/OsvvmLibraries/VHDL_LIBS/GHDL-4.0.0-dev TbAxi4_MemoryBurstPattern1
%% Log    PASSED    in Default,   Default BurstMode is ADDRESS_BUS_BURST_WORD_MODE 0 at 110 ns
%% Log    PASSED    in Default,   BurstMode Received : 0 at 110 ns
%% Log    ALWAYS    in Default,   Write with Addr = 8, 12 Words -- word aligned at 110 ns
ghdl:error: declaration of a too large object (1398340 kB > --max-stack-alloc=128 kB)
ghdl:error: simulation failed
Error: child process exited abnormally
Simulation Finish time 15:14:55, Elapsed time: 0:00:01 
SimulateError: See messages above in "simulate TbAxi4_MemoryBurstPattern1"
For tcl errorInfo, puts $::osvvm::SimulateErrorInfo
```

Error in TbAxi4_DemoMemoryReadWrite1.vhd
```
ghdl:error: bound check failure at OsvvmLibraries/AXI4/Axi4/TestCases/TbAxi4_DemoMemoryReadWrite1.vhd:146
ghdl:error: simulation failed
Error: child process exited abnormally
Simulation Finish time 15:17:01, Elapsed time: 0:00:02 
SimulateError: See messages above in "simulate TbAxi4_DemoMemoryReadWrite1"
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2023-09-23:
I will have a look.  The elaboration mechanism has changed a lot (and as a consequence external names are now supported).
I think a few bugs have been fixed after the initial change, so maybe the crashes will be fix when the docker images will be updated.

tgingold, 2023-09-24:
But in the test 1238 I cannot see the failures anymore (they are present in test 1236).
It is already fixed ?

JimLewis, 2023-09-24:
I see failures in 1236 to current.    The test case itself is not failing.   It is failing in reporting because there were test case errors.   

I don't quite have the knack for setting up the regression the way I want it to run.   I want all test cases to run, then the reporter to run (and finish - which currently it is not) and then produce passed/failed on the entire regression run - preferably in the test reporter - I thought at one point it did report like that.   It maybe some options on the reporter were updated.

tgingold, 2023-09-25:
I was looking for crashes and was not able to find them anymore.

I will have a look at errors.

tgingold, 2023-09-26:
I think I have identified the issue.

closed 2023-09-26 by tgingold

tgingold, 2023-09-27:
It is now fixed.  Thank you for your patience.

commit cfb18196 (2023-09-26): simul-vhdl_compile: handle constant with unbounded types

Fix #2497