ghdl/ghdl #2498: UNCONSTRAINT_TYPE_ELEMENT_OF_ARRAY__BUG__

fernandoka, 2023-09-25:
**Description**
When compiling the vhdl code for simulation, a uncatched exception rises. The proceess finish with errors.

**Expected behaviour**
At my understanding GHDL should not report a crash in case some constant is not fully constraine

**How to reproduce?**
It appears the cause of this crash is mainly due to an unconstrained type of an array type declaration. 

```
  constant C_A  : unsigned := to_unsigned(16#00#, 16); -- R/W
  constant C_B  : unsigned := to_unsigned(16#04#, 16); -- RO

  type regMap_tarr is array (natural range <>) of unsigned;
  constant C_REGISTERMAP : regMap_tarr := (
    0=>C_A,
    1=>C_B
  );

```

**Context**

    OS: GHDL -> Ubuntu Linux Kernel 20.04 LTS -> WSL2 -> Windows 10
    Origin:
        Compiled usign LLVM
        [3.0.0-dev]

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r151.g176b1453) [Dunoon edition]
Compiled with GNAT Version: 9.4.0
Target: x86_64-linux-gnu
/mnt/d/MK1/MAX10/GHDL_BUGS/TO_REPORT/UNCONSTRAINT_TYPE_ELEMENT_OF_ARRAY/sim/runs/i2c/
Command line:
/opt/ghdl-2.0.0-llvm/bin/ghdl1-llvm -O2 --workdir=./ghdl_i2c_.prj --std=08 --mb-comments --warn-unused -frelaxed-rules --no-vital-checks -fpsl -fexplicit -P/opt/ghdl-2.0.0-llvm/lib/ghdl/ieee/v08/ -P/opt/ghdl-2.0.0-llvm/lib/ghdl/ -fpic -c -o ./ghdl_i2c_.prj/i2c_tb.o /mnt/d/MK1/MAX10/GHDL_BUGS/TO_REPORT/UNCONSTRAINT_TYPE_ELEMENT_OF_ARRAY/sim/runs/i2c/i2c_tb.vhd
Exception STORAGE_ERROR raised
Exception information:
raised STORAGE_ERROR : stack overflow or erroneous memory access
Call stack traceback locations:
0x7fc601ac308e 0x7fc602dfed90 0x7fc602d11f38 0x7fc602d3169a 0x7fc602d413d6 0x4127e6 0x6e6a2c 0x6eb59d 0x6ebaeb 0x6a52f4 0x6a5421 0x6a5d92 0x6a6413 0x6a6f42 0x6a5f0c 0x6a88c2 0x6a897e 0x689092 0x68947b 0x68957a 0x6924e9 0x67e6ae 0x6f8a64 0x646357 0x6ff752 0x701163 0x7027ad 0x40b9a1 0x7fc601aa4081 0x40a72c 0xfffffffffffffffe
******************************************************************

```

**Additional context**
Addition of an example simulation project using Makefiles. Just move to '.\sim\runs\i2c' and type in a command line 'make' 
[UNCONSTRAINT_TYPE_ELEMENT_OF_ARRAY.zip](https://github.com/ghdl/ghdl/files/12713539/UNCONSTRAINT_TYPE_ELEMENT_OF_ARRAY.zip)

diogratia, 2023-09-26:
If you dump all the parochialism induced errors from your Makefile and file hierarchy as well as deadwood in i2c_tb.vhd to enhance readability:

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use     std.textio.all;
use     std.env.finish; -- use of vhdl-2008

entity i2c_tb is
end entity i2c_tb;

architecture str_tb of i2c_tb is

  constant C_CLK_PERIOD  : time := 20 ns; -- 50 Mhz

  constant C_A  : unsigned := to_unsigned(16#00#, 16); -- R/W
  constant C_B  : unsigned := to_unsigned(16#04#, 16); -- RO

  type regMap_tarr is array (natural range <>) of unsigned;
  constant C_REGISTERMAP : regMap_tarr := (
    0 => C_A,
    1 => C_B
  );

  file fd_log : text;
  signal s_clk          : std_logic := '1';
  signal s_rstn         : std_logic := '0';

BEGIN

clk_gen:
  s_clk <= not s_clk after C_CLK_PERIOD/2;

rst_gen:
    s_rstn <= '1' after 1 us;
stimulus: 
  process
  begin

     wait until s_rstn='1';
     write(output, LF & " ===========================================" & LF
                      & " +--      Starting  Test      --+" & LF
                      & " ===========================================" & LF & LF);

     wait until rising_edge(s_clk);  
      write(output, LF & " +--  Test has finished sucessfully!!" & LF);
      write(output, LF & " ===========================================" & LF
                       & " +--      Starting  Test      --+" & LF
                       & " ===========================================" & LF & LF);
      finish;
    end process;
END architecture str_tb;
```

and:

```none
%: ghdl -a --std=08 *.vhd
%  ghdl -e --std=08 i2c_tb
%  ghdl -r i2c_tb

 ===========================================
 +--      Starting  Test      --+
 ===========================================

 +--  Test has finished sucessfully!!

 ===========================================
 +--      Starting  Test      --+
 ===========================================

simulation finished @1020ns
```

I had trouble using your Makefile, requiring editing ../Makefile.ghdl to defined GHDL as a path to a ghdl-llvm executable (hint the nightlies don't call the executable binary ghdl-llvm). After that:

```none
%: make
if test -d ./ghdl_i2c_.prj ; then \
		rm -fr ./ghdl_i2c_.prj ; \
	fi
mkdir ./ghdl_i2c_.prj
/usr/local/bin/ghdl -i --workdir=./ghdl_i2c_.prj --std=08 --mb-comments --warn-unused -frelaxed-rules --no-vital-checks -fpsl -fexplicit /Users/dio_gratia/Desktop/UNCONSTRAINT_TYPE_ELEMENT_OF_ARRAY/sim/runs/i2c/i2c_tb.vhd /Users/dio_gratia/Desktop/UNCONSTRAINT_TYPE_ELEMENT_OF_ARRAY/sim/runs/i2c/i2c_tb.vhd
/usr/local/bin/ghdl -m -O2 --workdir=./ghdl_i2c_.prj --std=08 --mb-comments --warn-unused -frelaxed-rules --no-vital-checks -fpsl -fexplicit -o ./ghdl_i2c_.prj/exe_sim_i2c_tb.elf i2c_tb
analyze /Users/dio_gratia/Desktop/UNCONSTRAINT_TYPE_ELEMENT_OF_ARRAY/sim/runs/i2c/i2c_tb.vhd
elaborate i2c_tb
mkdir ./ghdl_i2c_.prj/obj
mv ./ghdl_i2c_.prj/*.o ./ghdl_i2c_.prj/obj
time ./ghdl_i2c_.prj/exe_sim_i2c_tb.elf -gG_LOG_FILE=./side.log --wave=./ghdl_i2c_.prj/wave_i2c_tb.ghw								# Use of 'time' to know the simulation elapsed time
./ghdl_i2c_.prj/exe_sim_i2c_tb.elf:error: cannot find in top entity generic 'g_log_file'
./ghdl_i2c_.prj/exe_sim_i2c_tb.elf:error: error during elaboration

real	0m0.096s
user	0m0.002s
sys	0m0.002s
make: *** [ghdl_run_sim] Error 1
%:  
```

reveals a different error message that appears to be due to the operation of make.

```none
%: which ghdl
/usr/local/bin/ghdl
%: ghdl --version
GHDL 4.0.0-dev (3.0.0.r529.g62e7cebd4) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 llvm 15.0.7 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
%: which make
/usr/bin/make
%: make --version
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i386-apple-darwin11.3.0
%:
```
Note the version of ghdl you are using is appears to be a bit dated, it's commit SHA fragment 176b1453 gives us a date around May 3, 2022.

tgingold, 2023-09-30:
As @diogratia said, the error cannot be reproduced; probably it has been fixed meanwhile.

closed 2023-09-30 by tgingold