ghdl/ghdl #2500: Natural generic in PSL repetion

reprographix, 2023-09-25:
**Description**
GHDL doesn't seem to support generics in PSL repetition.

```
ent.vhd:23:32:error: number expected
    --   {i_valid} |-> {[*0 to REG_COUNT]; o_valid}
                               ^
ent.vhd:23:32:error: missing ']'
    --   {i_valid} |-> {[*0 to REG_COUNT]; o_valid}
                               ^
ent.vhd:23:32:error: missing '}' after braced SERE
    --   {i_valid} |-> {[*0 to REG_COUNT]; o_valid}
                               ^
ent.vhd:23:31:error: ';' is expected instead of "reg_count"
    --   {i_valid} |-> {[*0 to REG_COUNT]; o_valid}
                              ^
ent.vhd:23:41:error: '<=' is expected instead of ']'
    --   {i_valid} |-> {[*0 to REG_COUNT]; o_valid}
                                        ^
ent.vhd:23:41:error: unexpected token ']' in a primary
    --   {i_valid} |-> {[*0 to REG_COUNT]; o_valid}
                                        ^
ent.vhd:23:41:error: ';' expected at end of signal assignment
    --   {i_valid} |-> {[*0 to REG_COUNT]; o_valid}
                                        ^
ent.vhd:23:41:error: (found: ']')
ent.vhd:23:41:error: unexpected token ']' in a concurrent statement list
    --   {i_valid} |-> {[*0 to REG_COUNT]; o_valid}
                                        ^
ent.vhd:23:51:error: '<=' is expected instead of '}'
    --   {i_valid} |-> {[*0 to REG_COUNT]; o_valid}
                                                  ^
ent.vhd:23:51:error: unexpected token '}' in a primary
    --   {i_valid} |-> {[*0 to REG_COUNT]; o_valid}
                                                  ^
ent.vhd:23:51:error: ';' expected at end of signal assignment
    --   {i_valid} |-> {[*0 to REG_COUNT]; o_valid}
                                                  ^
ent.vhd:23:51:error: (found: '}')
ent.vhd:23:51:error: unexpected token '}' in a concurrent statement list
    --   {i_valid} |-> {[*0 to REG_COUNT]; o_valid}
```

**Expected behaviour**
GHDL analyzes the expression with no error.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: ent.vhd
library ieee;
use     ieee.std_logic_1164.all;

entity ent is
    generic(
        REG_COUNT: natural
    );
    port(
        clk: in std_logic;
        rst: in std_logic;

        i_valid: in std_logic;
        o_valid: out std_logic
    );
end entity;

architecture arch of ent is
begin
    -- psl default clock is rising_edge(clk);
    -- psl assume {rst};

    -- psl f_delay_valid: assert always
    --   {i_valid} |-> {[*0 to REG_COUNT]; o_valid}
    --   sync_abort reset;
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 -fpsl ent.vhd
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS: Ubuntu 20.04 LTS
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `GHDL 4.0.0-dev (3.0.0.r529.g62e7cebd4)`

tgingold, 2023-09-25:
Ok.  I can fix the parse error, but your issue won't be fixed as currently a locally static value is required.
I hope to fix that, but not immediately.

reprographix, 2023-09-25:
I think the "number expected" error disappears when I run the non-minified reproducer (formal test with SymbiYosys, where the constant is initialized).

reprographix, 2023-09-25:
I can try a fix for the parser if you don't mind, I'll let you know if I struggle too much ;-)

tgingold, 2023-09-25:
Fine with me!

reprographix, 2023-09-25:
Ok I think I've got something, but I'm not sure I've fixed it the right way. I invite you to have a look at it, and tell me if I did it wrong.

tgingold, 2023-09-27:
At the end, I added parsing any expression.

closed 2023-09-27 by tgingold