ghdl/ghdl #2502: [Regression bug?] CONSTRAINT_ERROR

stnolting, 2023-09-25:
Hey there!

I am using the latest GHDL version via GitHub actions (`[ghdl/setup-ghdl-ci@nightl](uses: ghdl/setup-ghdl-ci@nightly)y`). Since last week (5 days ago?) GHDL keep crashing:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r540.g034c44c98) [Dunoon edition]
Compiled with GNAT Version: 9.4.0
Target: x86_64-linux-gnu
/home/runner/work/neorv32-riscof/neorv32-riscof/sim/
Command line:
/home/runner/work/_temp/ghdl/bin/ghdl -r --std=08 --work=neorv32 neorv32_riscof_tb -gMEM_FILE=./main.hex --max-stack-alloc=0 --ieee-asserts=disable --assert-level=error -gMEM_SIZE=52056 -gRISCV_B=true --stop-time=4ms
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : synth-vhdl_stmts.adb:3051 access check failed
******************************************************************
```

VHDL design sources: https://github.com/stnolting/neorv32/tree/c7590c9cc86b1646926f57d0ab0ab610e0471af8/rtl/core
Testbench: https://github.com/stnolting/neorv32-riscof/blob/main/sim/neorv32_riscof_tb.vhd

If you need further information I'll be happy to provide it :wink:

And by the way, thank you so much for GHDL and all you work!

tgingold, 2023-09-25:
The elaboration mechanism has changed.  I will investigate.

closed 2023-09-26 by tgingold

stnolting, 2023-09-27:
Seems like everything is fine again (at least on my side) :wink:
Thanks for fixing this so fast!

tgingold, 2023-09-27:
You're welcome!

commit e90faa8e (2023-09-26): simul-vhdl_elab: allow function calls in associations

Fix #2502