ghdl/ghdl #2504: GHDL synthesis crashes on multiple instances in GRLIB

JJ-Gaisler, 2023-09-28:
**Description**
We have recently been trying to get GRLIB (NOEL-V) through GHDL synthesis, we have currently gotten the code to synthesize by excluding certain design files and re-writing certain constructs. One thing that has been especially error prone has been indexing of different sorts. The analysis + elaborate steps goes fine but synthesis  crashes.

**Expected behaviour**
GHDL wouldn't crash

**How to reproduce?**
Since there are multiple instances of crashes the easiest way to reproduce the issue is to:

1.  Download [grlib](https://gaisler.com/products/grlib/grlib-gpl-2023.2-b4283.tar.gz)
3. Edit the file lib/gaisler/noelv/pkg/noelv.vhd and set AIA_SUPPORT to 1
4. cd to grlib-gpl-2023.2-b4283/designs/noelv-generic
5. make ghdl
6. Run the following synthesis command

```sh :image: ghdl/ghdl:bullseye-mcode
 ghdl --synth --out=verilog -fexplicit --ieee=synopsys --mb-comments
 --warn-no-binding --workdir=gnu/work --work=work
 -Pgnu/grlib -Pgnu/synplify -Pgnu/techmap -Pgnu/wizl -Pgnu/eth
 -Pgnu/opencores -Pgnu/gaisler -Pgnu/work
 work.noelvmp > noelv.v
```

- A crash occurs in aclint_ahb.vhd at line 222, changing the line as follows avoids the crash. A comparison between std_logic_vectors is perhaps not a good idea but an error should be thrown if this is an issue.

`if (r.mtime >= r.mtimecmp(i)) then ` to 
`if (unsigned(r.mtime) >= unsigned(r.mtimecmp(i))) then`

- The next file which causes a crash is ahbctrl.vhd
I haven't been able to find a workaround for this file, excluding it from the build to continue.

Is there a way to see which file GHDL is currently working on synthesizing to better be able to trace where the crash occurs? Backtraces in gdb for example? If there is an easier way to debug then I could probably help in trying to come up with rewrites which work (I guess these can be nice to have to understand why synthesis crashes?). In that case I can continue to update this issue.

**Context**
Please, provide the following information:

- OS: Linux OpenSuse 15.4
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `58d9ec09d082ea5ab4e059c72b53213d0b9f6ef2`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r381.g58d9ec09d) [Dunoon edition]
Compiled with GNAT Version: Community 2021 (20210519-103)
Target: x86_64-pc-linux-gnu
/home/jonathan/Downloads/grlib-gpl-2023.2-b4283/designs/noelv-generic/
Command line:
ghdl --synth --out=verilog -fexplicit --ieee=synopsys --mb-comments --warn-no-binding --workdir=gnu/work --work=work -Pgnu -Pgnu/grlib -Pgnu/synplify -Pgnu/techmap -Pgnu/wizl -Pgnu/eth -Pgnu/opencores -Pgnu/gaisler -Pgnu/work work.noelvmp
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : netlists-expands.adb:167
Call stack traceback locations:
0x53aa62 0x53acee 0x53ba2b 0x5afbb7 0x5da87e 0x5afa90 0x6511ba 0x62de09 0x409f44 0x7f74e881d24b 0x409f98 0xfffffffffffffffe
```

**Additional context**
```
#0  <__gnat_debug_raise_exception> (e=0xa1f520 <types.internal_error>, message=...) at s-excdeb.adb:43
#1  0x000000000065edf9 in ada.exceptions.complete_occurrence (x=x@entry=0x116ce90) at a-except.adb:929
#2  0x000000000065ee09 in ada.exceptions.complete_and_propagate_occurrence (x=x@entry=0x116ce90) at a-except.adb:940
#3  0x000000000065ee5c in <__gnat_raise_exception> (e=0xa1f520 <types.internal_error>, message=...) at a-except.adb:977
#4  0x000000000053aa64 in netlists.expands.extract_address (ctxt=ctxt@entry=0x114e090, addr_net=addr_net@entry=7080, ndims=ndims@entry=0) at /home/jonathan/Work/ghdl/gcc-12.3.0/gcc/vhdl/netlists-expands.adb:167
#5  0x000000000053acf0 in netlists.expands.expand_dyn_insert (ctxt=ctxt@entry=0x114e090, inst=inst@entry=5707, en=en@entry=0) at /home/jonathan/Work/ghdl/gcc-12.3.0/gcc/vhdl/netlists-expands.adb:446
#6  0x000000000053ba2d in netlists.expands.expand_gates (ctxt=ctxt@entry=0x114e090, m=m@entry=129) at /home/jonathan/Work/ghdl/gcc-12.3.0/gcc/vhdl/netlists-expands.adb:537
#7  0x00000000005afbb9 in synthesis.instance_passes (ctxt=0x114e090, m=m@entry=129) at /home/jonathan/Work/ghdl/gcc-12.3.0/gcc/vhdl/synthesis.adb:112
#8  0x00000000005da880 in synth.vhdl_insts.synth_instance (inst=...) at /home/jonathan/Work/ghdl/gcc-12.3.0/gcc/vhdl/synth-vhdl_insts.adb:1790
#9  synth.vhdl_insts.synth_all_instances () at /home/jonathan/Work/ghdl/gcc-12.3.0/gcc/vhdl/synth-vhdl_insts.adb:1802
#10 0x00000000005afa92 in synthesis.synth_design (design=design@entry=58340, inst=inst@entry=0xf1f3a0, encoding=<optimized out>) at /home/jonathan/Work/ghdl/gcc-12.3.0/gcc/vhdl/synthesis.adb:77
#11 0x00000000006511bc in ghdlsynth.perform_action (cmd=..., args=...) at /home/jonathan/Work/ghdl/gcc-12.3.0/gcc/vhdl/ghdldrv/ghdlsynth.adb:595
#12 0x000000000062de0b in ghdlmain.main () at /home/jonathan/Work/ghdl/gcc-12.3.0/gcc/vhdl/ghdldrv/ghdlmain.adb:628
#13 0x0000000000655b07 in ghdl_gcc () at /home/jonathan/Work/ghdl/gcc-12.3.0/gcc/vhdl/ghdldrv/ghdl_gcc.adb:43
#14 0x0000000000409f46 in main (argc=<optimized out>, argv=<optimized out>, envp=<optimized out>) at /home/jonathan/Work/ghdl/build/gcc-objs/gcc/b~ghdl_gcc.adb:972
#15 0x00007ffff7c9524d in __libc_start_main () from /lib64/libc.so.6
#16 0x0000000000409f9a in _start () at ../sysdeps/x86_64/start.S:120

```

tgingold, 2023-09-29:
I will have a look.

It is not that simple to display which file is being synthesized.  There are a first passes.
From the debugger, you can call routines from netlists-dump, like: `call debug_net(n)`.

tgingold, 2023-09-30:
I think this is now fixed.
ghdl didn't handle '2**x' and this is now supported.
Due to this error, it crashed after that.  It's also fixed.

Do not hesitate to report any issue with synthesis!

JJ-Gaisler, 2023-10-02:
Thanks for your reply.
I rebuilt GHDL to ```46e0978100e2eb71d50648a73566468337150a1a``` and repeated the aforementioned process. I still experience the same crashes, i.e. crashing on aclint_ahb:
`if (r.mtime >= r.mtimecmp(i))`

tgingold, 2023-10-02:
Yes, sorry.  I forgot to remove a debug flag.

tgingold, 2023-10-03:
I think this is now OK.
Thank you for your patience!

tgingold, 2023-10-03:
(There were issues with memories with only 1 element).

closed 2023-10-03 by tgingold

JJ-Gaisler, 2023-10-03:
Thank you for the update!
This seems to have solved a bunch of the issues we were seeing.

tgingold, 2023-10-03:
Great!