ghdl/ghdl #2505: External names in wait statements are not supported

yikkrubz, 2023-10-02:
**Description**
It seems external names in wait statements are not supported.
Analyzing and elaborating works. The program crashes with a sigabort when executed.
**Expected behaviour**
It should not crash.
**How to reproduce?**
top_tb.vhd:

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity top is
generic (clk_period : time);
end top;

architecture str of top is
signal rst : std_logic := '1';
begin

RELEASE_RESET : process
begin
wait for clk_period * 10;
rst <= '0';
wait for clk_period * 1;
wait;
end process;

end architecture;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use std.env.finish;

entity top_tb is
end top_tb;

architecture sim of top_tb is
constant clk_hz : integer := 100e6;
constant clk_period : time := 1 sec / clk_hz;
begin

DUT : entity work.top(str)
generic map(clk_period => clk_period);

SEQUENCER_PROC : process
begin
wait until << signal dut.rst : std_logic >>= '0';
finish;
end process;

end architecture;

ghdl -a --std=08 top_tb.vhd
ghdl -e --std=08 top_tb
ghdl -r top_tb 
make: *** [Makefile:18: run] Error 255

**Context**
GHDL 4.0.0-dev (3.0.0.r576.g46e097810) [Dunoon edition]
 Compiled with GNAT Version: 13.1.0
 llvm 16.0.6 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

- OS: macos 12.7
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `g46e097810`

tgingold, 2023-10-03:
This one is ok but you must use ghdl with the mcode backend.
External names are not yet supported with the llvm (or gcc) backend.

closed 2023-10-03 by tgingold

yikkrubz, 2023-10-03:
Thank you very much for your reply. I was not aware of that

commit 452c60e8 (2023-10-03): testsuite/gna: add a test

close #2505