ghdl/ghdl #2506: Alias of an external name causes a ghdl bug

yikkrubz, 2023-10-02:
**Description**
Analyzing the file B_pkg.vhd with a current version of ghdl raises a ghdl bug. The example works as expected when the code is compiled by Questa.
**Expected behaviour**

**How to reproduce?**
ghdl -a --std=08 A.vhd 
ghdl -a --std=08 B_pkg.vhd

**Context**
- OS:macos 12.7
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `g46e097810`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r576.g46e097810) [Dunoon edition]
Compiled with GNAT Version: 13.1.0
Target: x86_64-apple-darwin21
/Users/Peter/Downloads/tmp/hurz/external_name/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -o B_pkg.o B_pkg.vhd
Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
Exception information:
raised ADA.ASSERTIONS.ASSERTION_ERROR : vhdl-nodes.adb:862
Load address: 0x104ef9000
Call stack traceback locations:
0x106b862c1 0x105090c06 0x105274e7b 0x105275a34 0x10527647f 0x10527da38 0x10526e80f 0x1052509af 0x1052788f9 0x10527d368 0x10526e73d 0x105264007 0x1052641d9 0x1052647e3 0x10526d99f 0x10526e854 0x10526fc2c 0x105282ff5 0x105283037 0x105284c37 0x105284d1d 0x10524d0a0 0x10525eec8 0x10525f127 0x10524f0e5 0x1052501f3 0x105222b91 0x10534c69d 0x10534e130 0x10534f33f 0x104eff1bd
******************************************************************
ghdl:error: compilation error

**Additional context**
[external_name.zip](https://github.com/ghdl/ghdl/files/12782527/external_name.zip)

tgingold, 2023-10-03:
I can reproduce the crash.  I am working on a fix.

tgingold, 2023-10-03:
I am not sure the code is correct.  You need to use an absolute pathname.

yikkrubz, 2023-10-03:
I doublechecked: The example works as expected when the code is compiled by Questa. What is wrong?

tgingold, 2023-10-04:
It doesn't follow LRM2008 8.7 a 3:

3)For a relative pathname, the innermost concurrent region is initially 
identified, where a concurrent region is defined recursively to be
—A block declarative region (including an external block and any block 
equivalent to a generate statement), or
—A package declarative region (including a generic-mapped package 
equivalent to a package instantiation) declared immediately within a 
concurrent region.

closed 2023-10-04 by tgingold