ghdl/ghdl #2507: to_signed on negative integers causes vector truncated warning

avelure, 2023-10-03:
**Description**
a `to_signed` conversion on a negative literal or constant\variable integer value causes a `vector truncated` warning.
This seems to be a regression, as it was working in 3.0.0.r448.g2245a7fd1

**Expected behaviour**
No warning as long as the converted value can fit in the vector.

**How to reproduce?**

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity test;

architecture beh of test is
begin
  
  process
  begin
    assert signed'("11") = to_signed(-1, 2) report "to_signed" severity failure;
    wait;
  end process;

end architecture beh;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
```

```
$ghdl -a --std=08 test_signed_trunc.vhd
test_signed_trunc.vhd:13:28:warning: NUMERIC_STD.TO_SIGNED: vector truncated
    assert signed'("11") = to_signed(-1, 2) report "to_signed" severity failure;
                           ^
```
**Context**
- OS: ubuntu 22.04\win10
- Origin: nightly LLVM 3.0.0.r576.g46e097810

jklockars, 2023-10-03:
I was just about to mention this exact problem...

Here's an example fix (not very well tested, and it's been many years since I last programmed Ada):

diff --git a/src/synth/synth-vhdl_eval.adb b/src/synth/synth-vhdl_eval.adb
index 27557ae35..fafe0463f 100644
--- a/src/synth/synth-vhdl_eval.adb
+++ b/src/synth/synth-vhdl_eval.adb
@@ -15,7 +15,7 @@
 --
 --  You should have received a copy of the GNU General Public License
 --  along with this program.  If not, see <gnu.org/licenses>.
-
+with Ada.Text_IO;
 with Types; use Types;
 with Types_Utils; use Types_Utils;
 with Flags;
@@ -617,6 +617,7 @@ package body Synth.Vhdl_Eval is
       El_Type : constant Type_Acc := Get_Array_Element (Res_Type);
       Res : Memtyp;
       Bnd : Type_Acc;
+      Neg : Boolean := (Arg and (2 ** (Uns64'Size - 1))) /= 0;
       B : Uns64;
       D : Uns64;
    begin
@@ -630,10 +631,11 @@ package body Synth.Vhdl_Eval is
                           Std_Ulogic'Val (Std_Logic_0_Pos + D));
          B := Shift_Right_Arithmetic (B, 1);
       end loop;
-      if (Arg >= 0 and (B /= 0 or D /= 0))
-        or else (Arg < 0 and (B /= -1 or D /= 1))
+      if (not Neg and (B /= 0 or D /= 0))
+        or else (Neg and (B + 1 /= 0 or D /= 1))
       then
          Warning_Msg_Synth (+Loc, "NUMERIC_STD.TO_SIGNED: vector truncated");
       end if;
       return Res;
    end Eval_Signed_To_Log_Vector;

tgingold, 2023-10-03:
I am testing the fix.  Thank you @jklockars for it!

commit 93899a66 (2023-10-03): synth-vhdl_elab: fix warning in to_signed

Change the condition to check if Arg is negative
(as Arg is unsigned).

Fix #2507
closed 2023-10-03 by tgingold