ghdl/ghdl #2508: External Signal Alias Crashes with Mcode GHDL build

stfarley, 2023-10-04:
Attempting to use an external alias causes GHDL to Crash. This works fine with LLVM or GCC builds.
Likely related to 2506

./run_tc 

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r592.g522dc4317.dirty) [Dunoon edition]
Compiled with GNAT Version: Community 2020 (20200818-93)
Target: x86_64-pc-linux-gnu
/projects/ghdl_issues/new/
Command line:
ghdl --elab-run --std=08 -frelaxed-rules --work=work tb --stop-time=200ns --wave=gsim.ghw
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : elab-vhdl_context.adb:674 access check failed
******************************************************************
[tc.zip](https://github.com/ghdl/ghdl/files/12803291/tc.zip)

tgingold, 2023-10-04:
Thank you for the issue, now fixed.

closed 2023-10-04 by tgingold

stfarley, 2023-10-05:
Thanks, The fix works perfectly here as well.

commit 6f0a6d99 (2023-10-04): elab-vhdl_decls: handle top-level external names

Fix #2508