ghdl/ghdl #2509: GHDL crash while trying to simulate MEGA65

gardners, 2023-10-05:
**Description**

GHDL crashes.

**Expected behaviour**
GHDL doesn't crash, but continues running normally.

**How to reproduce?**

Attempt to simulate the provided commit ID of the  MEGA65.  As the error doesn't give more information, its a bit hard for me to narrow it down.  

I'm on Ubuntu 23.04, running GHDL 4.0.0 built yesterday from source commit 522dc4317646b4c2be81360086bc1297c23d53b9.

```
git clone https://github.com/MEGA65/mega65-core.git
git checkout a140586e819d7789bfcac3271d5b0e84d0f1274e
git submodule init
git submodule update
make simulate
```

If it's easier, I'm happy to run GHDL under a debugger of some sort, if that helps reveal useful information and saves you having to try to build the fairly complex MEGA65 target again.

**Context**
Please, provide the following information:

- OS: Ubuntu 23.04
- Origin:
  - [ ] Built from sources: 522dc4317646b4c2be81360086bc1297c23d53b9

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: 11.4.0
Target: x86_64-linux-gnu
/home/paul/Projects/mega65/mega65-core/
Command line:
ghdl/ghdl_mcode -r cpu_test --assert-level=warning
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-nodes.adb:862
******************************************************************
```

**Additional context**

N/A

gardners, 2023-10-05:
If it's any clue, here is what it outputs just before:

```
src/vhdl/viciv.vhdl:322:81:warning: cannot use signal value during elaboration [-Welaboration]
  signal x_chargen_start : unsigned(13 downto 0) := to_unsigned(safe_to_integer(frame_h_front),14);
                                                                                ^
src/vhdl/sprite.vhdl:150:61:warning: cannot use signal value during elaboration [-Welaboration]
    variable sprite_number_mod_4 : integer range 0 to 7 := (sprite_number mod 4) * 2;
                                                            ^
src/vhdl/sprite.vhdl:150:61:warning: cannot use signal value during elaboration [-Welaboration]
    variable sprite_number_mod_4 : integer range 0 to 7 := (sprite_number mod 4) * 2;
                                                            ^
src/vhdl/sprite.vhdl:150:61:warning: cannot use signal value during elaboration [-Welaboration]
    variable sprite_number_mod_4 : integer range 0 to 7 := (sprite_number mod 4) * 2;
                                                            ^
src/vhdl/sprite.vhdl:150:61:warning: cannot use signal value during elaboration [-Welaboration]
    variable sprite_number_mod_4 : integer range 0 to 7 := (sprite_number mod 4) * 2;
                                                            ^
src/vhdl/sprite.vhdl:150:61:warning: cannot use signal value during elaboration [-Welaboration]
    variable sprite_number_mod_4 : integer range 0 to 7 := (sprite_number mod 4) * 2;
                                                            ^
src/vhdl/sprite.vhdl:150:61:warning: cannot use signal value during elaboration [-Welaboration]
    variable sprite_number_mod_4 : integer range 0 to 7 := (sprite_number mod 4) * 2;
                                                            ^
src/vhdl/sprite.vhdl:150:61:warning: cannot use signal value during elaboration [-Welaboration]
    variable sprite_number_mod_4 : integer range 0 to 7 := (sprite_number mod 4) * 2;
                                                            ^
src/vhdl/sprite.vhdl:150:61:warning: cannot use signal value during elaboration [-Welaboration]
    variable sprite_number_mod_4 : integer range 0 to 7 := (sprite_number mod 4) * 2;
                                                            ^
```

gardners, 2023-10-05:
Fixed those warnings, but the crash still happens.

tgingold, 2023-10-05:
Are there missing files ?  I get:
```
$ make USE_LOCAL_GHDL=1 simulate
make: *** No rule to make target 'src/vhdl/s27kl0641-pgs-modified.vhd', needed by 'simulate'.  Stop.
```

tgingold, 2023-10-06:
My personal email is tgingold free fr

tgingold, 2023-10-08:
I think I need an archive with the sources and a simple to reproduce the crash.
When I do: `make USE_LOCAL_GHDL=1 GHDL=$GHDL simulate PATH=$PWD/bin:$PATH`, I got warnings and errors like:
```
src/vhdl/internal1541.vhdl:54:14:warning: (in default configuration of internal1541(romanesque_revival))
src/vhdl/machine.vhdl:1295:23:error: range of formal "xcounter_out" is different from formal range
      xcounter_out => xcounter_viciv,
                      ^
src/vhdl/machine.vhdl:1296:23:error: range of formal "ycounter_out" is different from formal range
      ycounter_out => ycounter_viciv,
                      ^
```

But` how to be sure to reproduce exactly what you did ?

gardners, 2023-10-09:
Email sent. Feel free to delete above comment with email details.

Let me see what I can cook up in terms of source archive.

gardners, 2023-10-14:
You should have the tarball by email now. Let me know if you need anything else.

commit 1ec1eb6d (2023-10-15): iterate_top_level: continue after un-elaborated package

Fix #2509
closed 2023-10-15 by tgingold