ghdl/ghdl #2510: GHDL Crashes when using UVVM AXILITE BFM (MCODE only)

stfarley, 2023-10-05:
**Description**
Attempting to use the AXILITE BFM with an MCODE GHDL Build results in the following Error
GCC and LLVM builds work as expected

**Expected behaviour**
The TC should just run and do basically nothing

**How to reproduce?**
./comp_run

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r592.g522dc4317.dirty) [Dunoon edition]
Compiled with GNAT Version: Community 2020 (20200818-93)
Target: x86_64-pc-linux-gnu
/projects/ghdl_issues/new/
Command line:
ghdl --elab-run --std=08 -frelaxed-rules --work=work tb --stop-time=200ns --wave=gsim.ghw
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : elab-vhdl_context.adb:674 access check failed
[tc.zip](https://github.com/ghdl/ghdl/files/12817150/tc.zip)

tgingold, 2023-10-05:
The `comp_run` file is missing

stfarley, 2023-10-05:
[comp_run.txt](https://github.com/ghdl/ghdl/files/12820839/comp_run.txt)

Sorry about that. I thought I had added it to the zip. (added the .txt ext so I could upload it.

tgingold, 2023-10-05:
I think I also need the reference to uvvm.  I don't know which version you are using.

stfarley, 2023-10-05:
I am using the current version from the "master" branch. the latest tag is v2023.03.21
git describe
v1.5.1-215-g6953b15

tgingold, 2023-10-06:
I can reproduce the crash.

tgingold, 2023-10-08:
I will fix the crash, but it is due to issues in osvvm:
in file vvc_methods_pkg.vhd line 133, a shared variable is declared using a protected type from a package instantiated just before.
This is not allowed: the body of the package and thus the body of the protected type are elaborated within the body of the vvc_methods_pkg and thus after the shared variable.  So the shared variable cannot be built.

There is a warning for this mistake, but it is disabled in osvvm.

stfarley, 2023-10-17:
With this TC and Code build I get a different error now. (LLVM and GCC works properly)
The UVVM libraries are compiled with the GHDL scripts.

/projects/uvvm/UVVM_ALL/bitvis_vip_scoreboard/src/generic_sb_pkg.vhd:546:14:error: type of variable "vr_sb_queue" referenced before its elaboration
    variable vr_sb_queue         : sb_queue_pkg.t_generic_queue;

GHDL 4.0.0-dev (3.0.0.r634.g649af4117.dirty) [Dunoon edition]
 Compiled with GNAT Version: Community 2020 (20200818-93)
 static elaboration, mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

tgingold, 2023-10-18:
Unfortunately, this is a real bug in uvvm.  Is the error message clear 
enough ?

I might try to add it to -frelax, but I am not sure how far it will be 
supported.

tgingold, 2023-10-25:
Yes, please it should be reported to the UVVM team.  Please add a reference to this issue.

stfarley, 2024-03-14:
This is fixed in UVVM 2024.03.08

Paebbels, 2024-12-19:
@stfarley are there any further actions for GHDL in the issue?

stfarley, 2025-03-09:
Hi Patrick,Sorry for the late reply in this.No further action is required.Thanks,Steve

closed 2025-03-09 by Paebbels