ghdl/ghdl #2511: GRLIB synthesis crash

JJ-Gaisler, 2023-10-06:
**Description**
GHDL synth crashes in recursive function for counting leading zeros.

**How to reproduce?**

1. Download [grlib](https://gaisler.com/products/grlib/grlib-gpl-2023.2-b4283.tar.gz)
2.  Edit the file lib/gaisler/noelv/core/alunv.vhd (Code example to follow)
3.  cd to grlib-gpl-2023.2-b4283/designs/noelv-generic
4.  make ghdl
5.   Run the following synthesis command

```
 ghdl --synth --out=verilog -fexplicit --ieee=synopsys --mb-comments \
 --warn-no-binding --workdir=gnu/work --work=work \
 -Pgnu/grlib -Pgnu/synplify -Pgnu/techmap -Pgnu/wizl -Pgnu/eth \
 -Pgnu/opencores -Pgnu/gaisler -Pgnu/work \
 work.noelvmp
 ```

In alunv.vhd there is a function for clz which crashes depending on if cnt_top is declared constant or not.
If not declared constant an error is produced and then synthesis crashes with a discriminant error. Taking the clz function from alunv and putting it in a separate entity doesn't lead to a crash but still yields the error. In the code snippet a few version of what causes the issue and how it can be circumvented is given.
```
  function clz(op_in : std_logic_vector) return unsigned is
    variable op      : std_logic_vector(op_in'length - 1 downto 0) := op_in;
    
    variable cnt_top : integer := log2(op'length);
    -- constant cnt_top : integer := log2(op'length);
    variable cnt     : unsigned(cnt_top downto 0);
    
    -- Using the expression of cnt_top doesn't cause a crash.
    -- variable cnt     : unsigned(log2(op'length) downto 0);

  begin
    if op'length = 1 then
      cnt(0) := not op(0);
    else
      if not all_0(hi_h(op)) then
        cnt := "0" & clz(hi_h(op));
      else
        cnt := uaddx(clz(lo_h(op)), u2vec(op'length / 2, cnt_top));
      end if;
    end if;

    return cnt;
  end;
```
The second issue we are seeing seems to be dependent on surrounding code, when running synthesis on the development version of grlib we get an invalid data crash (see below). I haven't been able to reproduce this issue with the GPL version of the code. I tried breaking out the code which crashes into a standalone entity but this wasn't enough to cause a crash. The crashing code is the following:

```   
function clz_orig(op_in : std_logic_vector) return unsigned is
     variable op      : std_logic_vector(op_in'length - 1 downto 0) := op_in;
     -- Changing cnt_top to constant or using the cnt_top expression directly doesn't lead to a crash
     variable cnt_top : integer := log2(op'length);
     variable cnt     : unsigned(log2(op'length) downto 0);
   begin
     if op'length = 1 then
       cnt(0) := not op(0);
     else
       if not all_0(hi_h(op)) then
         cnt := "0" & clz_orig(hi_h(op));
       else
         cnt := uaddx(clz_orig(lo_h(op)), u2vec(op'length / 2, cnt_top));
       end if;
     end if;

     return cnt;
   end;
```

**Context**
[... 239 lines cut ...]
#154 0x00000000005aa5a0 in synth.vhdl_decls.synth_declarations (syn_inst=0xa823b0, decls=<optimized out>, is_subprg=is_subprg@entry=true) at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_decls.adb:850
#155 0x00000000005ec63a in synth.vhdl_stmts.synth_dynamic_subprogram_call (init=..., call=11019184, sub_inst=0xa823b0, syn_inst=0xabc010) at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:2838
#156 synth.vhdl_stmts.synth_subprogram_call (syn_inst=syn_inst@entry=0xabc010, call=call@entry=228314, init=...) at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:3102
#157 0x00000000005ecc38 in synth.vhdl_stmts.synth_subprogram_call (syn_inst=syn_inst@entry=0xabc010, call=call@entry=228314) at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:3149
#158 0x00000000005ecde3 in synth.vhdl_stmts.synth_procedure_call (syn_inst=0xabc010, stmt=stmt@entry=228313) at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:3207
#159 0x00000000005e8f9f in synth.vhdl_stmts.synth_sequential_statement (c=..., stmt=228313) at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:4127
#160 0x00000000005ed08b in synth.vhdl_stmts.synth_sequential_statements (stmts=<optimized out>, c=...) at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:4189
#161 synth.vhdl_stmts.synth_if_statement (c=..., stmt=stmt@entry=228275) at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:1113
#162 0x00000000005e8e07 in synth.vhdl_stmts.synth_sequential_statement (c=..., stmt=228275) at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:4088
#163 0x00000000005e912b in synth.vhdl_stmts.synth_sequential_statements (stmts=228275, c=...) at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:4189
#164 synth.vhdl_stmts.synth_dynamic_for_loop_statement (stmt=228271, c=...) at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:3633
#165 synth.vhdl_stmts.synth_sequential_statement (c=..., stmt=228271) at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:4111
#166 0x00000000005edc40 in synth.vhdl_stmts.synth_sequential_statements (stmts=<optimized out>, c=...) at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:4189
#167 synth.vhdl_stmts.synth_process_statement (proc=226672, syn_inst=0x0) at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:4364
#168 synth.vhdl_stmts.synth_concurrent_statement (syn_inst=syn_inst@entry=0xf7ede0, stmt=stmt@entry=226672) at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:4887
#169 0x00000000005ee0ba in synth.vhdl_stmts.synth_concurrent_statements (syn_inst=syn_inst@entry=0xf7ede0, stmts=<optimized out>) at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:4953
#170 0x00000000005d66b2 in synth.vhdl_insts.synth_instance_design (syn_inst=syn_inst@entry=0xf7ede0, entity=entity@entry=114978, arch=arch@entry=188158) at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_insts.adb:1702
#171 0x00000000005da683 in synth.vhdl_insts.synth_instance (inst=...) at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_insts.adb:1787
#172 synth.vhdl_insts.synth_all_instances () at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_insts.adb:1805
#173 0x00000000005b1582 in synthesis.synth_design (design=design@entry=50678, inst=inst@entry=0xef4170, encoding=<optimized out>) at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/synthesis.adb:77
#174 0x00000000006553dc in ghdlsynth.perform_action (cmd=..., args=...) at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/ghdldrv/ghdlsynth.adb:595
#175 0x0000000000631f6b in ghdlmain.main () at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/ghdldrv/ghdlmain.adb:672
#176 0x0000000000659d27 in ghdl_gcc () at /home/jonathan/Work/ghdl/ghdl_sources/gcc-12.2.0/gcc/vhdl/ghdldrv/ghdl_gcc.adb:43
#177 0x000000000040a006 in main (argc=<optimized out>, argv=<optimized out>, envp=<optimized out>) at /home/jonathan/Work/ghdl/ghdl_sources/build/gcc-objs/gcc/b~ghdl_gcc.adb:974
```

tgingold, 2023-10-06:
I can reproduce the crash.

I am curious: why are you using variables in `clz` if you don't assign them ?

closed 2023-10-07 by tgingold

tgingold, 2023-10-08:
I think this is now fixed.  Thank you for the issue.

JJ-Gaisler, 2023-10-09:
> I can reproduce the crash.
> 
> I am curious: why are you using variables in `clz` if you don't assign them ?

Thanks for the update!

Good question, this all comes down to other EDA tools. Some ASIC tools only accept variables where constants would be the logical choice...

commit 2acd1236 (2023-10-07): synth_discrete_range_expression: allow defined values

Use Is_Static_Val to allow defined values for bounds.

Fix #2511