ghdl/ghdl #2515: Cant solve the error: Bound check failure: C://documents/uvvm-master/bitvis_vip_axistream/tb/mytbwithaxistreamvvc

gabrielzb19, 2023-10-10:
To: Any person familiar with UVVM or GHDL bugs.

I'm dealing with the problem stated in the title (Image attached below), it is just happening when using the axistream VVC functions, specifically the code stops at the end of the reset signal, the code is the following one:

### **Simulation with error message:**
[image]

### **Code:**
` library IEEE;
use     IEEE.std_logic_1164.all;
use     IEEE.numeric_std.all;
use     IEEE.math_real.all;
library uvvm_util;
context uvvm_util.uvvm_util_context;
library uvvm_vvc_framework;
use     uvvm_vvc_framework.ti_vvc_framework_support_pkg.all;
library bitvis_vip_axistream;
context bitvis_vip_axistream.vvc_context;

entity axis_vvc_supersimple is
end;

architecture bench of axis_vvc_supersimple is

  component axis_fifo
    generic(
      constant GC_DATA_WIDTH   : natural := 8;
      constant GC_USER_WIDTH   : natural := 1;
      constant GC_FIFO_DEPTH   : natural := 256
    );
    port(
      clk                      : in  std_logic;
      rst                      : in  std_logic;
      s_axis_tready            : out std_logic;
      s_axis_tvalid            : in  std_logic;
      s_axis_tdata             : in  std_logic_vector(GC_DATA_WIDTH - 1 downto 0);
      s_axis_tuser             : in  std_logic_vector(GC_USER_WIDTH - 1 downto 0);
      s_axis_tkeep             : in  std_logic_vector(GC_DATA_WIDTH / 8 - 1 downto 0);
      s_axis_tlast             : in  std_logic;
      m_axis_tready            : in  std_logic;
      m_axis_tvalid            : out std_logic;
      m_axis_tdata             : out std_logic_vector(GC_DATA_WIDTH - 1 downto 0);
      m_axis_tuser             : out std_logic_vector(GC_USER_WIDTH - 1 downto 0);
      m_axis_tkeep             : out std_logic_vector(GC_DATA_WIDTH / 8 - 1 downto 0);
      m_axis_tlast             : out std_logic;
      empty                    : out std_logic := '1'
      --full          : out STD_LOGIC := '0'
    );
  end component;
  constant c_max_bytes         : natural   := 100;
  constant C_CLK_PERIOD        : time      :=10 ns;
  signal   clock_ena           : boolean   :=false;
  signal clk                   : STD_LOGIC;
  signal rst                   : STD_LOGIC :='1';
  constant GC_VVC_IS_MASTER    : boolean   := true;
  constant GC_INSTANCE_IDX     : natural   := 1;
  constant GC_DATA_WIDTH       : natural   := 8;
  constant GC_USER_WIDTH       : natural   := 1;
  constant C_S_AXIS_TDEST_WIDTH: natural   := 1;
  constant C_S_AXIS_TID_WIDTH  : natural   :=1;
  constant GC_FIFO_DEPTH       : natural   := 256;
  signal s_axis_tready         : STD_LOGIC;
  signal s_axis_tvalid         : STD_LOGIC;
  signal s_axis_tdata          : STD_LOGIC_VECTOR(GC_DATA_WIDTH - 1 downto 0);
  signal s_axis_tuser          : STD_LOGIC_VECTOR(GC_USER_WIDTH - 1 downto 0);
  signal s_axis_tkeep          : STD_LOGIC_VECTOR(GC_DATA_WIDTH / 8 - 1 downto 0);
  signal s_axis_tlast          : STD_LOGIC;
  signal m_axis_tready         : STD_LOGIC;
  signal m_axis_tvalid         : STD_LOGIC;
  signal m_axis_tdata          : STD_LOGIC_VECTOR(GC_DATA_WIDTH - 1 downto 0);
  signal m_axis_tuser          : STD_LOGIC_VECTOR(GC_USER_WIDTH - 1 downto 0);
  signal m_axis_tkeep          : STD_LOGIC_VECTOR(GC_DATA_WIDTH / 8 - 1 downto 0);
  signal m_axis_tlast          : STD_LOGIC;
  signal empty                 : STD_LOGIC:= '1' ;
  --signal full                  : STD_LOGIC:= '0' ;
  constant C_SCOPE             : string := C_TB_SCOPE_DEFAULT;
  signal axistream_if          : t_axistream_if
  (
    tdata  (GC_DATA_WIDTH-1 downto 0),          
    tkeep  ((GC_DATA_WIDTH / 8) - 1 downto 0),  
	tuser  (GC_USER_WIDTH-1 downto 0),              
    tstrb  ((GC_DATA_WIDTH / 8) - 1 downto 0),
    tdest  (C_S_AXIS_TDEST_WIDTH - 1 downto 0),
    tid    (C_S_AXIS_TID_WIDTH - 1 downto 0)
  );
  signal axistream_if_s      : t_axistream_if
  (
    tdata  (GC_DATA_WIDTH-1 downto 0),           
	tkeep  ((GC_DATA_WIDTH / 8) - 1 downto 0),   
	tuser  (GC_USER_WIDTH-1 downto 0),           
    tstrb  ((GC_DATA_WIDTH / 8) - 1 downto 0),
    tdest  (C_S_AXIS_TDEST_WIDTH - 1 downto 0),
    tid    (C_S_AXIS_TID_WIDTH - 1 downto 0)
  );
  

begin
  axistrm_vvc_m: entity bitvis_vip_axistream.axistream_vvc
                        generic map(
                                    GC_VVC_IS_MASTER=> true, 
                                    GC_DATA_WIDTH   => GC_DATA_WIDTH,
                                    GC_USER_WIDTH   => GC_USER_WIDTH,
                                    GC_INSTANCE_IDX => GC_INSTANCE_IDX
                        )
                        port map(
                              clk                   => clk,
                              axistream_vvc_if      => axistream_if
                        );
  axistrm_vvc_s: entity bitvis_vip_axistream.axistream_vvc
                        generic map(
                                    GC_VVC_IS_MASTER=> false, 
                                    GC_DATA_WIDTH   => GC_DATA_WIDTH,
                                    GC_USER_WIDTH   => GC_USER_WIDTH,
                                    GC_INSTANCE_IDX => GC_INSTANCE_IDX+1
                        )
                         port map(
                              clk                   => clk,
                              axistream_vvc_if      => axistream_if_s
                        );
  uut: axis_fifo generic map ( 
                                    GC_DATA_WIDTH   => GC_DATA_WIDTH,
                                    GC_USER_WIDTH   => GC_USER_WIDTH,  
                                    GC_FIFO_DEPTH   => GC_FIFO_DEPTH )
                    port map (      clk             => clk,
                                    rst             => rst,
                                    s_axis_tready   => axistream_if.tready,
                                    s_axis_tvalid   => axistream_if.tvalid,
                                    s_axis_tdata    => axistream_if.tdata,
                                    s_axis_tuser    => axistream_if.tuser,
                                    s_axis_tkeep    => axistream_if.tkeep,
                                    s_axis_tlast    => axistream_if.tlast,
                                    m_axis_tready   => axistream_if_s.tready,
                                    m_axis_tvalid   => axistream_if_s.tvalid,
                                    m_axis_tdata    => axistream_if_s.tdata,
                                    m_axis_tuser    => axistream_if_s.tuser,
                                    m_axis_tkeep    => axistream_if_s.tkeep,
                                    m_axis_tlast    => axistream_if_s.tlast,
                                    empty           => empty
                      --              full            => full
				    );
  --FRAMEWORK INSTANTIATION
  i_ti_uvvm_engine : entity uvvm_vvc_framework.ti_uvvm_engine;
  --CLOCK AND RESET SIGNALS
  clock_generator(clk, clock_ena, C_CLK_PERIOD, "AXI Stream Clock Signal");
  gen_pulse(rst,'0', 2*C_CLK_PERIOD, "AXI Stream Reset signal");
  wait_until_given_time_after_rising_edge(clk,5*C_CLK_PERIOD);
  p_axistream_vvc: process
    variable axistream_bfm_config : t_axistream_bfm_config := C_AXISTREAM_BFM_CONFIG_DEFAULT;
    variable v_data_array : t_byte_array(0 to 10);
      begin
      await_uvvm_initialization(VOID);
      wait for 15 ns;
      axistream_bfm_config.clock_period := C_CLK_PERIOD; 
      log(ID_LOG_HDR, "AXI STREAM VVC BEGINS HERE ");
      clock_ena <= true;   
 
      v_data_array(0) := x"AA";
      v_data_array(1) := x"BB";
      v_data_array(2) := x"CC";
      v_data_array(3) := x"DD";

      axistream_transmit(AXISTREAM_VVCT, GC_INSTANCE_IDX ,v_data_array(0),"Written Data 0",C_SCOPE);
      axistream_transmit(AXISTREAM_VVCT, GC_INSTANCE_IDX ,v_data_array(1),"Written Data 1",C_SCOPE);
      axistream_transmit(AXISTREAM_VVCT, GC_INSTANCE_IDX ,v_data_array(2),"Written Data 2",C_SCOPE);
      axistream_transmit(AXISTREAM_VVCT, GC_INSTANCE_IDX ,v_data_array(3),"Written Data 3",C_SCOPE);

      log("Begin of 50 clk periods waiting");
      wait for 400 ns;
      log("End of 50 clk periods waiting");
      std.env.stop;
      wait;
  end process;
end bench; `

Paebbels, 2023-10-10:
Have you asked that question at UVVM? Bound check errors are usually not a simulator error, but a bug in the framework or user code.

gabrielzb19, 2023-10-11:
Hi.
Yes I’ve asked it in its forum but they suggest me the changing to ModelSim and then it worked, now I’m focused on finding the reason why it didn’t work in GHDL, I think this
Open source  tool is powerful and I’m really interested to keep using it.

tgingold, 2023-10-11:
Which version of ghdl are you using (ghdl -v) ?  I got:

```
*****************************************************************************************************
 This is a *** LICENSED PRODUCT *** as given in the LICENSE.TXT in the root directory.
*****************************************************************************************************

/home/tgingold/devel/vhdl-testsuite/UVVM/UVVM-2023_03_21/script/../uvvm_util/script/../src/license_pkg.vhd:65:7:(report note): 

=====================================================================================================
=====================================================================================================
This info section may be turned off via C_SHOW_UVVM_UTILITY_LIBRARY_INFO in adaptations_pkg.vhd

Important Simulator setup: 
- Set simulator to break on severity 'FAILURE' 
- Set simulator transcript to a monospace font (e.g. Courier new)

UVVM Utility Library setup:
- It is recommended to go through the two powerpoint presentations provided with the download
- There is a Quick-Reference in the doc-directory
- In order to change layout or behaviour - please check the src*/adaptations_pkg.vhd
  This is intended for personal or company customization

License conditions are given in LICENSE.TXT
=====================================================================================================
=====================================================================================================

UVVM: ID_CONSTRUCTOR                     0.0 ns  AXISTREAM_VVC,1                VVC instantiated.
UVVM: ID_CONSTRUCTOR_SUB                 0.0 ns  AXISTREAM_VVC,1                Command queue instantiated and will give a warning when reaching 1000 elements in queue.
UVVM: ID_CONSTRUCTOR_SUB                 0.0 ns  AXISTREAM_VVC,1                Result queue instantiated and will give a warning when reaching 1000 elements in queue.
UVVM: ID_CONSTRUCTOR                     0.0 ns  AXISTREAM_VVC,2                VVC instantiated.
UVVM: ID_CONSTRUCTOR_SUB                 0.0 ns  AXISTREAM_VVC,2                Command queue instantiated and will give a warning when reaching 1000 elements in queue.
UVVM: ID_CONSTRUCTOR_SUB                 0.0 ns  AXISTREAM_VVC,2                Result queue instantiated and will give a warning when reaching 1000 elements in queue.
UVVM: ID_CMD_INTERPRETER_WAIT            0.0 ns  AXISTREAM_VVC,2                  ..Interpreter: Waiting for command
UVVM: ID_CMD_INTERPRETER_WAIT            0.0 ns  AXISTREAM_VVC,1                  ..Interpreter: Waiting for command
UVVM: ID_CMD_EXECUTOR_WAIT               0.0 ns  AXISTREAM_VVC,1                  ..Executor: Waiting for command
UVVM: ID_CMD_EXECUTOR_WAIT               0.0 ns  AXISTREAM_VVC,2                  ..Executor: Waiting for command
UVVM: 
UVVM: 
UVVM: ID_LOG_HDR                        15.0 ns  TB seq.                        AXI STREAM VVC BEGINS HERE 
UVVM: -------------------------------------------------------------------------------------------------------------------------------------------------------------------------
UVVM: ID_CLOCK_GEN                      15.0 ns  TB seq.                        Starting clock AXI Stream Clock Signal
UVVM: ID_UVVM_SEND_CMD                  15.0 ns  TB seq.                        ->axistream_transmit(AXISTREAM_VVC,1,     1 words): 'Written Data 0'. [1]
UVVM: ID_CMD_INTERPRETER                15.0 ns  AXISTREAM_VVC,1                  axistream_transmit(AXISTREAM_VVC,1,     1 words). Command received  [1]
UVVM: ID_UVVM_CMD_ACK                   15.0 ns  TB seq.                            ACK received.   [1]
UVVM: ID_CMD_EXECUTOR                   15.0 ns  AXISTREAM_VVC,1                  axistream_transmit(AXISTREAM_VVC,1,     1 words) - Will be executed  [1]
UVVM: ID_CMD_INTERPRETER_WAIT           15.0 ns  AXISTREAM_VVC,1                  ..Interpreter: Waiting for command
UVVM: ID_UVVM_SEND_CMD                  15.0 ns  TB seq.                        ->axistream_transmit(AXISTREAM_VVC,1,     1 words): 'Written Data 1'. [2]
UVVM: ID_CMD_INTERPRETER                15.0 ns  AXISTREAM_VVC,1                  axistream_transmit(AXISTREAM_VVC,1,     1 words). Command received  [2]
UVVM: ID_UVVM_CMD_ACK                   15.0 ns  TB seq.                            ACK received.   [2]
UVVM: ID_CMD_INTERPRETER_WAIT           15.0 ns  AXISTREAM_VVC,1                  ..Interpreter: Waiting for command
UVVM: ID_UVVM_SEND_CMD                  15.0 ns  TB seq.                        ->axistream_transmit(AXISTREAM_VVC,1,     1 words): 'Written Data 2'. [3]
UVVM: ID_CMD_INTERPRETER                15.0 ns  AXISTREAM_VVC,1                  axistream_transmit(AXISTREAM_VVC,1,     1 words). Command received  [3]
UVVM: ID_UVVM_CMD_ACK                   15.0 ns  TB seq.                            ACK received.   [3]
UVVM: ID_CMD_INTERPRETER_WAIT           15.0 ns  AXISTREAM_VVC,1                  ..Interpreter: Waiting for command
UVVM: ID_UVVM_SEND_CMD                  15.0 ns  TB seq.                        ->axistream_transmit(AXISTREAM_VVC,1,     1 words): 'Written Data 3'. [4]
UVVM: ID_CMD_INTERPRETER                15.0 ns  AXISTREAM_VVC,1                  axistream_transmit(AXISTREAM_VVC,1,     1 words). Command received  [4]
UVVM: ID_UVVM_CMD_ACK                   15.0 ns  TB seq.                            ACK received.   [4]
[... cut ...]
UVVM: ID_PACKET_DATA                    50.0 ns  AXISTREAM_VVC,1                axistream_transmit(1 words[8b])=> Tx x"DD", byte# 0. 'Written Data 3'  [4]
UVVM: ID_PACKET_COMPLETE                57.5 ns  AXISTREAM_VVC,1                axistream_transmit(1 words[8b])=> Tx DONE. 'Written Data 3'  [4]
UVVM: ID_CMD_EXECUTOR_WAIT              57.5 ns  AXISTREAM_VVC,1                  ..Executor: Waiting for command
UVVM: ID_SEQUENCER                     415.0 ns  TB seq.                        End of 50 clk periods waiting
simulation stopped @415ns
```

gabrielzb19, 2023-10-11:
GHDL version 3.0 and you ?
[image]

tgingold, 2023-10-12:
So apparently the bug has been fixed in development branch of ghdl.

gabrielzb19, 2023-10-12:
I've solved it by compiling on Ubuntu 20.04.1, there I've installed GHDL 4.0.0 and I got the same result as you.

gabrielzb19, 2023-10-16:
@tgingold can you try to compile and run the following code for Axilite on the same GHDL version you run the previous code, please ? I just want to make sure that I've installed GHDL correctly.

```ruby
library IEEE;
use     IEEE.Std_logic_1164.all;
use     IEEE.Numeric_Std.all;
library uvvm_util;
context uvvm_util.uvvm_util_context;
library uvvm_vvc_framework;
use     uvvm_vvc_framework.ti_vvc_framework_support_pkg.all;
library bitvis_vip_axilite;
context bitvis_vip_axilite.vvc_context;

entity axilite_supersimple_tb is
end;

architecture bench of axilite_supersimple_tb is

  component axilite_supersimple
  	generic (
  		C_S_AXI_DATA_WIDTH	    : integer	:= 32;
  		C_S_AXI_ADDR_WIDTH	    : integer	:= 4
  	);
  	port (
  		S_AXI_ACLK	            : in std_logic;
  		S_AXI_ARESETN	        : in std_logic;
  		S_AXI_AWADDR	        : in std_logic_vector(C_S_AXI_ADDR_WIDTH-1 downto 0);
  		S_AXI_AWPROT	        : in std_logic_vector(2 downto 0);
  		S_AXI_AWVALID	        : in std_logic;
  		S_AXI_AWREADY	        : out std_logic;
  		S_AXI_WDATA	            : in std_logic_vector(C_S_AXI_DATA_WIDTH-1 downto 0);
  		S_AXI_WSTRB	            : in std_logic_vector((C_S_AXI_DATA_WIDTH/8)-1 downto 0);
  		S_AXI_WVALID	        : in std_logic;
  		S_AXI_WREADY	        : out std_logic;
  		S_AXI_BRESP	            : out std_logic_vector(1 downto 0);
  		S_AXI_BVALID	        : out std_logic;
  		S_AXI_BREADY	        : in std_logic;
  		S_AXI_ARADDR	        : in std_logic_vector(C_S_AXI_ADDR_WIDTH-1 downto 0);
  		S_AXI_ARPROT	        : in std_logic_vector(2 downto 0);
  		S_AXI_ARVALID	        : in std_logic;
  		S_AXI_ARREADY	        : out std_logic;
  		S_AXI_RDATA	            : out std_logic_vector(C_S_AXI_DATA_WIDTH-1 downto 0);
  		S_AXI_RRESP	            : out std_logic_vector(1 downto 0);
  		S_AXI_RVALID	        : out std_logic;
  		S_AXI_RREADY	        : in std_logic
  	);
  end component;

  constant C_S_AXI_DATA_WIDTH   : integer	:= 32;
  constant C_S_AXI_ADDR_WIDTH	: integer	:= 4;
  constant C_CLK_PERIOD         : time      := 10 ns;
  constant C_AXILITE_VVC_IDX    : integer   := 1;
  signal   clock_ena            : boolean   := true;
  signal   S_AXI_ACLK           : std_logic;
  signal   S_AXI_ARESETN        : std_logic := '1';
  signal   S_AXI_AWADDR         : std_logic_vector(C_S_AXI_ADDR_WIDTH-1 downto 0);
  signal   S_AXI_AWPROT         : std_logic_vector(2 downto 0);
  signal   S_AXI_AWVALID        : std_logic;
  signal   S_AXI_AWREADY        : std_logic;
  signal   S_AXI_WDATA          : std_logic_vector(C_S_AXI_DATA_WIDTH-1 downto 0);
  signal   S_AXI_WSTRB          : std_logic_vector((C_S_AXI_DATA_WIDTH/8)-1 downto 0);
  signal   S_AXI_WVALID         : std_logic;
  signal   S_AXI_WREADY         : std_logic;
  signal   S_AXI_BRESP          : std_logic_vector(1 downto 0);
  signal   S_AXI_BVALID         : std_logic;
  signal   S_AXI_BREADY         : std_logic;
  signal   S_AXI_ARADDR         : std_logic_vector(C_S_AXI_ADDR_WIDTH-1 downto 0);
  signal   S_AXI_ARPROT         : std_logic_vector(2 downto 0);
  signal   S_AXI_ARVALID        : std_logic;
  signal   S_AXI_ARREADY        : std_logic;
[... 78 lines cut ...]
    shared_axilite_vvc_config(1).bfm_config.clock_period := C_CLK_PERIOD;
    log(ID_LOG_HDR, "AXI LITE VVC BEGINS HERE ");
    --WRITING PROCESS
    log(ID_LOG_HDR,"WRITING PROCESS");
    v_data:=x"A0A1A2A3";
    v_addr:=x"D";
    S_AXI_AWADDR<=std_logic_vector(v_addr);          
    axilite_write(AXILITE_VVCT, C_AXILITE_VVC_IDX, v_addr, v_data, "Written Data");               
    --AWAITING FOR ACK IS NEEDED 
    --await_completion(AXILITE_VVCT, C_AXILITE_VVC_IDX, 1000 ns);  
    --READING PROCESS
    log(ID_LOG_HDR,"READING PROCESS");
    S_AXI_ARADDR<=std_logic_vector(v_addr);
    axilite_read(AXILITE_VVCT, 1, v_addr, "Read Data From Register",C_SCOPE);
    wait_until_given_time_after_rising_edge(S_AXI_ACLK,15*C_CLK_PERIOD);  
    --CHECKING PROCESS
    log(ID_LOG_HDR,"CHECKING PROCESS");
    axilite_check(AXILITE_VVCT, 1, v_addr, v_data, "Checked Correctly",ERROR,C_SCOPE);
    wait_until_given_time_after_rising_edge(S_AXI_ACLK,15*C_CLK_PERIOD);  
    std.env.stop;
    wait;
  end process;

end;
```