ghdl/ghdl #2516: GHDL crashes analyzing assignments to aggregates

HoT-1, 2023-10-10:
**Description**
GHDL crashes when analyzing the code below.

**Expected behaviour**
The code is valid VHDL. GHDL should be able to analyze.

**How to reproduce?**
Run the command with the file below:

```vhd :file: test.vhd
LIBRARY ieee;
USE ieee.std_logic_1164.ALL;

ENTITY test IS
  GENERIC(
    MASTER_ID_WIDTH          : positive :=  4;
    ID_EXTENSION_WIDTH       : positive :=  4
  );
  PORT(
    bid_extension  : OUT std_logic_vector(ID_EXTENSION_WIDTH-1 DOWNTO 0);
    rid_extension  : OUT std_logic_vector(ID_EXTENSION_WIDTH-1 DOWNTO 0)
  );
END ENTITY;

ARCHITECTURE rtl OF test IS

  CONSTANT SLAVE_ID_WIDTH : positive := MASTER_ID_WIDTH + ID_EXTENSION_WIDTH;

  SIGNAL master_bid           : std_logic_vector(MASTER_ID_WIDTH-1 DOWNTO 0);
  SIGNAL master_rid           : std_logic_vector(MASTER_ID_WIDTH-1 DOWNTO 0);
  SIGNAL master_bid_extended  : std_logic_vector(SLAVE_ID_WIDTH-1 DOWNTO 0);
  SIGNAL master_rid_extended  : std_logic_vector(SLAVE_ID_WIDTH-1 DOWNTO 0);

BEGIN

  (bid_extension, master_bid) <= master_bid_extended;
  (rid_extension, master_rid) <= master_rid_extended;

END ARCHITECTURE;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 test.vhd
```

**Context**

- OS: RHEL 8.5
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: b67ace3f4

The following `GHDL Bug occurred` block is shown in the log:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r619.gb67ace3f4.dirty) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/user/workspace/ghdl_issues/GHDL crashes analyzing assignments to aggregates/
Command line:
/home/user/install/oss-cad-suite/libexec/ghdl1-llvm --std=08 -P/home/user/install/oss-cad-suite/lib/ghdl/ieee/v08/ -P/home/user/install/oss-cad-suite/lib/ghdl/ -fpic -c -o test.o test.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap8.adb:4634
Call stack traceback locations:
0x7f336b8b8d12 0x7f3370438b6b 0x7f3370439be5 0x7f337043c13e 0x7f337043c2d1 0x7f3370440190 0x7f3370445e88 0x7f33704cca7f 0x7f337041407c 0x7f33704d38ac 0x7f33704d530c 0x7f33704d6957 0x7f337008dfb5 0x7f336b1550b1 0x7f337008c82c 0xfffffffffffffffe
******************************************************************
```

**Additional context**
Interestingly the problem only occurs if both assignments exist. Remove either one of them and the problem disappears.

tgingold, 2023-10-11:
I can reproduce the crash.  Thank you for the small reproducer, it 
always help.

HoT-1, 2023-10-11:
Thank you for your dedication.
You're awesome!

commit 038a8854 (2023-10-12): trans-chap8: handle unconstrained target aggregates

Fix #2516
closed 2023-10-12 by tgingold