ghdl/ghdl #2517: Not Compatible with macOS 14

jumper385, 2023-10-11:
**Description**
I am getting an error which says the following: 
"This version of “gtkwave-bin” is not compatible with macOS 14 or later and needs to be updated. Contact the app developer for more information."

[image]

**Expected behaviour**
The app should just start

**How to reproduce?**
Update your Mac Device to macOS Sonoma (MacOS14) and setup gtkwave.

**Context**
Please, provide the following information:

- OS: MacOS 14 (Running on Mac M1 - Macbook Air)
- Origin:
  - [ ] homebrew

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

Xiretza, 2023-10-11:
That sounds like a problem with GTKWave, not ghdl.

diogratia, 2023-10-11:
It is a problem with GTKWave which is also undergoing major changes. There also hasn't been a macOS compatible 
GTKWave application since 3.3.107 some 2 years and more in the past. See [GTKWave issue 250](https://github.com/gtkwave/gtkwave/issues/250#).

There's an interim solution where randomplum provides a [GTKWave Homebrew Tap](https://github.com/randomplum/homebrew-gtkwave) allowing you to build a command line binary that will be installed in /usr/local/bin prior to the next GTKWave release following which someone will likely invest the effort to produce a Sonoma compatible application bundle by modifying the Formula. 

The Tap is registered and only requires three Homebrew brew commands:
```none
brew uninstall gtkwave
brew untap randomplum/gtkwave
brew install --HEAD randomplum/gtkwave/gtkwave
```
the first two aren't needed if you've never installed GTKWave from Homebrew previously.

Because this involved compiling GTKWave you'll need the macOS Command Line Tools be installed.

The installed binary isn't fully functional nor are builds for other targets as yet while randomplum has at least made the Tap compatible with Retina Display Macs (HiDPI). The Tap also use a copy of the GTKWave Github repository of a particular commit and tag sourced on randomplums repository because there can be daily changes to GTKWave. The good news is the Finder snapshot will work with the GTKWave window again while the GTKWave grab to file scales incorrectly and print to file fails. The Tap's author randomplum is now a GTKWave developer looking after macOS.

The compile isn't particular time or memory intensive, unlike the original method Tony used to build the App on a Snow Leopard virtual machine. I'd duplicated the effort exactly once some years ago and find the Homebrew endeavor refreshingly easy to the point of sparking interest in researching meson and ninja used by Homebrew to package, build, and install Casks, Kegs and Taps. There's a real emphasis on reproducible results without requiring expertise by end users although when as in this case hitting an unreleased target you also rely on the Tap/Formula developer to make it bullet proof.

njcoburn, 2024-12-09:
I had the same problem as the original poster, this evening - after updating my 2020 M1 Mac to macOS Sequoia 15.1.1.

The solution above worked for me.

closed 2024-12-19 by Paebbels