ghdl/ghdl #2518: [Issue] Xilinx BRAM not synthesized nor simulated

pidgeon777, 2023-10-13:
My VHDL source contains this instance of a Xilinx BRAM, generated with `Vivado 2020.2`:

```vhdl
UUT_VER : entity work.dp_ram_32bit_1024word_xlx
port map(
    clka => CLK_PROG,
    wea(0) => wren_dpRam_ver,
    addra => wraddress_dpRam_ver,
    dina => data_dpRam_ver,

    clkb => CLK_IN,
    enb => rden_dpRam_ver,
    addrb => rdaddress_dpRam_ver,
    doutb => q_dpRam_ver_tmp
);
```

The `dp_ram_32bit_1024word_xlx` entity is defined inside the file `dp_ram_32bit_1024word_xlx_stub.vhd`:

```vhdl
-- Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.
-- --------------------------------------------------------------------------------
-- Tool Version: Vivado v.2020.2 (win64) Build 3064766 Wed Nov 18 09:12:45 MST 2020
-- Date        : Mon Jul 18 17:11:37 2022
-- Host        : PC running 64-bit major release  (build 9200)
-- Command     : write_vhdl -force -mode synth_stub
--               c:/Work/Projects/TEST/TEST.srcs/sources_1/ip/dp_ram_32bit_1024word_xlx/dp_ram_32bit_1024word_xlx_stub.vhdl
-- Design      : dp_ram_32bit_1024word_xlx
-- Purpose     : Stub declaration of top-level module interface
-- Device      : xczu7ev-ffvc1156-1-i
-- --------------------------------------------------------------------------------
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;

entity dp_ram_32bit_1024word_xlx is
  Port ( 
    clka : in STD_LOGIC;
    wea : in STD_LOGIC_VECTOR ( 0 to 0 );
    addra : in STD_LOGIC_VECTOR ( 9 downto 0 );
    dina : in STD_LOGIC_VECTOR ( 31 downto 0 );
    clkb : in STD_LOGIC;
    enb : in STD_LOGIC;
    addrb : in STD_LOGIC_VECTOR ( 9 downto 0 );
    doutb : out STD_LOGIC_VECTOR ( 31 downto 0 )
  );

end dp_ram_32bit_1024word_xlx;

architecture stub of dp_ram_32bit_1024word_xlx is
attribute syn_black_box : boolean;
attribute black_box_pad_pin : string;
attribute syn_black_box of stub : architecture is true;
attribute black_box_pad_pin of stub : architecture is "clka,wea[0:0],addra[9:0],dina[31:0],clkb,enb,addrb[9:0],doutb[31:0]";
attribute x_core_info : string;
attribute x_core_info of stub : architecture is "blk_mem_gen_v8_4_4,Vivado 2020.2";
begin
end;
```

which is the file I import in the 'work' library when simulating the design, among the other sources needed for simulation:

```
...
ghdl -i --workdir=ghdl_lib -PC:/GHDL/lib/ghdl/vendors/altera -PC:/GHDL/lib/ghdl/vendors/xilinx-vivado "IP/Xilinx/RAM/dp_ram_32bit_1024word_xlx.vhd"
...
```

Problem is that when simulating my top entity, the DPRAM doesn't produce any output.

Also, when synthesizing the design, the DPRAM appears not to be synthesized (there is no trace of it in the synthesized netlist output).

Actually the following files are generated by Vivado:

- `dpram_32bit_1024word_xil_stub.v`
- `dpram_32bit_1024word_xil_stub.vhd`
- `dpram_32bit_1024word_xil_sim_netlist.v`
- `dpram_32bit_1024word_xil_sim_netlist.vhd`

Those sources contains many `protect` statements (I don't know if this could be related to my issue).

I already compiled all of the `Xilinx` libraries with the included `ghdl` script (`PowerShell` version) successfully, but at this point I don't know how I could use them.

I don't know if this is a GHDL bug or not, but my question would be how to simulate and synthesize the DPRAM in my design.

Xiretza, 2023-10-13:
Quite simply there's nothing there that could be simulated - as the filename suggests, this is only a stub that defines the entity's ports. The real simulation model is locked behind Digital Restrictions Management (those `protect` statements) and can't be accessed by anything other than vivado.

For synthesis, you need to use the actual FPGA primitives, not the wrappers generated by Vivado wizards.

Alternatively, a much easier solution is to just use pure-VHDL memory implementations and let the synthesis tool's memory inference do its thing.

pidgeon777, 2023-10-13:
> Quite simply there's nothing there that could be simulated - as the filename suggests, this is only a stub that defines the entity's ports. The real simulation model is locked behind Digital Restrictions Management (those `protect` statements) and can't be accessed by anything other than vivado.

Sounds clear.

> For synthesis, you need to use the actual FPGA primitives, not the wrappers generated by Vivado wizards.

Do you know where I could have a look at that?

> Alternatively, a much easier solution is to just use pure-VHDL memory implementations and let the synthesis tool's memory inference do its thing.

So, you mean implementing a generic RAM in VHDL? Then the Vivado synthesys tool would automatically make use of the necessary BRAMs?

If yes, do you know if there is any example which could I take a look at?

Thanks again 👍.

pidgeon777, 2023-10-13:
> > For synthesis, you need to use the actual FPGA primitives, not the wrappers generated by Vivado wizards.
> 
> Do you know where I could have a look at that?

Ok, after a bit of research, you mean for example writing a wrapper around `blk_mem_gen`?

> > Alternatively, a much easier solution is to just use pure-VHDL memory implementations and let the synthesis tool's memory inference do its thing.
> 
> So, you mean implementing a generic RAM in VHDL? Then the Vivado synthesys tool would automatically make use of the necessary BRAMs?

I found this, if it's what you were referring to?:

https://docs.xilinx.com/r/en-US/ug1580-updatemem/Using-XPM-Memory-in-Vivado

diogratia, 2023-10-13:
>Alternatively, a much easier solution is to just use pure-VHDL memory implementations and let the synthesis tool's memory inference do its thing.

[UG901 Vivado Synthesis](https://docs.xilinx.com/v/u/2021.2-English/ug901-vivado-synthesis) Chapter 4.RAM HDL Coding Techniques, Memory Inference Capabilities and RAM HDL Coding Guideline, Simple Dual-Port Block RAM Example,s See Simple Dual-Port Block RAM with Dual Clocks (VHDL) in the current PDF, pages 127-128.

You'd supply a simulation entity and architecture pair instead of instantiating a black box and being tied to a particular vendor's simulation and synthesis tools by relying on Vivado Synthesis to infer memory correctly. Here incidentally giving you simulation portability and synthesis portability to other vendors capable of inferring RAM. 

The subset of vendors capable of inferring RAM approaches a perfect subset, although you could end up with particular vendor attributes cluttering up a design specification. It's a capability inherited from IEEE Std 1076.6-2004 IEEE Standard for VHDL Register Transfer Level (RTL) Synthesis (withdrawn) 6. Modeling hardware elements, 6.5 ROM and RAM memories, 6.5.2 Random-access memory (RAM).

pidgeon777, 2023-10-15:
@diogratia 

Thanks for the info! 👍

closed 2023-10-17 by tgingold