ghdl/ghdl #2519: GHDL crashes analyzing aggregate of sliced vectors

HoT-1, 2023-10-16:
**Description**
GHDL crashes while analyzing the code below.

**Expected behaviour**
The code is valid VHDL. GHDL should be able to analyze.

**How to reproduce?**
Run the command with the file below:

```vhd :file: test_rtl.vhd
LIBRARY ieee;
USE ieee.std_logic_1164.ALL;

ENTITY test IS
  GENERIC(
    ADDR_WIDTH      : positive := 32;
    MASTER_ID_WIDTH : positive :=  4
  );
END ENTITY;

ARCHITECTURE rtl OF test IS

  CONSTANT MASTER_IDX_WIDTH : positive := 2;
  CONSTANT SLAVE_ID_WIDTH   : positive := MASTER_ID_WIDTH+MASTER_IDX_WIDTH;

  FUNCTION to_slv(awid, awaddr: std_logic_vector) RETURN std_logic_vector IS
  BEGIN
    RETURN (awid(SLAVE_ID_WIDTH-1 DOWNTO 0), awaddr(ADDR_WIDTH-1 DOWNTO 0));
  END FUNCTION;

BEGIN

END ARCHITECTURE;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 test_rtl.vhd
```

**Context**

- OS: RHEL 8.5
- Origin:
  - [x] Built from sources: 1ec1eb6d9

The following `GHDL Bug occurred` block is shown in the log:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r636.g1ec1eb6d9.dirty) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/user/workspace/ghdl_issues/GHDL crashes analyzing aggregate of sliced vectors/
Command line:
/home/user/install/oss-cad-suite/libexec/ghdl1-llvm --std=08 -P/home/user/install/oss-cad-suite/lib/ghdl/ieee/v08/ -P/home/user/install/oss-cad-suite/lib/ghdl/ -fpic -c -o test_rtl.o test_rtl.vhd
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-chap3.adb:45 access check failed
Call stack traceback locations:
0x7faf4ae698c7 0x7faf4ae79bee 0x7faf4ae3a547 0x7faf4ae3ad6a 0x7faf4ae3b83a 0x7faf4ae3e8ad 0x7faf4ae3ed7d 0x7faf4ae3eb2d 0x7faf4ade3fd7 0x7faf4adfb69a 0x7faf4adfb8dd 0x7faf4ae82663 0x7faf4ae21638 0x7faf4ae0534f 0x7faf4ae8c053 0x7faf4add35d6 0x7faf4ae92e80 0x7faf4ae948e0 0x7faf4ae95f2b 0x7faf4aa4cfb5 0x7faf45b140b1 0x7faf4aa4b82c 0xfffffffffffffffe
******************************************************************
```

**Additional context**
N/A

gabrielzb19, 2023-10-17:
Hi. 
The compilation and running goes well when you save the file with the same name as the entity name, in this case, the name of the file should be replaced by test. So the command would be likes this:

`ghdl -a --std=08 test.vhd`

HoT-1, 2023-10-17:
> Hi. The compilation and running goes well when you save the file with the same name as the entity name, in this case, the name of the file should be replaced by test. So the command would be like this:
> 
> `ghdl -a --std=08 test.vhd`

I cannot confirm.

gabrielzb19, 2023-10-17:
I can because I've tested your code in GHDL 4.0.0 & Ubuntu 20.04.1 and the compilation, entity elaboration and running worked perfectly.

HoT-1, 2023-10-17:
> I can because I've tested your code in GHDL 4.0.0 & Ubuntu 20.04.1 and the compilation, entity elaboration and running worked perfectly.

There is no GHDL 4.0.0.

Also you are missing the point: I am reporting an issue I found to contribute to the improvement of GHDL. I am not looking for a way to circumvent the issue for which there are numerous.

gabrielzb19, 2023-10-17:
I'm not avoiding the problem I'm just following the rules stated in its official website to compile a file with GHDL.
<img src="data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAssAAAEeCAYAAAB8E1jaAAAAAXNSR0IArs4c6QAAAARnQU1BAACxjwv8YQUAAAAJcEhZcwAAEnQAABJ0Ad5mH3gAAHESSURBVHhe7b1PiGtbdt+/+vK6abtfm2fTdNpIleJC8OTAj2ACQlCBW5CJQZ4YDZRERTzITAJDBIkEwQMTkBxQwCBNQgYOpSQ1EJ5YkEmgLqSgEBjjiSYhcCmXhDtN0236j/vRflC/tfbe55x9js6ffaQjlarq++ne75Z0tvb/P2uvvc7eXyOiJ3YAAAAAAACAGO/MvwAAAAAAAIAYEJYBAAAAAABIAcIyAAAAAAAAKUBYBgAAAAAAIAUIywAAAAAAAKQAYRkAAAAAAIAUICwDAAAAAACQAoRlAAAAAAAAUoCwDAAAAAAAQAoQlgEAAAAAAEgBwjIAAAAAAAApQFgGAAAAAAAgBQjLAAAAAAAApABhGQAAAAAAgBQgLAMAAAAAAJAChGUAAAAAAABSeFZh+Ts3RNU/IfqG+bwv3+qXGx4AAAAAAHjbfI3dk/7z+Iiw/M3/R/SDPyD6pfluH0RY/vXfjIb3jX9G9Bv/mugz8/nL/0n0w/9qPliIv1/7Xfb3l0TfT3guuPg5FqrszN8+P24R/dz8XRaq/CTP/8B8wfX1synR3/4f8zkO+69yeX/5X7ic/5f5jklKr/AV14dfloXjAgAAAAA4MM8qLJdNXFj+xr8i+u5vsxDJAtfPReD6LRbaOiygWQKY+s0/5u9YMCMR0izhzcfFjwiDnyV8n4gRKG1B0SdJqPzqr4h+NIouKMTfV3+YLUhuCZ+MHedWXJy/L/88KuRKGF//a1N+jF8WP0uJ+zt/wmFyfHFhOYl4HlziSixnKc9/QrTmMlJwPX/xezpvX7H7jNvET/02wPj5jiwuMuoEAAAAAG+X12uzzMLPd3+HBSIWnEVI+oIFr+ofaUHum3Xjh/k5C1hrFpq+z/5EsErCxU8RvsWCnfAZC/JJJiMisEl84n7AQiexwPhdTn8RRND8rmh4WfgNwmKh88uN8WCIxMV+P+PfVK24fskCry9kCj//M10Gdhn6SJzEYXypP2YifkXotQVu17g+43r9ggXiNL7D9fzZX3B9cb39kN33WVD+Nn8X/82vcBvx8esEAAAAAMAmEJZFC/u9GxaUjPsOf44LcqKR+558b/u1bIT95zbx70RI8uNQzhLMfL7gMCN+fBezR7bT8T0O5+vme0GEHxEERXMocYom8QciEPJ3zwoLbN9m4ffHIgSz4P6rGUKfIALkT/+K//jN7fpIQ8ol0Mha2t1fsiBqf44jcf1Iyod/myWMJsKC57c5jT/JCD9AyoD9/mgPDe7nv2f+iCF5F9OeSDo431KG9m++5M/ftATkX+E8y3cAAAAAADZKWFZayN9hYYEFOKVlZCFLtHdJ2kz5/jdE0DH+impafS2tuJ+JWUMCf/sHoR8/PRLPV3/JAp32EmiOvzJp/tFfsPDDAo/P1zmNX97rv0UQ+inHW4Zd9L58o85lyELZz1mYEw1skoY2juQlic//SC8U4gubX/1t/g+X7d9ZWlpXfslllqY5FlT6+V+/bH2+I1rsqVsZf4uF1q/+PN9vWlxqoZEi0Ku8/8122H/P38lvvqU/0i+4vQSfuS2JgP0L8QMAAAAAYKGEZREmRQsbaB1ZyPI1jL5wEcCCnmxvK2GE/YlpwkGFUBaI1La6xGtpIr/4Xf4Pf+fbxQYaWIPY6f69CIv8+89YEPp7/bUWpp4Rif9LEdSYX3B600wxfL7FgvDnYgMcEy5/aBYSYqYRX9hI3iMLiyJwmaUugLgsf8MsqmzzCaW557wk2TBvIVpl/ifPnjktLgX/VhZan3e2y07lPUHo/aUxP/k6h6vgMGSxIqYYaheCy8tvI0X44/9I9PFPif4b/9v4R+ZLwx/z9wAAAAB42bxTwiT/8VXMltUXLmy7TiFJEDkkIiirEzNYQLdJE4psAsHIhz+LzfKzYeL/hREUfy5CM3/+tVgZi/Drm558m4XpH/9hunApi4TvJ2ha5aW2UuGwv8d1IS83RtLCaReTjx/H6ieNL1jA/fLPzIc00uKy+Nsp139C2RVBFitiimHvQhSh8W/4P5zGD79P9D84rH/+71lA/hdEnjy8JKr9QP4AAAAAwEvmpF/wE9tlsTWW48OKaknl5IrP/iH/IZpSFqrEnlkENTlZ4rkQ8wNZmPy6EYTl9AXBtp0V7JfuRHNvv/SWyF/zb/gflV9GndrBAqCrjXMEFlSTFk+yaEk6BeQbFf1vkCd2UmfflBcFYzbmytyBFzh5Gui0uCJwGLKT8M3fjcah6j1hoeCnU+02GNRihQX9z/mfXUxW/un/R/S/b/Xfi/9O9C//A9Hj94imf0r0kRc88/+snwEAAADg5fLO33b/zAgTPr5w4WtBd0KEI/NnUWRrX5kfJG3DG7K0p3/3lxy3EUJ/woKXCHNyPvJPWFjzNbfHRr1ExvnxBWFlRsFpSzR3KQILyUrAZaFZkLwnaaxd8O2E7XpXixYWTJOE11/yd3Z+xIl5g8pnzERHmTsYE5Q0suKKI/bvX3I+5Xg8H5X3hIWCsvvmcCPnUHMeJa27mqz8u99nIdn8rfi/RJP/pDXNH/4t/82fAQAAAPCyUZplZTvLAuQXvnD1W9pedEu4yMHX6omgImfmfo+FmFT71wz80xxEw5q2DS9pVkKmMT0Q2175jY8IcT/jtMgpGWQEuh/es9DEzhfqjoosHLh84qdF+C/Uxc1d8viGybfUlbxcZ7/Qp/LOn0W7G3/5T+olLkj6SBlKvfuniAhSF7Jo+Vme6UQenE45BSRr8bVLXD9kodw+R1ryLgJ0ZKFg4k4KV2y/XQRzAAAAALxNgktJRFARQUm0ioLcdPcTFiJsjVvihRA2IiCzgKbCYGFNzjj+lYzfiBbx879hwTVm7+pfGpGEfZGEnJ38uRGQJb2/YOE4foOf7UfdCPfnFDk+TaXBErZsRDsqwrqLn7Q0i+ApeRdN+bc5r3nlkFvGggjIYsttPiZdWiIkXkpi+Y2nWRY78iKhS/n4+UpCwiVTLgGclsjFIQm4xCVh/8JqA4Jdfj5S75I3WYjIKSk/4/T4+UoKQ1B9gP/NLHsAAAAAvCle1Q1+ACQhCxW1U8ELhZ/+GQvJO9gnAwAAAOBtAmEZvA1EG98hZbcOzTEAAAAAXIGwDAAAAAAAQAonfXQcAAAAAAAAzwmEZQAAAAAAAFLYEpYbo1saNcyH0mnQ6PaWZh11x1khJF23s46+Hc3/zGEFznoGAAAAAABAGYTCsteh0eyWejWiWk+EzxntINMejUX/ki4vtRsvzZcAAAAAAACUiBGWGzSaNKl631WC53LMQuhwSB/1QwAAAAAAAN4kWlj2zqlKG7r/uFIfFauV/B8AAAAAAIA3ixaWVw+0pgqdvVef9sJrjGgW2BGPqNEZJdoTe/y972+WYCQdfd6hc/M9AAAAAAAAx8KYYSzoer5Rtspis7wzLChPOID1uKtsibs3j9RqbgdYaQ5oQHc0FJtjsfuo9aIvFUo4/Ds/nOEdUX2fdAEAAAAAALADwQt+q2mbuiy4bvjvWmu0w8t9HnVaLNEux9RfaPuN1WJKQxbCt1jeUHu6IOVrcUdr/qd67keYHM4NXuIDAAAAAABHJhCWhdXimu5Ztt1QjZqToqdhvKezCv/28ZP5nE62H/dwAAAAAAAAOCQRYVmzofthl+abCjWvDnbgMgAAAAAAACdPgrAsrOhBbCN2oBJ7S/C9qIl3IB4OAAAAAAAAx0YJy16jQ52GZXPxvkEXYjZ8tzBfuKBfEqRaKzDfkBMtxPy4GAu6E/vk2gXpJHnU6Mz2e/EQAAAAAACAHVDCsrxGd9Ya0O3thJqVCjV7LaJ5l/pFZGVmNR2qS02aEznybUZXdJ38gl8Oi36X5ssq9VQ4A7ogHS4AAAAAAADH5GvsnvSfgked2YBo2KZpSReSNEa31KMxXRaVvAEAAAAAAHhmYsLynngeySuBC3P1n5h3DHpNWo8vC2upAQAAAAAAeG5KFpY7NBs0qRK807eh5XgYnJcMAAAAAADAS6JcYRkAAAAAAIBXRMrRcQAAAAAAAIDyhGUxwZh1qNClf2BPGjS6vaXbhHKXFytv5ZnvTq1uvAZ1RjNO20jZuR8XXW6zUcccT5hAae3ZxFX8/nhdh1YaTr5OT52XOEY5pbloG0sfN3JBv3jB7FHvhTlmXECI9wuQwQ7jWHnC8mpKN+sm4dK/02DRv6TLS+1O7ti9xohmkx7V6Ya6l30q+u6nJ78PJscZjRInXI+jEWHc+JvNqBO0zQX1L7t081in3oR/n9RmT7A9n3SdHhxeXM38Ojf1HlvpyAvFM9vPbBS9sv8ljlGnlmb0CwDAS2eHcSwQlpUAEpuMiq76F9dzqraypHWz2ow50fAVi+m1klw+yiVKdCL08STRnqqzsl8EsqLr1dQJKe3+oni6uZ1OelW67+rJsTteU605iQm8cgTihHrVexobf5fDG/r4yTxWrGgxbdOl/L7HgnRCA8xvzy8P3b5i2nwuU2ljYX9Pb4dxP1tjhIQVb6vBLoIVDguyDU/qKSEMH7P613yih5tuIBR158T1Ntiqt5uh76dL43WNmrwYsv0412lqmuVhct6jmp2EMuQFW1TATynDGOW3w/3GjdfYL94Gx5wvXuDcBDJwG6uOxXPIq0pYlkFeCSD2ZNS9ofsH9Qt3HKX1zdyKZyyzXpMmicLg28Qun8C9krP3GldNqizHOx8l6J1XuYDu6aMZgVeLa9q696ZxRc3KksY8UAcHsaxYME8atRd9Gi8r1ExqtC9RE+lEjS6sPDXkus4IZqJj11WFy2VpPrdjB7BXmtsCawQReCc9aqpdBL89d2l488gPc67Vf39GlbU/CPHixjpVZ/XxnjZUIftW/NVC6juocDUYip/6ByuBLnWameZiRMa6mzVVe5OUhW8Gp9YOX22/AACcOs8lr77z1FXSG5p32zS1JyMWLuzPGl/STtbECUW1DjLB3cjWWfXc+k24ipErs4Mt95iNSeSZ2o5vbMUb2bIXzRD/JhqOQ1w8eY5mMS2TXQBKM8dhm7hmMhnKhKvCiWnxSmDLRm+fhYbKWxiWpD1Shh6XT3wFt2t8HJdcf558jbppW7E6jtthrUS6qtQpkH+8D1SvbOjR0hor4W95xyKfGwu5X926pt3GpT3ntzGN3b6SyjD6vEPn5vtyWdJSshtIy3K1/Ya/K37TpiZloaHwqCNHSW7m1I3sIqyMYEv06ZHFWV/iNX3GLxpZGG3sit0iWu+uZNdpfpp3ZcULM3 [... cut ...] SVc+jHPZxjcVpt9YDlk9J+svuFzD91aqkXi8yzSStmn+nipyxEKBnT48WMZqMRjUb87+CMbhIup4mw1Zfz2bmf7sDOce2QL7c25jbWraZt7jtc+xPzXPrOePvdgawx4bh5d8M1XxF2So9Le3bvX0nl/JrslnHdtUGOcEq85cyGV12zCTemN2KLCwAAAADw1lGa5beOXKspNv3ZWmU5Ik1WeBCUAQAAAADeCm9asyw3BemtoA0tx8PwnGAAAAAAAAAYmGEAAAAAAACQAswwAAAAAAAASAHCMgAAAAAAAClAWAYAAAAAACAFCMsAAAAAAACkAGEZAAAAAACAFCAsAwAAAAAAkAKEZQAAAAAAAFKAsAwAAAAAAEAKEJYBAAAAAABIAcIyAAAAAAAAKUBYBgAAAAAAIAUIywAAAAAAAKQAYRkAAAAAAIAUICwDAAAAAACQAoRlAAAAAAAAEiH6/wHerIjPVsrhoQAAAABJRU5ErkJggg==">

tgingold, 2023-10-17:
@gabrielzb19 I think this issue reproduces only with the llvm or gcc backend; with the mcode backend it is not triggered with the reproducer (the function is not called).

@HoT-1 I will work on that, but not immediately.  As a workaround, you can concatenate using `&`.

commit 170bc1df (2023-11-14): trans-chap7: handle globally static slice for aggregates.

Fix #2519
closed 2023-11-14 by tgingold