ghdl/ghdl #2520: Conversions between arrays with closely related elements

nickg, 2023-10-17:
**Description**
The 2008 LRM relaxed requirements for closely related array types to allow conversions when the elements are only closely related rather than exactly the same. This should allow conversion between, for example, an array of reals and an array of integers.

**Expected behaviour**
The test case below passes with Questa and Riviera-PRO but not with GHDL

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: ent.vhd
entity ent is
end entity;

architecture test of ent is
begin

    p1: process is
        variable i : integer_vector(1 to 3);
        variable r : real_vector(1 to 3);
    begin
        i := (1, 2, 3);
        r := real_vector(i);
        assert r = (1.0, 2.0, 3.0);
        r := (4.2, 5.9, 7.1);
        i := integer_vector(r);
        assert i = (4, 6, 7);
        r := (1.0, real'left, 5.3);
--      i := integer_vector(r);     -- Range error
        wait;
    end process;

end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 ent.vhd
ghdl --elab-run ent
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [X] Built from sources: `g1ec1eb6d9ecc`

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

diogratia, 2023-10-17:
If you leave off the leading g (1ec1eb6d9ecc) we'll get a link automagically. The --version command will tell us the commit the same way.

You have nothing present that indicates the back end of the version of ghdl you are using.

```none
%: ghdl -a --std=08 ent.vhdl
%: ghdl --elab-run --std=08 ent
ent.vhdl:13:9:@0ms:(assertion error): Assertion violation
ent.vhdl:16:9:@0ms:(assertion error): Assertion violation
%: ghdl --version 
GHDL 4.0.0-dev (3.0.0.3.0.0.r636.g1ec1eb6d9) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 llvm 15.0.7 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
%: $GHDL --version
GHDL 4.0.0-dev (3.0.0.r636.g1ec1eb6d9) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 static elaboration, mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
%: $GHDL -a --std=08 ent.vhdl
%: $GHDL -r --std=08 ent
ent.vhdl:13:9:@0ms:(assertion error): Assertion violation
ent.vhdl:16:9:@0ms:(assertion error): Assertion violation
%: 
```
Can we take it this is either gcc back end or otherwise build or platform specific?

tgingold, 2023-10-18:
Yes, this is not yet supported.