ghdl/ghdl #2522: No overloaded function found matching 'conv_std_logic_vector'

rachitnigam, 2023-10-28:
**Description**
I'm trying to convert a VHDL file to Verilog using GHDL. The file uses `CONV_STD_LOGIC_VECTOR` which doesn't seem to be recognized by GHDL. I *think* I'm importing the right libraries because `CONV_STD_LOGIC_VECTOR` seems to be defined in [`std_arith_logic`](https://github.com/ghdl/ghdl/blob/master/libraries/synopsys/std_logic_arith.vhdl#L173). 

**Expected behaviour**
The file can be transformed into Verilog with hitting any errors.

**How to reproduce?**
The code is:
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.std_logic_arith.all;
use ieee.std_logic_unsigned.all;

entity FPMult is
   port ( clk, rst : in std_logic;
          X : in  std_logic_vector(23+8+2 downto 0);
          Y : in  std_logic_vector(23+8+2 downto 0);
          R : out  std_logic_vector(23+8+2 downto 0)   );
end entity;

architecture arch of FPMult is
   component IntMultiplier_UsingDSP_9_9_18_unsigned_F300_uid4 is
      port ( clk, rst : in std_logic;
             X : in  std_logic_vector(8 downto 0);
             Y : in  std_logic_vector(8 downto 0);
             R : out  std_logic_vector(17 downto 0)   );
   end component;

signal bias, bias_d1 :  std_logic_vector(24 downto 0);
begin
   bias <= CONV_STD_LOGIC_VECTOR(4.1943e+06,25);
end architecture;
```

```vhd :file: ent.vhd
entity ent is
end entity;

architecture a of ent is
begin
  process begin
    report "Hello world" severity note;
    wait;
  end process;
end;
```

The command to run:
```sh
ghdl synth --out=verilog -fsynopsys -fexplicit flopoco.vhdl -e FPMult
```

The reported error message:
```
flopoco.vhdl:23:33:error: no overloaded function found matching 'conv_std_logic_vector'
   bias <= CONV_STD_LOGIC_VECTOR(4.1943e+06,25);
```

Version information:
GHDL 3.0.0 (2.0.0.r1417.g7de967c51) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 llvm 15.0.7 code generator

rachitnigam, 2023-10-28:
Some debugging help from @JulianKemmerer showed that replacing:
```vhdl
   bias <= CONV_STD_LOGIC_VECTOR(4.1943e+06,25);
```
with
```vhdl
   bias <= CONV_STD_LOGIC_VECTOR(41943000,25);
```
makes the command work.

Maybe the problem is that the number is being confused for a float instead of an integer?

diogratia, 2023-10-28:
e.g. IEEE Std 1076-2008:

15.5 Abstract literals
15.5.1 General

There are two classes of abstract literals: real literals and integer literals. A real literal is an abstract literal
that includes a point; an integer literal is an abstract literal without a point. Real literals are the literals of the
type universal_real. Integer literals are the literals of the type universal integer.
--

rachitnigam, 2023-10-28:
Ah, so this is in fact the correct behavior! Just another question, is there a `STD_CONV_LOGIC_VECTOR` version for real literals or should FloPoCo's code generator should actually be generating values in integer form here?

diogratia, 2023-10-28:
>Ah, so this is in fact the correct behavior! Just another question, is there a STD_CONV_LOGIC_VECTOR version for real literals or should FloPoCo's code generator should actually be generating values in integer form here?

The basic issue here is that no one is performing any maintenance on Synopsys packages and they are not standardized. When the were widely used prior to IEEE Std 1076.3-1997 IEEE Standard VHDL Synthesis Packages reals were neither convertible to vector forms nor envisioned to be used in any other form that describing scaled physical types. Today reals are widely recognized in generating constants or initial values by synthesis tools. We also have a synthesis eligible package for floating point arithmetic standardized in the -2008 VHDL standard.

/usr/local/lib/ghdl/src/ieee2008/float_generic_pkg  has 
```vhdl
function to_slv (arg : UNRESOLVED_float) return STD_LOGIC_VECTOR;
```
and
```vhdl
  -- real to float
  function to_float (
    arg                     : REAL;
    constant exponent_width : NATURAL    := float_exponent_width;  -- length of FP output exponent
    constant fraction_width : NATURAL    := float_fraction_width;  -- length of FP output fraction
    constant round_style    : round_type := float_round_style;  -- rounding option
    constant denormalize    : BOOLEAN    := float_denormalize)  -- Use IEEE extended FP
    return UNRESOLVED_float;
```
Note the defaults.

[FloPoCo](https://gitlab.inria.fr/flopoco/flopoco/-/tree/master) contains no reference to a declaration for conv_to_std_logic_vector with a parameter type of real nor an indication of whose library they'd expect to use. In any event for the era the tool comes from (2009 or earlier) ISE synthesis tools would expect Synopsys packages (as shown in flopoco-master/tools/testInputIEEE.sh). In ghdl the synopsys packages used were those submitted to VASG as a basis for development of IEEE Std 1076.3.

/usr/local/lib/ghdl/src/synopsys
```none
%: grep -i conv_std_logic_vector std_logic_arith.vhdl
    function CONV_STD_LOGIC_VECTOR(ARG: INTEGER; SIZE: INTEGER)
    function CONV_STD_LOGIC_VECTOR(ARG: UNSIGNED; SIZE: INTEGER)
    function CONV_STD_LOGIC_VECTOR(ARG: SIGNED; SIZE: INTEGER)
    function CONV_STD_LOGIC_VECTOR(ARG: STD_ULOGIC; SIZE: INTEGER)
    function CONV_STD_LOGIC_VECTOR(ARG: INTEGER; SIZE: INTEGER) return STD_LOGIC_VECTOR is
    function CONV_STD_LOGIC_VECTOR(ARG: UNSIGNED; SIZE: INTEGER) return STD_LOGIC_VECTOR is
    function CONV_STD_LOGIC_VECTOR(ARG: SIGNED; SIZE: INTEGER) return STD_LOGIC_VECTOR is
    function CONV_STD_LOGIC_VECTOR(ARG: STD_ULOGIC; SIZE: INTEGER) return STD_LOGIC_VECTOR is
%:
```
Using -2008:
```vhdl
library ieee;
use ieee.std_logic_1164.all;
-- use ieee.std_logic_arith.all;
-- use ieee.std_logic_unsigned.all;
use ieee.float_pkg.all;
use ieee.numeric_std_unsigned.all;

entity FPMult is
   port ( clk, rst : in std_logic;
          X : in  std_logic_vector(23+8+2 downto 0);
          Y : in  std_logic_vector(23+8+2 downto 0);
          R : out  std_logic_vector(23+8+2 downto 0)   );
end entity;

architecture arch of FPMult is
   component IntMultiplier_UsingDSP_9_9_18_unsigned_F300_uid4 is
      port ( clk, rst : in std_logic;
             X : in  std_logic_vector(8 downto 0);
             Y : in  std_logic_vector(8 downto 0);
             R : out  std_logic_vector(17 downto 0)   );
   end component;

signal bias, bias_d1 :  std_logic_vector(24 downto 0);
begin
   -- bias <= CONV_STD_LOGIC_VECTOR(4.1943e+06,25);
   bias <= resize(to_slv(to_float(4.1943e+06)), 25);
end architecture;
```
successfully analyzes:
```none
%: ghdl -a --std=08 FPMult.vhdl
%: 
```
and introduces no errors during elaboration or simulation.

You'd expect that with the eventual adoption of the -2008 VHDL standard tools like FloPoCo might eventually adopt floating point types (which are grudgingly supported by FPGA vendors today in lieu of licensing their floating point cores).  While type real is tied to the implementation type universal_real, IEEE floating point types (instantiated package float_pkg) support various standard sizes as well as customization through generics and  a new instantiated float_generic_pkg package.

It looks like for purposes of your question FloPoCo should have generated an abstract literal of type universal_integer.

Note that IEEE package numeric_std is purposely incompatible with Synopsys package std_logic_arith to encourage migration some 14+ years after the former was standardized. In that time Synopsys has until purportedly recently also de-emphasized VHDL support. 

This ghdl issue can be closed.

diogratia, 2023-10-29:
9.3.6 Type conversions

>A type conversion provides for explicit conversion between closely related types.
>  
>type_conversion ::= type_mark ( expression )
>  
>...  
>  
>Explicit type conversions are allowed between *closely related types.* In particular, a type is closely related to itself. Other types are closely related only under the following conditions:
>>    
>>— *Abstract numeric types*—Any abstract numeric type is closely related to any other abstract numeric
type.
>>  
>>...

closed 2023-11-04 by tgingold