ghdl/ghdl #2523: [Bug] `--vendor-library=NAME` ha no effect during synthesys

pidgeon777, 2023-10-28:
**Description**

`--vendor-library=NAME` is a synth option which allows to replace any unit from `LIB_NAME` with a black box during synthesys:

https://ghdl.github.io/ghdl/using/Synthesis.html

I'm using that option as an argument to the synthesys:

`ghdl make -Wall -fsynopsys -fexplicit --workdir=ghdl_lib -PC:/GHDL/lib/ghdl/vendors/altera -PC:/GHDL/lib/ghdl/vendors/xilinx-vivado --std=08 -frelaxed Top_ENT && MKDIR ghdl_out & ghdl synth -Wall -fsynopsys -fexplicit --workdir=ghdl_lib -PC:/GHDL/lib/ghdl/vendors/altera -PC:/GHDL/lib/ghdl/vendors/xilinx-vivado --std=08 -frelaxed --out=verilog --latches --vendor-library=unisim --vendor-library=altera_mf Top_ENT`

Problem is that `ghdl` reports an error, `signal assignment not allowed here` for a Xilinx source file:

`C:\Xilinx\Vivado\2020.2\data\vhdl\src\unisims\primitive\IBUFDS.vhd`

of an entity which probably gets instantiated in one of my source files, which belongs to the `unisim` library.

Also, all of the source files defined in the `altera_mf` library get parsed and analyzed, even if the `--vendor-library` argument is applied:

```log
sources_1/new/FINDER/fifo_512word_32bit.vhd:61:14:warning: (in default configuration of fifo_512word_32bit(syn))
C:\altera\16.0\quartus\eda\sim_lib\altera_mf.vhd:48462:5:warning: port "eccstatus" of entity "dcfifo_mixed_widths" is not bound [-Wbinding]
    DCFIFO_MW : DCFIFO_MIXED_WIDTHS
    ^
C:\altera\16.0\quartus\eda\sim_lib\altera_mf.vhd:48406:14:warning: (in default configuration of dcfifo(behavior))
C:\altera\16.0\quartus\eda\sim_lib\altera_mf.vhd:47598:14:warning: declaration of "mem_data" hides signal "mem_data" [-Whide]
    variable mem_data : LPM_MEMORY := (OTHERS => ZEROS_R);
             ^
C:\altera\16.0\quartus\eda\sim_lib\altera_mf.vhd:47596:34:warning: extra signal "write_aclr" in sensitivity list [-Wsensitivity]
    process (aclr, wrclk, rdclk, write_aclr, read_aclr)
                                 ^
C:\altera\16.0\quartus\eda\sim_lib\altera_mf.vhd:47596:46:warning: extra signal "read_aclr" in sensitivity list [-Wsensitivity]
    process (aclr, wrclk, rdclk, write_aclr, read_aclr)
                                             ^
C:\altera\16.0\quartus\eda\sim_lib\altera_mf.vhd:47505:8:warning: signal "mem_data" is never referenced [-Wunused]
signal mem_data : LPM_MEMORY := (OTHERS => ZEROS_R);
       ^
C:\altera\16.0\quartus\eda\sim_lib\altera_mf.vhd:47513:8:warning: signal "i_showahead_flag" is never referenced [-Wunused]
signal i_showahead_flag : std_logic := '0';
       ^
sources_1/new/fifo_kword_mbit.vhd:252:17:warning: instance "fifo_512word_32bit_xil_inst" of component "fifo_512word_32bit_xil" is not bound [-Wbinding]
                fifo_512word_32bit_xil_inst : fifo_512word_32bit_xil
```

**Expected behaviour**

I would expect `--vendor-library=unisim` to skip the `IBUFDS` entity analysis defined in the Xilinx `unisim` library, so that the error is not generated during synthesys (`IBUFDS` instances would be considered as black boxes?).

**How to reproduce?**

```sh
ghdl make -Wall -fsynopsys -fexplicit --workdir=ghdl_lib -PC:/GHDL/lib/ghdl/vendors/altera -PC:/GHDL/lib/ghdl/vendors/xilinx-vivado --std=08 -frelaxed Top_ENT && MKDIR ghdl_out & ghdl synth -Wall -fsynopsys -fexplicit --workdir=ghdl_lib -PC:/GHDL/lib/ghdl/vendors/altera -PC:/GHDL/lib/ghdl/vendors/xilinx-vivado --std=08 -frelaxed --out=verilog --latches --vendor-library=unisim --vendor-library=altera_mf Top_ENT
```

**Context**

Please, provide the following information:

- OS: Windows 10 Pro
- Origin: Official ghdl repository, built from sources.

tgingold, 2023-11-04:
How did you instantiate `IBUFDS` ?  A reproducer would be nice.
It is expected that files are analyzed.

pidgeon777, 2024-04-03:
> How did you instantiate `IBUFDS` ? A reproducer would be nice. It is expected that files are analyzed.

@tgingold My VHDL file content, which I want to synthesize, is:

```vhdl
-------------------------------------------------------------------------------
--
-- (c) Copyright 2020-2023 Advanced Micro Devices, Inc. All rights reserved.
--
-- This file contains confidential and proprietary information
-- of AMD and is protected under U.S. and
-- international copyright and other intellectual property
-- laws.
--
-- DISCLAIMER
-- This disclaimer is not a license and does not grant any
-- rights to the materials distributed herewith. Except as
-- otherwise provided in a valid license issued to you by
-- AMD, and to the maximum extent permitted by applicable
-- law: (1) THESE MATERIALS ARE MADE AVAILABLE "AS IS" AND
-- WITH ALL FAULTS, AND AMD HEREBY DISCLAIMS ALL WARRANTIES
-- AND CONDITIONS, EXPRESS, IMPLIED, OR STATUTORY, INCLUDING
-- BUT NOT LIMITED TO WARRANTIES OF MERCHANTABILITY, NON-
-- INFRINGEMENT, OR FITNESS FOR ANY PARTICULAR PURPOSE; and
-- (2) AMD shall not be liable (whether in contract or tort,
-- including negligence, or under any other theory of
-- liability) for any loss or damage of any kind or nature
-- related to, arising under or in connection with these
-- materials, including for any direct, or any indirect,
-- special, incidental, or consequential loss or damage
-- (including loss of data, profits, goodwill, or any type of
-- loss or damage suffered as a result of any action brought
-- by a third party) even if such damage or loss was
-- reasonably foreseeable or AMD had been advised of the
-- possibility of the same.
--
-- CRITICAL APPLICATIONS
-- AMD products are not designed or intended to be fail-
-- safe, or for use in any application requiring fail-safe
-- performance, such as life-support or safety devices or
-- systems, Class III medical devices, nuclear facilities,
-- applications related to the deployment of airbags, or any
-- other applications that could lead to death, personal
-- injury, or severe property or environmental damage
-- (individually and collectively, "Critical
-- Applications"). Customer assumes the sole risk and
-- liability of any use of AMD products in Critical
-- Applications, subject only to applicable laws and
-- regulations governing limitations on product liability.
--
-- THIS COPYRIGHT NOTICE AND DISCLAIMER MUST BE RETAINED AS
-- PART OF THIS FILE AT ALL TIMES.
--
-------------------------------------------------------------------------------
-- Project    : Series-7 Integrated Block for PCI Express
-- File       : PcieBlock_x1_pcie_bram_7x.vhd
-- Version    : 3.3
--  Description : single bram wrapper for the mb pcie block
--                The bram A port is the write port
--                the      B port is the read port
--
--
---------------------------------------------------------------------------------

library ieee;
use ieee.std_logic_1164.all;
use ieee.std_logic_misc.all;
use ieee.std_logic_unsigned.all;

library unisim;
[... 537 lines cut ...]
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_TDP_MACRO.vhd|1006 col 5 warning| declaration of "device" hides generic "device" [-Whide]
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_TDP_MACRO.vhd|1106 col 5 warning| declaration of "device" hides generic "device" [-Whide]
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_TDP_MACRO.vhd|1233 col 5 warning| declaration of "device" hides generic "device" [-Whide]
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_TDP_MACRO.vhd|1783 col 4 warning| declaration of "web1" hides if generate statement [-Whide]
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_TDP_MACRO.vhd|1792 col 4 warning| declaration of "web2" hides if generate statement [-Whide]
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_TDP_MACRO.vhd|1801 col 2 warning| declaration of "web1" hides if generate statement [-Whide]
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unisims\primitive\RAMB18E1.vhd|3214 col 36| no declaration for "gsr"
```

The error is at the last log line.

Also, it seems this `gsr` signal is a "global" one, defined in `C:\Xilinx\Vivado\2023.2\data\vhdl\src\unisims\unisim_VCOMP.vhd`:

```vhdl
-----------------------------------------
-----------   FPGA Globals --------------
-----------------------------------------
signal GSR : std_logic := '0';
```

This `gsr` signal is also causing me similar issues on the **Xilinx**  `XADC` entity, and other **Xilinx** primitives.

I wonder what I'm doing wrong, and how these global signals from **Xilinx** can be managed, in the synthesis phase.

Also, all of those Xilinx primitives should be defined as black boxes when synthesizing, or it would be still possibile to synthesize them someway?

tgingold, 2024-04-04:
You are running two commands: ghdl make and  ghdl synth.  Which one generates the error ?  I suppose this is the first one (ghdl make).
It would be nice to have a smaller reproducer!

And, these primitives should be defined as black box (except maybe unimacros)

pidgeon777, 2024-04-04:
This is what I did, we consider the source file I provided above as the input file.

1. `ghdl import -Wall -fsynopsys -fexplicit --workdir=ghdl_lib -PC:/GHDL/LLVM/lib/ghdl/vendors/altera -PC:/GHDL/LLVM/lib/ghdl/vendors/xilinx-vivado --std=08 -frelaxed FPGA\PcieBlock_x1_pcie_bram_7x.vhd`

2. `ghdl make -Wall -fsynopsys -fexplicit --workdir=ghdl_lib -PC:/GHDL/LLVM/lib/ghdl/vendors/altera -PC:/GHDL/LLVM/lib/ghdl/vendors/xilinx-vivado --std=08 -frelaxed PcieBlock_x1_pcie_bram_7x`

```log
FPGA\PcieBlock_x1_pcie_bram_7x.vhd:212:5:warning: declaration of "impl_target" hides generic "impl_target" [-Whide]
    impl_target   : string)
    ^
FPGA\PcieBlock_x1_pcie_bram_7x.vhd:224:5:warning: declaration of "width" hides generic "width" [-Whide]
    WIDTH      : integer;
    ^
FPGA\PcieBlock_x1_pcie_bram_7x.vhd:239:5:warning: declaration of "width" hides generic "width" [-Whide]
    WIDTH  : integer)
    ^
FPGA\PcieBlock_x1_pcie_bram_7x.vhd:192:12:warning: function "pad_val" is never referenced [-Wunused]
  function pad_val (
           ^
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_SDP_MACRO.vhd:220:5:warning: declaration of "bram_size" hides generic "bram_size" [-Whide]
    bram_size : in string;
    ^
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_SDP_MACRO.vhd:221:5:warning: declaration of "device" hides generic "device" [-Whide]
    device : in string;
    ^
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_SDP_MACRO.vhd:280:5:warning: declaration of "bram_size" hides generic "bram_size" [-Whide]
    bram_size : in string;
    ^
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_SDP_MACRO.vhd:281:5:warning: declaration of "device" hides generic "device" [-Whide]
    device : in string
    ^
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_SDP_MACRO.vhd:392:5:warning: declaration of "bram_size" hides generic "bram_size" [-Whide]
    bram_size : in string;
    ^
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_SDP_MACRO.vhd:393:5:warning: declaration of "device" hides generic "device" [-Whide]
    device : in string
    ^
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_SDP_MACRO.vhd:396:14:warning: variable "message" is never referenced [-Wunused]
    variable Message : LINE;
             ^
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_SDP_MACRO.vhd:423:5:warning: declaration of "bram_size" hides generic "bram_size" [-Whide]
    bram_size : in string;
    ^
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_SDP_MACRO.vhd:424:5:warning: declaration of "device" hides generic "device" [-Whide]
    device : in string
    ^
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_SDP_MACRO.vhd:427:14:warning: variable "message" is never referenced [-Wunused]
    variable Message : LINE;
             ^
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_SDP_MACRO.vhd:454:5:warning: declaration of "device" hides generic "device" [-Whide]
    device : in string
    ^
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_SDP_MACRO.vhd:485:5:warning: declaration of "bram_size" hides generic "bram_size" [-Whide]
    bram_size : in string;
    ^
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_SDP_MACRO.vhd:486:5:warning: declaration of "device" hides generic "device" [-Whide]
    device : in string
    ^
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_SDP_MACRO.vhd:528:5:warning: declaration of "bram_size" hides generic "bram_size" [-Whide]
    bram_size : in string;
    ^
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_SDP_MACRO.vhd:529:5:warning: declaration of "device" hides generic "device" [-Whide]
    device : in string
    ^
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_SDP_MACRO.vhd:564:5:warning: declaration of "bram_size" hides generic "bram_size" [-Whide]
    bram_size : in string;
    ^
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_SDP_MACRO.vhd:565:5:warning: declaration of "device" hides generic "device" [-Whide]
    device : in string
    ^
[... 642 lines cut ...]
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_TDP_MACRO.vhd:1106:5:warning: declaration of "device" hides generic "device" [-Whide]
    device : in string;
    ^
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_TDP_MACRO.vhd:1233:5:warning: declaration of "device" hides generic "device" [-Whide]
    device : in string;
    ^
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_TDP_MACRO.vhd:1783:4:warning: declaration of "web1" hides if generate statement [-Whide]
   web1 : web_pattern <= (WEB & WEB) when (BRAM_SIZE = "18Kb" and web_width = 1 ) else
   ^
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_TDP_MACRO.vhd:1792:4:warning: declaration of "web2" hides if generate statement [-Whide]
   web2 : web_pattern <= (WEB & WEB) when (BRAM_SIZE = "9Kb" and web_width = 1 ) else
   ^
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unimacro\BRAM_TDP_MACRO.vhd:1801:2:warning: declaration of "web1" hides if generate statement [-Whide]
 web1 : web_pattern_bl <= ("00" & WEB & WEB) when (BRAM_SIZE = "18Kb" and web_width = 1  ) else
 ^
elaborate pcieblock_x1_pcie_bram_7x
```

3. `ghdl synth -Wall -fsynopsys -fexplicit --workdir=ghdl_lib -PC:/GHDL/LLVM/lib/ghdl/vendors/altera -PC:/GHDL/LLVM/lib/ghdl/vendors/xilinx-vivado --std=08 -frelaxed --out=verilog --vendor-library=unisim --vendor-library=unimacro PcieBlock_x1_pcie_bram_7x > ghdl_out\PcieBlock_x1_pcie_bram_7x_ghdl_synth.v`

```log
C:\Xilinx\Vivado\2023.2\data\vhdl\src\unisims\primitive\RAMB18E1.vhd:3214:36: no declaration for "gsr"
    gsr_dly              <= TO_X01(GSR)         after 0 ps;
                                   ^
```

pidgeon777, 2024-04-08:
Hello @tgingold, just wanted to ask if these steps are enough or if you need more information to replicate the issue. If yes, I'll be more than willing to provide them 👍.

tgingold, 2024-04-11:
As I don't have vivado installed on my machine, it would be nice if you could create a standalone reproducer.