ghdl/ghdl #2524: Can not build GHDL with llvm

RealTigerCZ, 2023-10-29:
**Description**
Encountered compilation error when running `make` with ghdl 3.0.0. 

**Expected behaviour**
Expected to do `make` and then make install without problems.

**How to reproduce?**
Firstly I was trying   `sudo dnf install ghdl` but it didn't work with this error message.

```
Poslední kontrola metadat: před 2:59:52, Ne 29. října 2023, 10:44:07.
Chyba: 
 Problém: conflicting requests
  - nothing provides libgnat-12.so()(64bit) needed by ghdl-0.38~dev-16.20201208git83dfd49.fc37.x86_64 from fedora
(try to add '--skip-broken' to skip uninstallable packages)
```

I didn't find any good suggestions what to do, except trying fedora37 packages. I didn't try that, because lastly I broken many things with it.

So I've downloaded 3.0.0 ghdl release (source code)  because there wasn't any package compiled directly for Fedora 38. Then I was trying to compile ghdl as  described in https://ghdl.github.io/ghdl/development/building/LLVM.html
I installed gnat 13.2.1, and i had only llvm 16.0.6 so i needed older version. I ended up with manual compiling llvm 14.0.6.

Then in build folder `../configure --with-llvm-config --prefix=/usr/local`. 
But when I tried `make` It stated great with compiling some thing but it crashed with this error:

```
ranlib lib/ghdl/libgrt.a
sed -e "\!^.[/\\]!d" -e "/-shared/d" -e "/-static/d" -e "/-lgnat/d" \
  -e "\X-L/Xd" < grt/grt-files > grt/grt-files.in
echo "@/libgrt.a" > lib/ghdl/grt.lst
for i in -ldl -lm; do echo $i >> lib/ghdl/grt.lst; done
cat grt/grt-files.in >> lib/ghdl/grt.lst
make -f ../libraries/Makefile.inc LIBSRC_DIR="../libraries" LIBDST_DIR="lib/ghdl" enable_gplcompat="false" LN="ln -s" CP="cp" MKDIR="mkdir" GHDL=/home/tigercz/Stažené/install_folder/ghdl-3.0.0/build/ghdl_llvm GHDL_FLAGS="--GHDL1=/home/tigercz/Stažené/install_folder/ghdl-3.0.0/build/ghdl1-llvm " vhdl.libs.all libs.vhdl.standard
make[1]: Vstupuje se do adresáře „/home/tigercz/Stažené/install_folder/ghdl-3.0.0/build“
rm -f -f lib/ghdl/std/v87/std-obj87.cf
cd lib/ghdl/std/v87; \
for i in std/v87/textio.vhdl std/v87/textio-body.vhdl; do \
  cmd="/home/tigercz/Stažené/install_folder/ghdl-3.0.0/build/ghdl_llvm -a --GHDL1=/home/tigercz/Stažené/install_folder/ghdl-3.0.0/build/ghdl1-llvm  --std=87 --bootstrap --work=std ../../src/$i"; \
  echo $cmd; eval $cmd || exit 1; \
done
/home/tigercz/Stažené/install_folder/ghdl-3.0.0/build/ghdl_llvm -a --GHDL1=/home/tigercz/Stažené/install_folder/ghdl-3.0.0/build/ghdl1-llvm --std=87 --bootstrap --work=std ../../src/std/v87/textio.vhdl
ghdl1-llvm: /home/tigercz/Stažené/install_folder/llvm-project-llvmorg-14.0.6/llvm/lib/IR/Attributes.cpp:95: static llvm::Attribute llvm::Attribute::get(llvm::LLVMContext&, AttrKind, uint64_t): Assertion `Attribute::isIntAttrKind(Kind) && "Not an int attribute"' failed.
/home/tigercz/Stažené/install_folder/ghdl-3.0.0/build/ghdl_llvm:error: compilation error
make[1]: *** [../libraries/Makefile.inc:173: lib/ghdl/std/v87/std-obj87.cf] Chyba 1
make[1]: Opouští se adresář „/home/tigercz/Stažené/install_folder/ghdl-3.0.0/build“
make: *** [Makefile:389: libs.vhdl.llvm] Chyba 2
```
ghdl compiled 
```
GHDL 3.0.0 (tarball) [Dunoon edition]
 Compiled with GNAT Version: 13.2.1 20231011 (Red Hat 13.2.1-
 llvm 14.0.6 code generator
Written by Tristan Gingold.
```
But there are missing libraries. (like "std")

- OS: Fedora 38 6.5.7-200
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `7de967c`

tgingold, 2023-11-04:
Try with the master branch.

RealTigerCZ, 2023-11-04:
Ok, i tried now:
```console
git clone --depth=1 https://github.com/ghdl/ghdl.git
cd ghdl
mkdir build
cd build/ 
../configure --with-llvm-config --prefix=/usr/local
```
Which printed this:
```
gcc (GCC) 13.2.1 20231011 (Red Hat 13.2.1-4)
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Use full IEEE library
Build machine is: x86_64-redhat-linux
Debugging is enabled with llvm 14.0.6
Creating Makefile
Creating default_paths.ads
Creating ghdl.gpr
Creating scripts/gcc/Make-lang.in
for d in ieee/v87 ieee/v93 ieee/v08 std/v87 std/v93 std/v08 src/ieee src/ieee/v87 src/ieee/v93 src/ieee2008 src/std src/std/v87 src/std/v93 src/std/v08 src/synopsys src/synopsys/v08 src/upf src/vital95 src/vital2000; do \
  mkdir -p lib/ghdl/$d; \
done
Generate ghdlsynth_maybe.ads
Generate grt-readline.ads
```

Then i did
```console
make
```

It started to compile and after a while with long log it ended up with compilation error again.
```
/home/tigercz/Stažené/install_folder/ghdl/build/ghdl_llvm -a --GHDL1=/home/tigercz/Stažené/install_folder/ghdl/build/ghdl1-llvm --std=87 --bootstrap --work=std ../../src/std/v87/textio.vhdl
ghdl1-llvm: /home/tigercz/Stažené/install_folder/llvm-project-llvmorg-14.0.6/llvm/lib/IR/Attributes.cpp:95: static llvm::Attribute llvm::Attribute::get(llvm::LLVMContext&, AttrKind, uint64_t): Assertion `Attribute::isIntAttrKind(Kind) && "Not an int attribute"' failed.
/home/tigercz/Stažené/install_folder/ghdl/build/ghdl_llvm:error: compilation error
make[1]: *** [../libraries/Makefile.inc:187: lib/ghdl/std/v87/std-obj87.cf] Chyba 1
make[1]: Opouští se adresář „/home/tigercz/Stažené/install_folder/ghdl/build“
make: *** [Makefile:397: libs.vhdl.llvm] Chyba 2
```

Then I tried this (because this 1ec1eb6 commit has a green checkmark with it):
```console
git clone --depth=20 https://github.com/ghdl/ghdl.git
git reset --hard 1ec1eb6
cd ghdl
mkdir build
cd build/ 
../configure --with-llvm-config --prefix=/usr/local
make
```
And same error. 

I tried it compile even with new llvm 16 (although its not supported) it still didn't compile (but with different error).

`sudo dnf install ghdl` Still doesn't not work.

tgingold, 2023-11-05:
Ok, I probably need to rebuild llvm with assertions to reproduce.

tgingold, 2023-11-05:
It doesn't fail for me with llvm 17.  I will try with llvm 14.

tgingold, 2023-11-05:
Should now be ok.  Thank you for your patience.

commit 3c64c548 (2023-11-05): llvm-cbindings.cpp: fix assertion failure.

Fix #2524
closed 2023-11-05 by tgingold