ghdl/ghdl #2527: FST support for custom types

albydnc, 2023-11-01:
I am trying to work on a quite large and complex design (GHW ~ 300 MB). GTKwave is not capable of opening such file, resulting into enormous RAM usage and hanging.
I tried to switch to FST since it should be much better for handling big designs in GTKwave. The issue is that my design has a lot of subtypes, many of which are arrays and they are not exported into the FST format (nor the VCD).
E.g. I have arrays of `std_logic_vector` which are ignored completely.

Is it possible to export these into FST? Is there a workaround?

Paebbels, 2023-11-04:
As I remember, *FST* (*fast*) is a format invented by GTKwave as a faster replacement for VCD and compressed VCD. So it's limited to Verilog features. (VCD - Value-Change-Dump is a format defined by Verilog.)

Internally, GTKwave reads all waveform formats for representation. So when GTKwave has problems to handle a 300MB GHW file, I assume it also has problems with a 300 MB FST file. Have you already reported that issue to GTKwave? Tony Bybell can analyze which parts of GTKwave consumes most memory when opening such large files. It might be interesting to describe if your simulation is
* long (less signals, but many data per signal),
* big (many signals, but medium runtime),
* complex (many signals and deep hierarchy as well as complex datatypes (e.g. many enums)

For GHDL and GHW, you can limit the number of signals written into a waveform. See the [--read-wave-opt=<FILENAME>](https://ghdl.github.io/ghdl/using/Simulation.html#export-waveforms) simulation option. This should enhance simulation speed (less disk I/O) and filesize/complexity (less hierarchy levels).

tgingold, 2023-11-04:
If you recompile a very recent ghdl with the mcode backend, the support of FST has been improved.  But as mentioned by @Paebbels GTKWave has a limited support for record/arrays.

ekiwi, 2024-03-07:
@albydnc The [surfer waveform viewer](http://surfer-project.org/) has a different GHW implementation than GTKWave and is now able to load your file. You will have to use a local installation though, the web version is significantly slower which makes a difference for large files like yours. Feel free to report any problems you find to the [surfer issue tracker](https://gitlab.com/surfer-project/surfer/-/issues). GHW support is very new, so there are definitely still some bugs. One major issue that we are aware of is that enums are currently rendered as numbers instead of their string values. We are working on fixing that.

ekiwi, 2024-03-15:
Surfer correctly renders enums now. This is what your file looks like opened in Surfer:

[image]

Opening it does take ~10-20s, but memory usage is under control (< 400MB). FST would be much faster to load still, because of the way it stores each signal trace separately.