ghdl/ghdl #2528: GHDL Bug occurred during vhdl-2008 alias to hierachy

wm2015email, 2023-11-02:
**Description**

Reporting bug as requested...

```txt
ghdl -a -fsynopsys --std=08 --work=work C:\Users\fn269e\sandbox\hdlc\test1\test1.vhd

ghdl -e -fsynopsys --std=08 test1

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0 (2.0.0.r1417.g7de967c51) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: x86_64-w64-mingw32
C:\Users\fn269e\sandbox\hdlc\test1\
Command line:
C:\Users\fn269e\sandbox\rincon_tools\GHDL\bin\ghdl.exe -e -fsynopsys --std=08 test1
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-chap4.adb:1724 access check failed
Load address: 0x7ff6c38a0000
Call stack traceback locations:
0x7ff6c3cb06a3 0x7ff6c3cb174e 0x7ff6c3cb343f 0x7ff6c3cd6260 0x7ff6c3c98b0b 0x7ff6c3c8bf4a 0x7ff6c3c9d74c 0x7ff6c3c97b48 0x7ff6c3ce1c8e 0x7ff6c3c1cfd4 0x7ff6c3c074f7 0x7ff6c3ce320b 0x7ff6c38a36a0 0x7ff6c38a12ec 0x7ff6c38a1404 0x7ff87c9d7342 0x7ff87dfc26af
******************************************************************
```

**Expected behaviour**
Not to get a core dump.

**How to reproduce?**

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity bfm is
end entity;

architecture beh of bfm is
    signal data :unsigned(7 downto 0);
    signal en   :std_logic;
begin

end architecture;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test1 is
end entity;

architecture beh of test1 is

    alias probe_data is <<signal .test1.ibfm.data  :unsigned(7 downto 0)>>;
    alias probe_en   is <<signal .test1.ibfm.en    :std_logic>>;

begin

    ibfm: entity work.bfm;
	
    -- The testbench process
    process
    begin
        probe_en   <= '1';
        probe_data <= (others => '1');

        wait for 100 ns;
        probe_en   <= '0';
        probe_data <= (others => '0');
        
        wait;
    end process;
	
end architecture;
```

```bash
ghdl -a -fsynopsys --std=08 --work=work C:\Users\fn269e\sandbox\hdlc\test1\test1.vhd
ghdl -e -fsynopsys --std=08 test1
```

**Context**

see above.  this is running on windows 11. using GHDL 3.0.0 (2.0.0.r1417.g7de967c51

**Additional context**
see above

tgingold, 2023-11-05:
This looks valid but yet not handled by ghdl.

diogratia, 2023-11-05:
Jim pointed to [LCS-2016-I05](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_I05) several [years ago](https://stackoverflow.com/posts/comments/83763761). The question and answer the comment addresses on Stackoverflow seems to cover this case which articulated a bit clearer could be restated as the alias can't be elaborated until the target has been elaborated.

14.4.2.6 Alias declarations

>Elaboration of an alias declaration consists of the elaboration of the subtype indication to establish the subtype associated with the alias, followed by the creation of the alias as an alternative name for the named entity. The creation of an alias for a composite object involves a check that the subtype associated with the alias includes a matching element for each element of the named object. It is an error if this check fails.

GHDL appears to crash under various conditions when the alias target is evaluated and is not yet elaborated such as here. Such a crash likely represents a lack of exception handling. 

Here any alias would be required to be declared after the instantiated design unit its target external name references an element of. Here in the testbench process. I'm on an iPad at the moment or I'd demonstrate.

(And you wouldn't believe the contortions finding the two linked references on an iPad with only a vague idea of the LCS number but remembering it was Jim commenting on it. I also don't have ready access to IEEE Std 1076-2019 on my iPad to determine if the approved LCS was actually implemented. Stopping and correcting that deficiency and the example has been updated with the LCS.)

The issue comes about because elaboration is done in order and an external name alias target is not available until ibfm is elaborated in the failing code.

tgingold, 2023-11-05:
Hum, according to 14.2:

During elaboration of a design hierarchy, if an external name or alias of an external name appears in a
declaration or statement being elaborated, then in the following cases, the declaration of the object denoted
by the external name or alias shall have been previously elaborated:
—If the external name or alias is a primary or a prefix of a primary in an expression that is evaluated
during elaboration of the design hierarchy, when the primary is read during evaluation of the
expression.
—If the external name or alias, or a name in which the external name or alias is a prefix, is associated as
an actual in an association element in a port map aspect, when the association element is elaborated.

Does the elaboration of an alias declaration evaluate the name ?  Not yet clear to me.

diogratia, 2023-11-06:
Note you can get the bfm and test1 entity/architecture pairs to analyze, elaborate and simulate by adding the reserved word force to the assignments in test1 architecture architecture for a recent llvm version (GHDL 4.0.0-dev (3.0.0.r636.g1ec1eb6d9), a nightly build) if you also convert probe_data to natural. The same commit's mcode version nightly requires the alias declarations appears as process declarative items instead of block declarative items (seemingly in line with Jim's [LCS](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_I05)). 

Both versions ghdl analyzed, elaborated and ran.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
-- use ieee.numeric_std.all;

entity bfm is
end entity;

architecture beh of bfm is
    signal data :natural;
    signal en   :std_logic;
begin

end architecture;

library ieee;
use ieee.std_logic_1164.all;
-- use ieee.numeric_std.all;

entity test1 is
end entity;

architecture beh of test1 is
begin

    ibfm: entity work.bfm;

    -- The testbench process
    process
        alias probe_data is <<signal .test1.ibfm.data  :natural>>;
        alias probe_en   is <<signal .test1.ibfm.en    :std_logic>>;
    begin
        probe_en   <=  force '1';
        probe_data <=  force natural'high;

        wait for 100 ns;
        probe_en   <=  force '0';
        probe_data <= force 0;
        wait;

    end process;

end architecture;
```
The alias declarations here as process declarative items allowing the mcode version to succeed. The issue with probe_data as unsigned may or may not be related to the above quoted semantics from 14.4.2.6.

I ran a matrix of different versions of test1 to attempt to understand how an accidentally contrived test case using force for a Stackoverflow question that worked differed from bfm and test1.

```vhdl
entity tb is
end entity;

entity device_under_test is
end entity;

architecture external_events of device_under_test is
    signal alpha_value:     natural;
    signal alpha_update:    natural;
    signal alpha_val_in:    natural;
    signal countv:          boolean := TRUE; -- First falling edge at
    signal countu:          boolean := TRUE; -- 500 sn CHANGED
begin
VAL_INPUT:
    process (alpha_value)
    begin
        alpha_val_in <= alpha_value + 1;
    end process;

COUNTERS_EVENTS:             -- CHANGED event 'clock' generators
    process
    begin
        wait for 500 ns;
        countv <= not countv;
        countu <= not countu; -- wnat the TRUE to FALSE transition at 500 ns
    end process;
COUNT_UPDATE:
    process
    begin
        wait until rising_edge(countu);
        alpha_update <= alpha_update + 1;
    end process;

COUNT_VALUE:
    process
    begin
        wait until rising_edge(countv);
        alpha_value <= alpha_val_in;
    end process;
    
end architecture;

architecture external_events of tb is
    signal alpha_force:     boolean;
begin
DUT:
    entity work.device_under_test;
    
FORCE_PROCESS:   -- Force process
  process
      alias alpha_value is <<signal .tb.dut.alpha_value: natural>>;
      alias alpha_val_in is <<signal .tb.dut.alpha_val_in: natural>>;
      alias countv is <<signal .tb.dut.countv: boolean>>;
      alias countu is <<signal .tb.dut.countu: boolean>>; -- 'clock' ADDED
  begin
    -- Force beefore increment
    wait until falling_edge(countu);
    alpha_val_in <= force 13;  -- From 1 to 2 us
    alpha_force <= true;
    countv <= force FALSE;
    wait for 50 ns;
    countv <= force  TRUE;
    wait for 50 ns;
    countv <= release;
    alpha_val_in <= release;
    alpha_force <= false;
    -- Force over increment
    wait until falling_edge(countu);
    alpha_val_in <= force 41;  -- From 5 to 7 us
    alpha_force <= true;
    wait until falling_edge(countu);
    alpha_val_in <= release;
    alpha_force <= false;
    wait;
  end process;

end architecture;
```

```none
~/opt/share/3.0.0.r636.g1ec1eb6d9/ghdl-macos-11-mcode/bin/ghdl -a  --std=08 tb.vhdl
~/opt/share/3.0.0.r636.g1ec1eb6d9/ghdl-macos-11-mcode/bin/ghdl -e  --std=08 tb
~/opt/share/3.0.0.r636.g1ec1eb6d9/ghdl-macos-11-mcode/bin/ghdl -r --std=08 tb  --wave=tb.ghw --stop-time=4us
ghdl:info: simulation stopped by --stop-time @4us
/usr/local/bin/gtkwave tb.gtkw &
%: 
```
[image]

The llvm back end version did not record any waveform transactions to the tb.ghw dump file (length 307 bytes), while the mcode version did (length 1070).

The message here is "There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE" seems applicable to the use of external names and their aliases currently in ghdl as well as a lack of portability between ghdl mcode and llvm versions.

commit 1c3a7353 (2023-11-08): translate: handle drivers for external signal names.

Fix #2528
closed 2023-11-08 by tgingold