ghdl/ghdl #2530: Different behaviour using variable and signal in simulus sim

albydnc, 2023-11-06:
**Description**
I am using a file to drive some signals on a bus. The files are read into a table and then the table is sent to the bus using a procedure.
In GHDL, I see a different behaviour if the table is a signal or a variable, while in both QuestaSim and RivieraPro they behave the same way.
- Correct behaviour: output is driven
- Observed behaviour: output is uninitialized

**Expected behaviour**
I expect no difference in output between declaring the `data_table` as variable or as signal.

**How to reproduce?**
The MWE shows the two behaviour. First is the case with variable `data_table_var`, which works in the right way. Second is the signal case `data_table_sig`, which shows the wrong behaviour.

```vhd :file: file_driver.vhd
library ieee;
context ieee.ieee_std_context;
use ieee.math_real.all;
use ieee.std_logic_textio.all;

library std;
use std.textio.all;
use std.env.finish;

entity file_driver is
end entity file_driver;

architecture sim of file_driver is
  constant FILE_NAME    : string    := "data.txt";
  constant CLOCK_PERIOD : time      := 4 ns;
  signal clk            : std_logic := '1';
  signal bus_data       : std_logic_vector(31 downto 0);
  signal bus_addr       : std_logic_vector(24 downto 0);
  signal bus_wren       : std_logic;

  type data_array is array(0 to 127) of std_logic_vector(31 downto 0);
  signal data_table_sig : data_array;

begin

  clk <= not clk after CLOCK_PERIOD /2;

  SEQUENCER_PROC : process
    file infile             : text;
    variable read_buffer    : line;
    variable line_content   : string(1 to 32);
    variable line_read      : std_logic_vector(31 downto 0);
    variable table_address  : integer range 0 to 127 := 0;
    variable data_table_var : data_array;

    procedure write_table (DATA_ADDR : std_logic_vector(24 downto 0);
    DATA_TBL                         : data_array;
    NVALUES                          : integer) is
  begin
    wait until rising_edge(clk);
    bus_addr <= (others => 'Z');
    bus_wren <= '0';
    bus_data <= (others => 'Z');
    --
    -- load the table of data
    for index_table in 0 to NVALUES - 1 loop
      --
      wait until rising_edge(clk);
      bus_addr <= std_logic_vector(unsigned(DATA_ADDR) + to_unsigned(index_table * 4, 25));
      bus_wren <= '1';
      bus_data <= DATA_TBL(index_table);
      --
      wait until rising_edge(clk);
      bus_addr <= (others => 'Z');
      bus_wren <= '0';
      bus_data <= (others => 'Z');
    end loop;

    wait until rising_edge(clk);
    bus_wren <= '0';
    bus_addr <= (others => 'Z');
    bus_data <= (others => 'Z');
  end write_table;

begin

  --WRITING TO BUS USING VARIABLE -- correct behaviour
  file_open(infile, FILE_NAME, read_mode);
  data_table_var := (others => (others => '0'));
  while (not(endfile(infile))) loop
    readline(infile, read_buffer);
    read(read_buffer, line_content);
    for index_word in 0 to 31 loop
      if line_content(index_word + 1) = '1' then
        line_read(index_word) := '1';
      else
        line_read(index_word) := '0';
      end if;
    end loop;
    data_table_var(table_address) := line_read;
    table_address               := table_address + 1;
  end loop;

  write_table(25X"0A00", data_table_var, table_address);
  file_close(infile);
  wait for 50 ns;

  --WRITING TO BUS USING SIGNAL -- wrong behaviour
  file_open(infile, FILE_NAME, read_mode);
  table_address := 0;
  data_table_sig <= (others => (others => '0'));
  while (not(endfile(infile))) loop
    readline(infile, read_buffer);
    read(read_buffer, line_content);
    for index_word in 0 to 31 loop
      if line_content(index_word + 1) = '1' then
        line_read(index_word) := '1';
      else
        line_read(index_word) := '0';
      end if;
    end loop;
    data_table_sig(table_address) <= line_read;
    table_address := table_address + 1;
  end loop;

  write_table(25X"0A00", data_table_sig, table_address);
  file_close(infile);
  wait for 50 ns;
  finish;
end process;

end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a file_driver.vhd
ghdl --elab-run file_driver --wave=file_driver.vhd
```
Here is the example data stimulus file
[data.txt](https://github.com/ghdl/ghdl/files/13265412/data.txt)

**Context**
- OS: Docker on CentOS7
- Origin: image `ghdl/vunit` all flavors (mcode, llvm, gcc)

tgingold, 2023-11-06:
For signals, you need to add a `wait for 0 ns` before the call to write_table.  Otherwise `data_table_sig` is not updated.
Looks like both QuestaSim and Riviera wrongly optimize your design.

albydnc, 2023-11-06:
Thanks for the clarification!

closed 2023-11-06 by albydnc