ghdl/ghdl #2531: Missing operators for real * integer and real / integer

kraigher, 2023-11-06:
The VHDL standard defines:
- `universal_real * universal_integer`
- `universal_integer * universal_real`
- `universal_real / universal_integer`

```vhdl
    variable x : real := 2.0 / 2;  
    variable y : real := 2.0 * 2;  
    variable z : real := 2 * 2.0;  
```

These are missing from GHDL, I tried the latest release just now.

tgingold, 2023-11-06:
It's implemented, but your typing is not correct.  Compare with:
```vhdl
entity repro2 is
end;

architecture behav of repro2 is
begin
  process
    variable x : real := real (2.0 / 2);  
    variable y : real := real (2.0 * 2);  
    variable z : real := real (2 * 2.0);  
  begin
    wait;
  end process;
end;
```

tgingold, 2023-11-07:
Typing is tricky: if you allow more implicit conversions, ambiguities due to overloading will appear very quickly.

kraigher, 2023-11-07:
Yes overloading complicates the tools. In this case I see that the existing rules for doing an implicit concerning could apply. The standard says a conversion should be done only if the code without conversion wouldnt work. And only in the inner most context.

tgingold, 2023-11-07:
No, you can only convert literals, attributes or division of two physical values (see LRM 9.3.6)

kraigher, 2023-11-07:
Yes you probably right. It must have read "operator" when the standard said "operand" the last time I read it. Thanks for the explaination and sorry to bother you!

closed 2023-11-07 by kraigher

tgingold, 2023-11-07:
I first copied main.hex to `sim/`, and then:

```
cd sim
sh ghdl_setup.sh
sh ghdl_run.sh
```

commit c91a5618 (2023-11-08): testsuite/gna: add a reproducer for #2531

close #2531