ghdl/ghdl #2532: ERROR | Using custom simulation arguments

tthompsongmsi, 2023-11-06:
**Description**
Trying to run neorv32-riscof verification test. Recursively cloned the repo. I get an ERROR using custom simulation arguments and storage error raised.

**Expected behaviour**
Test should be functional from the get go.  

**How to reproduce?**
recursively clone neorv32-riscof 

installed using main.yml file as guideline:
https://github.com/stnolting/neorv32-riscof/blob/main/.github/workflows/main.yml

**Context**
Please, provide the following information:

- OS: ubuntu 20.04 in podman container
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [X] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

   ERROR | Using custom simulation arguments: -gMEM_SIZE=52056 -gRISCV_B=true
   ERROR | 
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r660.g3e6a6909c) [Dunoon edition]
Compiled with GNAT Version: 9.4.0
Target: x86_64-linux-gnu

Command line:
ghdl -r --std=08 --work=neorv32 neorv32_riscof_tb -gMEM_FILE=./main.hex --max-stack-alloc=0 --ieee-asserts=disable --assert-level=error -gMEM_SIZE=52056 -gRISCV_B=true --stop-time=4ms
Exception STORAGE_ERROR raised
Exception information:
raised STORAGE_ERROR : stack overflow or erroneous memory access
******************************************************************

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2023-11-07:
It would be nice to have a reproducer: a small shell script to analyze all the vhdl files and start the simulation, as well as any generated files.

tgingold, 2023-11-07:
Thanks.  I didn't notice there were shell scripts in sim/
But, I also need `main.hex`.

tthompsongmsi, 2023-11-07:
oh that's strange, main.hex is in the sim directory but it's an empty file.

tgingold, 2023-11-07:
Ok, but I cannot reproduce the crash.  I get:
```
/neorv32_riscof_tb.vhd:134:3:@0ms:(assertion warning): TB: actual memory size = 65536 bytes
../neorv32/rtl/core/neorv32_top.vhd:343:5:@0ms:(assertion note): The NEORV32 RISC-V Processor by Stephan Nolting,version 0x01090007, github.com/stnolting/neorv32
../neorv32/rtl/core/neorv32_top.vhd:349:5:@0ms:(assertion note): NEORV32 Processor Configuration: I-CACHE D-CACHE WISHBONE SYSINFO 
../neorv32/rtl/core/neorv32_cpu.vhd:142:3:@0ms:(assertion note): NEORV32 CPU Configuration: RV32IBU_Zicsr_Zicntr_Zifencei
../neorv32/rtl/core/neorv32_cpu.vhd:163:3:@0ms:(assertion warning): NEORV32 CPU WARNING! Assuming this is a simulation.
../neorv32/rtl/core/neorv32_cpu_cp_bitmanip.vhd:173:3:@0ms:(assertion note): NEORV32 CPU: Implementing bit-manipulation (B) sub-extensions Zba Zbb Zbc Zbs 
../neorv32/rtl/core/neorv32_wishbone.vhd:118:3:@0ms:(assertion note): NEORV32 PROCESSOR CONFIG NOTE: Ext. Bus Interface - PIPELINED Wishbone protocol, auto-timeout (8 cycles), LITTLE-endian byte order, ASYNC RX, ASYNC TX
./neorv32_riscof_tb.vhd:246:13:@34055ns:(assertion note): Finishing simulation.
simulation finished @34055ns
```

I use a slightly more recent version that yours, but I doubt it will fix the issue.

tthompsongmsi, 2023-11-07:
oh weird, I'm not sure what I'm doing wrong then. I execute the ./run.sh file and get all those errors as shown above. may I ask how you're running the the simulation?

tgingold, 2023-11-11:
I have just cloned the neorv32-riscof reposritory, copy your main.hex file in sim.
I run ghdl_setup.sh and ghdl_run.sh in sim.

tthompsongmsi, 2023-11-11:
Thank you for your help. I've tried this on my laptop and in several
different Linux containers and it still doesn't work for me. Must be
something with my setup.

Thanks,

Trevor Thompson
Member of Technical Staff
Green Mountain Semiconductor
(802)503-7125

tgingold, 2023-11-12:
Let's try to investigate a little bit more,
Does it work during CI ?

What is your version of ghdl (output of `ghdl -v`) ?

If you run within gdb, you should get a signal.  What is the backtrace ?
(Try `gdb --args ghdl -r ...`, then inside gdb, `run` and when it stops `where`).

tthompsongmsi, 2023-11-13:
I made a linux container with Ubuntu 20.04 and freshly cloned the
neorv32-riscof repo. I then went through and installed all the
prerequisites and ran the commands to ensure they're installed correctly.
Here's the output from those:

riscv32-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv32-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/trevor/Documents/riscof/Documents/neorv32-riscof/riscv-gcc/bin/../libexec/gcc/riscv32-unknown-elf/13.2.0/lto-wrapper
Target: riscv32-unknown-elf
Configured with:
/home/stnolting/tmp/rvgcc/riscv-gnu-toolchain/gcc/configure
--target=riscv32-unknown-elf --prefix=/home/stnolting/tmp/riscv/bin
--disable-shared --disable-threads --enable-languages=c,c++
--with-pkgversion= --with-system-zlib --enable-tls --with-newlib
--with-sysroot=/home/stnolting/tmp/riscv/bin/riscv32-unknown-elf
--with-native-system-header-dir=/include --disable-libmudflap
--disable-libssp --disable-libquadmath --disable-libgomp --disable-nls
--disable-tm-clone-registry --src=.././gcc --disable-multilib
--with-abi=ilp32 --with-arch=rv32i --with-tune=rocket
--with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 13.2.0 ()

 riscv_sim_RV32 -h
Usage: riscv_sim_RV32 [options] <elf_file>
-d --enable-dirty-update
-m --enable-misaligned
-P --enable-pmp
-N --enable-next
-z --ram-size
-C --disable-compressed
-I --disable-writable-misa
-F --disable-fdext
-i --mtval-has-illegal-inst-bits
-b --device-tree-blob
-t --terminal-log
-p --show-times
-a --report-arch
-T --test-signature
-g --signature-granularity
-h --help
-v --trace
-V --no-trace
-l --inst-limit
-x --enable-zfinx

ghdl -v
GHDL 4.0.0-dev (3.0.0.r671.g11a721ba1) [Dunoon edition]
 Compiled with GNAT Version: 9.4.0
 static elaboration, mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is
NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

riscof --version
RISC-V Architectural Test Framework., version 1.25.3

I placed the main.hex file I sent you in the sim directory for this new
linux container I made. I ran ghdl_setup.sh then ghdl_run.sh and got the
following message:

./ghdl_setup.sh
./ghdl_run.sh
Using custom simulation arguments:

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r671.g11a721ba1) [Dunoon edition]
Compiled with GNAT Version: 9.4.0
Target: x86_64-linux-gnu
/home/trevor/Documents/riscof/Documents/neorv32-riscof/sim/
Command line:
ghdl -r --std=08 --work=neorv32 neorv32_riscof_tb -gMEM_FILE=./main.hex
--max-stack-alloc=0 --ieee-asserts=disable --assert-level=error
--stop-time=4ms
Exception STORAGE_ERROR raised
Exception information:
raised STORAGE_ERROR : stack overflow or erroneous memory access
******************************************************************

*Does it work during CI ?*
Please excuse my ignorance, still new to github and stuff. I'm not sure
what CI is.

*What is your version of ghdl (output of ghdl -v) ?*
GHDL 4.0.0-dev (3.0.0.r671.g11a721ba1) [Dunoon edition]
 Compiled with GNAT Version: 9.4.0
 static elaboration, mcode code generator
Written by Tristan Gingold.

*If you run within gdb, you should get a signal. What is the backtrace ?*
*(Try gdb --args ghdl -r ..., then inside gdb, run and when it
stops where).*

(gdb) run
Starting program: /home/trevor/Documents/riscof/.local/bin/ghdl -r --std=08
--work=neorv32 neorv32_riscof_tb -gMEM_FILE=main.hex --max-stack-alloc=0
--iee-asserts=disable --assert-level=error --stop-time=4ms
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x0000555555638060 in binary_file.write_64 (sect=0x55555607b880, pc=8192,
b=0)
    at
/home/trevor/Documents/riscof/Downloads/ghdl/src/ortho/mcode/binary_file.adb:406
406      Sect.Data (Pc + 0 .. Pc + 7) := To_B8 (B);

Thanks,

*Trevor Thompson*
Member of Technical Staff
Green Mountain Semiconductor, Inc.
(802) 503-7125

tgingold, 2023-11-13:
That's interesting.  Can you post the full backtrace ?

tthompsongmsi, 2023-11-14:
I'm not sure if I'm running gdb correctly. I enter the following in the
terminal:
gdb --args ghdl -r --std=08 --work=neorv32 neorv32_riscof_tb \
  -gMEM_FILE=main.hex \
  --max-stack-alloc=0 \
  --ieee-asserts=disable \
  --assert-level=error \
  --stop-time=4ms

Once inside gdb I enter run at the prompt which gives me this error:
*command line*:1:1:error: unit entity "neorv32_riscof_tb" has not been
analyzed
[Inferior 1 (process 14721) exited with code 01]

I then enter backtrace -full and get this:
No stack.

I also tried starting gdb as follows:
gdb -e ghdl_setup.sh -e ghdl_run.sh

Once inside I ran the following:
(gdb) shell ./ghdl_setup.sh
(gdb) shell ./ghdl_run.sh
Using custom simulation arguments:

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r671.g11a721ba1) [Dunoon edition]
Compiled with GNAT Version: 9.4.0
Target: x86_64-linux-gnu
/home/trevor/Documents/riscof/Documents/neorv32-riscof/sim/
Command line:
ghdl -r --std=08 --work=neorv32 neorv32_riscof_tb -gMEM_FILE=./main.hex
--max-stack-alloc=0 --ieee-asserts=disable --assert-level=error
--stop-time=4ms
Exception STORAGE_ERROR raised
Exception information:
raised STORAGE_ERROR : stack overflow or erroneous memory access
******************************************************************
(gdb) bt
No stack.
(gdb) bt -full
No stack.
(gdb)

Thanks,

*Trevor Thompson*
Member of Technical Staff
Green Mountain Semiconductor, Inc.
(802) 503-7125

tthompsongmsi, 2023-11-14:
Sorry I must have been doing something wrong. Here's the full backtrace:
(gdb) run
Starting program: /home/trevor/Documents/riscof/.local/bin/ghdl -r --std=08
--work=neorv32 neorv32_riscof_tb -gMEM_FILE=main.hex --max-stack-alloc=0
--ieee-asserts=disable --assert-level=error --stop-time=4ms
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x0000555555638060 in binary_file.write_64 (sect=0x55555607b880, pc=8192,
b=0)
    at
/home/trevor/Documents/riscof/Downloads/ghdl/src/ortho/mcode/binary_file.adb:406
406      Sect.Data (Pc + 0 .. Pc + 7) := To_B8 (B);
(gdb) bt -full
#0  0x0000555555638060 in binary_file.write_64 (sect=0x55555607b880,
pc=8192,
    b=0)
    at
/home/trevor/Documents/riscof/Downloads/ghdl/src/ortho/mcode/binary_file.adb:406
No locals.
#1  0x000055555563857e in binary_file.gen_64 (b=0)
    at
/home/trevor/Documents/riscof/Downloads/ghdl/src/ortho/mcode/binary_file.adb:478
No locals.
#2  0x000055555563a5a4 in binary_file.gen_addr (offset=0)
    at
/home/trevor/Documents/riscof/Downloads/ghdl/src/ortho/mcode/binary_file.adb:773
No locals.
#3  0x000055555563a5d5 in binary_file.gen_abs (sym=1759, offset=0)
    at
/home/trevor/Documents/riscof/Downloads/ghdl/src/ortho/mcode/binary_file.adb:785
No locals.
#4  0x0000555555661da3 in ortho_code.x86.emits.emit_const (val=7175)
    at
/home/trevor/Documents/riscof/Downloads/ghdl/src/ortho/mcode/ortho_code-x86-emits.adb:3294
        decl = 1547
        off = 0
        h = 21845
--Type <RET> for more, q to quit, c to continue without paging--
        l = 1432519081
#5  0x0000555555661ea4 in ortho_code.x86.emits.emit_const (val=7173)
    at
/home/trevor/Documents/riscof/Downloads/ghdl/src/ortho/mcode/ortho_code-x86-emits.adb:3309
        i = 0
        e = 7175
        <R204b> = 1
        <ortho_code__x86__emits__emit_const__B_25__L_26__T205b___U> = 1
        h = 21845
        l = 1432577906
#6  0x0000555555662000 in ortho_code.x86.emits.emit_init_value (decl=1548,
    val=7173)
    at
/home/trevor/Documents/riscof/Downloads/ghdl/src/ortho/mcode/ortho_code-x86-emits.adb:3343
        sym = 1760
        dtype = 196
#7  0x0000555555662abf in ortho_code.x86.abi.expand_init_value (decl=1548,
    val=7173)
    at
/home/trevor/Documents/riscof/Downloads/ghdl/src/ortho/mcode/ortho_code-x86-abi.adb:233
No locals.
#8  0x0000555555658c39 in ortho_code.decls.new_init_value (decl=1548,
val=7173)
    at
/home/trevor/Documents/riscof/Downloads/ghdl/src/ortho/mcode/ortho_code-decls.adb:376
No locals.
#9  0x000055555566b78b in ortho_mcode.finish_init_value (decl=1548,
val=7173)
    at
/home/trevor/Documents/riscof/Downloads/ghdl/src/ortho/mcode/ortho_mcode.adb:145
[... 167 lines cut ...]
        cmd = 0x555555daa2f0
        first_arg = 4
--Type <RET> for more, q to quit, c to continue without paging--
        action_success = 85
        <R375b> = 9
        <R474b> = 91
        <R374b> = 1
        <ghdlmain__main__L_13__T376b___L> = 1
        <ghdlmain__main__L_13__T376b___U> = 9
        <R475b> = 110
        <ghdlmain__main__L_19__T476b___L> = 21845
        <ghdlmain__main__L_19__T476b___U> = 1437833599
        args = 0x555555d0af08
        arg_index = 10
#31 0x0000555555b08b4a in ghdl_jit ()
    at
/home/trevor/Documents/riscof/Downloads/ghdl/src/ghdldrv/ghdl_jit.adb:39
No locals.

Thanks,

*Trevor Thompson*
Member of Technical Staff
Green Mountain Semiconductor, Inc.
(802) 503-7125

tgingold, 2023-11-14:
Thank you.  Instead of the backtrace, you can do: `print cur_sect.all` ?
I am trying to understand the reason of the crash...

tthompsongmsi, 2023-11-15:
(gdb) print cur_sect.all
$1 = (next => 0x55555607b930, flags => 2, name => 0x55555607b918, link =>
0x0, align => 2, esize => 0,
  pc => 8192, insn_pc => 0, data => 0xffffffffffffffff, data_max => 16384,
first_reloc => 0x555555d165a0,
  last_reloc => 0x5555562b75a0, nbr_relocs => 507, number => 0, vaddr => 0,
img_off => 0, seg => (
    base => (system.address) 0xffffffffffffffff, size => 16384))

Question for you. These are the results when I execute the following
command:
gdb --args ghdl -r --std=08 --work=neorv32 neorv32_riscof_tb \
  -gMEM_FILE=main.hex \
  --max-stack-alloc=0 \
  --ieee-asserts=disable \
  --assert-level=error \
  --stop-time=4ms

This is the run command from inside the ghdl_run.sh file. Do I need to be
doing something with the ghdl_setup.sh file when debugging with gdb?

Thanks,

*Trevor Thompson*
Member of Technical Staff
Green Mountain Semiconductor, Inc.
(802) 503-7125

tgingold, 2023-11-16:
Thank you very much.  I have the impression the memory limit is very low inside the container.
Can you try: `strace ghdl ...` ?
It displays all the system calls.  Near the end, I suppose there is a call to `mremap` which fails.

It would also be interesting to know the ram size limit using `ulimit -a`.

The output is:
```
$ $GHDL -r --std=08 --work=neorv32 neorv32_riscof_tb -gMEM_FILE=./main.hex --max-stack-alloc=0 --ieee-asserts=disable --assert-level=error --stop-time=4ms
./neorv32_riscof_tb.vhd:134:3:@0ms:(assertion warning): TB: actual memory size = 8192 bytes
../neorv32/rtl/core/neorv32_top.vhd:343:5:@0ms:(assertion note): The NEORV32 RISC-V Processor by Stephan Nolting,version 0x01090007, github.com/stnolting/neorv32
../neorv32/rtl/core/neorv32_top.vhd:349:5:@0ms:(assertion note): NEORV32 Processor Configuration: I-CACHE D-CACHE WISHBONE SYSINFO 
../neorv32/rtl/core/neorv32_cpu.vhd:142:3:@0ms:(assertion note): NEORV32 CPU Configuration: RV32IU_Zicsr_Zicntr_Zifencei
../neorv32/rtl/core/neorv32_cpu.vhd:163:3:@0ms:(assertion warning): NEORV32 CPU WARNING! Assuming this is a simulation.
../neorv32/rtl/core/neorv32_wishbone.vhd:118:3:@0ms:(assertion note): NEORV32 PROCESSOR CONFIG NOTE: Ext. Bus Interface - PIPELINED Wishbone protocol, auto-timeout (8 cycles), LITTLE-endian byte order, ASYNC RX, ASYNC TX
./neorv32_riscof_tb.vhd:246:13:@34055ns:(assertion note): Finishing simulation.
simulation finished @34055ns
```

tthompsongmsi, 2023-11-16:
Output from ulimit -a :
$ ulimit -a
core file size          (blocks, -c) 0
data seg size           (kbytes, -d) unlimited
scheduling priority             (-e) 0
file size               (blocks, -f) unlimited
pending signals                 (-i) 61637
max locked memory       (kbytes, -l) 8192
max memory size         (kbytes, -m) unlimited
open files                      (-n) 524288
pipe size            (512 bytes, -p) 8
POSIX message queues     (bytes, -q) 819200
real-time priority              (-r) 0
stack size              (kbytes, -s) 8192
cpu time               (seconds, -t) unlimited
max user processes              (-u) 61637
virtual memory          (kbytes, -v) unlimited
file locks                      (-x) unlimited

Output of strace:
$ strace ghdl -r --std=08 --work=neorv32 neorv32_riscof_tb \
>   -gMEM_FILE=main.hex \
>   --max-stack-alloc=0 \
>   --ieee-asserts=disable \
>   --assert-level=error \
>   --stop-time=4ms
execve("/home/trevor/Documents/riscof/.local/bin/ghdl", ["ghdl", "-r",
"--std=08", "--work=neorv32", "neorv32_riscof_tb", "-gMEM_FILE=main.hex",
"--max-stack-alloc=0", "--ieee-asserts=disable", "--assert-level=error",
"--stop-time=4ms"], 0x7fff4faa9ac8 /* 47 vars */) = 0
brk(NULL)                               = 0x55c85d1aa000
arch_prctl(0x3001 /* ARCH_??? */, 0x7ffed0e6d550) = -1 EINVAL (Invalid
argument)
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or
directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=25146, ...}) = 0
mmap(NULL, 25146, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7fc7827a0000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0
\22\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=18848, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) =
0x7fc78279e000
mmap(NULL, 20752, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) =
0x7fc782798000
mmap(0x7fc782799000, 8192, PROT_READ|PROT_EXEC,
MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1000) = 0x7fc782799000
mmap(0x7fc78279b000, 4096, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE,
3, 0x3000) = 0x7fc78279b000
mmap(0x7fc78279c000, 8192, PROT_READ|PROT_WRITE,
MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0x7fc78279c000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libm.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3,
"\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\300\323\0\0\0\0\0\0"...,
832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=1369384, ...}) = 0
mmap(NULL, 1368336, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) =
0x7fc782649000
mmap(0x7fc782656000, 684032, PROT_READ|PROT_EXEC,
MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xd000) = 0x7fc782656000
mmap(0x7fc7826fd000, 626688, PROT_READ,
MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xb4000) = 0x7fc7826fd000
mmap(0x7fc782796000, 8192, PROT_READ|PROT_WRITE,
MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x14c000) = 0x7fc782796000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libz.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3,
[... 1027 lines cut ...]
write(2, ":", 1:)                        = 1
write(2, "1", 11)                        = 1
write(2, ":", 1:)                        = 1
write(2, "1", 11)                        = 1
write(2, ":", 1:)                        = 1
write(2, "\33[1;31m", 7)                = 7
write(2, "error:", 6error:)                   = 6
write(2, "\33[0;1m", 6)                 = 6
write(2, " ", 1 )                        = 1
write(2, "unit ", 5unit )                    = 5
write(2, "entity \"neorv32_riscof_tb\"", 26entity "neorv32_riscof_tb") = 26
write(2, " has not been analyzed", 22 has not been analyzed)  = 22
write(2, "\33[0m", 4)                   = 4
write(2, "\n", 1
)                       = 1
futex(0x7fce8db6a1e0, FUTEX_WAKE_PRIVATE, 2147483647) = 0
exit_group(1)                           = ?
+++ exited with 1 +++

Thanks,

*Trevor Thompson*
Member of Technical Staff
Green Mountain Semiconductor, Inc.
(802) 503-7125

tgingold, 2023-11-17:
The crash is due to:
```
mremap(0x7fc7827a3000, 8192, 16384, MREMAP_MAYMOVE) = -1 ENOMEM (Cannot
allocate memory)
```
ie, it cannot allocate 8KB.
I have no idea why.  Are you running within a container ?
It could be interesting to run within gdb, and then when it crashes, using the `ps` system command to see the amount of allocated memory.  But there is something weird.  ghdl is not very resource intensive.

tthompsongmsi, 2023-11-17:
I've tried running this on my laptop and not in a container which runs
Rocky Linux 9 with no success. I could replicate these commands on my
laptop and see if it's the same.

Thanks,

Trevor Thompson
Member of Technical Staff
Green Mountain Semiconductor
(802)503-7125

tgingold, 2023-11-19:
What is your kernel version ?
There was a similar issue #2359

tthompsongmsi, 2023-11-19:
Currently it is 5.14... I can try upgrading to the latest stable kernel and
see if it works.

Thanks,

Trevor Thompson
Member of Technical Staff
Green Mountain Semiconductor
(802)503-7125

tthompsongmsi, 2023-11-20:
I updated to kernel 6.6.1 and this fixed my issue. The test now
successfully passes. Thanks for the help.
Thanks,

*Trevor Thompson*
Member of Technical Staff
Green Mountain Semiconductor, Inc.
(802) 503-7125

tgingold, 2023-11-20:
Not an easy issue!
Thank you for your patience.

closed 2023-11-25 by tgingold