ghdl/ghdl #2534: GHDL Bug occurred

andreasorio, 2023-11-09:
Dear Sirs

I apologize for me English

It is quite difficult to describe this issue: a block instance create the following message:

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r484.g3488cdeb3) [Dunoon edition]
Compiled with GNAT Version: 10.5.0
Target: x86_64-linux-gnu
/home/andsor/git/d-matrix-8t/Fracarro/vhdl/
Command line:
/usr/local/libexec/gcc/x86_64-pc-linux-gnu/12.3.0/ghdl1 --ghdl--work=fracarro --ghdl--workdir=fracarro --ghdl-fsynopsys --ghdl-fexplicit -P./xilinx-ise/ -P/usr/local/lib/ghdl/ieee/v93/ -P/usr/local/lib/ghdl/ -quiet -o fracarro/TS_AES_128.s TS_AES_128.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-nodes.adb:862
Call stack traceback locations:
0x7f2a6305f5a2 0x850bc5 0x8527cf 0x9c340f 0x9e9218 0x9b6e59 0x9b71a8 0xa057eb 0xa06fa1 0x9dcf75 0x9d696b 0x9dad4b 0x991f58 0x75df18 0xebb455 0x754a52 0x763653 0xa0a979 0x751ff7 0x7f2a62a29d8e 0x7f2a62a29e3e 0x756bf3 0xfffffffffffffffe
******************************************************************

Execution of /usr/local/libexec/gcc/x86_64-pc-linux-gnu/12.3.0/ghdl1 terminated by unhandled exception
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-nodes.adb:862
Call stack traceback locations:
0x7f2a6305f5a2 0x850bc5 0x8527cf 0x9c340f 0x9e9218 0x9b6e59 0x9b71a8 0xa057eb 0xa06fa1 0x9dcf75 0x9d696b 0x9dad4b 0x991f58 0x75df18 0xebb455 0x754a52 0x763653 0xa0a979 0x751ff7 0x7f2a62a29d8e 0x7f2a62a29e3e 0x756bf3 0xfffffffffffffffe
ghdl:error: compilation error

I have two different entities with the same interface (both of them compiled fine) : with the following code
....
  inst_RAM_channel_configuration: entity fracarro.RAM_DP_generic
    generic map (
      SIZE => 2**ID_INDEX_WIDTH,
      DATA_WIDTH => 5,
..

ghdl ends fine, while with 

....
  inst_RAM_channel_configuration: entity fracarro.RAM_DP_DIST_generic
    generic map (
      SIZE => 2**ID_INDEX_WIDTH,
      DATA_WIDTH => 5,
.....

ghdl crashes

Do you need more information?

Thank you in advance
Best regards

Andrea Sorio

tgingold, 2023-11-09:
I fear we need an MWE or a reproducer: a set of files and instructions 
to reproduce the crash from the set of files.

andreasorio, 2023-11-16:
Dear Tgingold

   Here follow the source code that generate the issue.

In the TS_AES_128.vhd file on line 670 if you put the wrong value empty string
to INIT_DATA gdhl crashes, while if you put the correct value (0 => 0),
ghdl works correctly and does not crash.
I invoke the ghdl in the following mode:
ghdl -a --work=fracarro --workdir=fracarro -P./xilinx-ise --ieee=synopsys -fexplicit TS_AES_128.vhd

Is this information sufficient?

   Thank you in advance

   Best Regards

   Andrea Sorio

   Da "tgingold" ***@***.***
   A "ghdl/ghdl" ***@***.***
   Cc "andreasorio" ***@***.***,"Author" ***@***.***
   Data Thu, 09 Nov 2023 09:57:56 -0800
   Oggetto Re: [ghdl/ghdl] GHDL Bug occurred (Issue #2534)

   I fear we need an MWE or a reproducer: a set of files and instructions
   to reproduce the crash from the set of files.

tgingold, 2023-11-16:
Yes, it is sufficient, but the source code is missing.

andreasorio, 2023-11-17:
Do you mean the source code about RAM_DP_generic.vhdl?

   Do you need other source files?

   Thank you in advance

   Andrea

   Da "tgingold" ***@***.***
   A "ghdl/ghdl" ***@***.***
   Cc "andreasorio" ***@***.***,"Author" ***@***.***
   Data Thu, 16 Nov 2023 09:59:52 -0800
   Oggetto Re: [ghdl/ghdl] GHDL Bug occurred (Issue #2534)

   Yes, it is sufficient, but the source code is missing.

tgingold, 2023-11-18:
On https://github.com/ghdl/ghdl/issues/2534  I cannot see any source file.

How did you file them ?

andreasorio, 2023-11-18:
I have attached them to last email.
I sent you an archive with library, did you receive it?
In this email i have attached two archives : crash_files.tar.xz and RAM_DP_generic.vhdl.tar.bz2
Thakyou
Best regards
Andrea

   Da "tgingold" ***@***.***
   A "ghdl/ghdl" ***@***.***
   Cc "andreasorio" ***@***.***,"Author" ***@***.***
   Data Sat, 18 Nov 2023 08:52:39 -0800
   Oggetto Re: [ghdl/ghdl] GHDL Bug occurred (Issue #2534)

   On #2534 I cannot see any source file.

   How did you file them ?

tgingold, 2023-11-19:
No, I haven't received anything.  Please, use the github web interface to attach your files.

andreasorio, 2023-11-19:
OK?
[crash_files.zip](https://github.com/ghdl/ghdl/files/13402742/crash_files.zip)

tgingold, 2023-11-19:
Yes, I got it.  But I need the sources; the object files are useless.

andreasorio, 2023-11-20:
The issue  occurs with incorrect assignment of a generic parameters (INIT_DATA => "") in the instance of "RAM_DP_generic" into TS_AES_128.vhd
All other blocks are instantiated correctly.
Why do I have to give you all the source codes?
In RAM_DP_generic there are no references to the other blocks.
Maybe you need to sign an NDA, I have to ask my PM.

tgingold, 2023-11-20:
The P1 is an automatic given name to an anonymous process.

If you cannot give the sources, you should create a reproducer: remove as much as possible from the existing sources while still keeping the issue.

tgingold, 2023-11-25:
I can reproduce the error, but not the crash:
```
> $GHDL -a -fsynopsys --work=fracarro types_pkg.vhd functions_pkg.vhd RAM_DP_generic.vhd TS_AES_128.vhd
TS_AES_128.vhd:64:21:error: can't associate string literal with generic "init_data"
      INIT_DATA =>  "",   -- *****  The empty string is the cause of ghdl crashing, with the right parameter (0 => 0) ghdl works
                    ^
RAM_DP_generic.vhd:28:5:error: (type of generic "init_data" is arrayof_natural)
```

tgingold, 2023-11-25:
But the version of ghdl you are using is ~200 commits before the head.  So probably the crash has been fixed.
Could you try with a more recent version.

andreasorio, 2023-11-26:
I just updated the ghdl source code to the latest version (r700) and it now handles the wrong vhdl code correctly.

I have another question, I don't know how to interpret the following run time error:
ghdl -r top_tb --stop-time=200us --wave=top_tb.ghw

./top_tb:error: bound check failure at txt_util.vhd:43
in process .top_tb(rtl).dut@top(rtl).multi_pid_filter_aes_1@multi_pid_filter_aes(rtl).ts_filter_remapper_counter_analyzer_multi_1@ts_filter_remapper_counter_analyzer_multi(ts_filter_remapper_counter_analyzer_multi_behavioral).inst_binary_encoder@binary_encoder(binary_encoder_behavioral).P1
./top_tb:error: error during elaboration

thankyou in advance
best regards
Andrea Sorio

tgingold, 2023-11-27:
Bound check failure means an index is out of array range, or an value is out of the range of a subtype.

Paebbels, 2024-12-19:
@tgingold can we close this issue?

closed 2024-12-22 by tgingold