ghdl/ghdl #2538: External Reference issue

stfarley, 2023-11-18:
Here is a simple TC that attempts to use an external reference.

With code I get an error.
src/work/tb.vhd:17:61:error: component instance "th" is not yet elaborated
alias clk                               is      <<signal.tb.th.clk:std_logic>>;
                                       
with LLVM and gcc versions the simulation just doesn't run.

[TC.zip](https://github.com/ghdl/ghdl/files/13401089/TC.zip)

tgingold, 2023-11-19:
It is not fully clear what are the vhdl 2008 rules, but to be safe, you can only use external names after the object has been elaborated.

So move the declaration of `clk` after the instantiation `th` (you need to move it inside a block).

stfarley, 2023-11-19:
Hi tgingold,

I gather from what you are suggesting is that the alias must be treated like a variable and placed in the declaration area of a process that follows the instantiation of the target. (This would the preclude using the alias in a sensitivity list)
This does seem to work. 
I was under the impression that you could declare the alias like a signal (In the architecture declaration area) but couldn't reference it until after the target component was instantiated.

avelure, 2023-11-21:
There is a workaround.
You could wrap your process in a block statement, then you can add the alias in the declaration part of the block and use the alias in the process sensitivity list.

commit db706e2b (2023-11-25): simul-vhdl_elab: fix handling of global external signal names.

Fix #2538
closed 2023-11-25 by tgingold