ghdl/ghdl #2543: Mingw mcode: std.env.stop causes stackfault

avelure, 2023-11-22:
**Description**
There seems to be an issue with stack unwinding with mcode backend on windows.

**Expected behaviour**
No stack fault.

**How to reproduce?**

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity;

architecture tb of test is
begin
   p_main: process
   begin
    std.env.stop;
    wait;
  end process p_main;
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent.vhd
ghdl --elab-run test
```

Tested on win10 with latest nightly ucrt64 mcode.

tgingold, 2023-11-23:
Yes, mcode on win64 doesn't support well errors and backtraces.  It needs a little bit of work to emit win64 unwind info.

avelure, 2023-11-27:
But would you expect an error or backtrace in this case?

tgingold, 2024-04-10:
Now fixed.

closed 2024-04-10 by tgingold