ghdl/ghdl #2544: GHDL does not support generic data type

Galactic-Explorer, 2023-11-23:
**Description**
The generic data type a feature of VHDL 2008 is not supported by GHDL

**Expected behaviour**
What you expected to happen, and what is happening instead.

```vhd :file: generic_fixed_delay.vhd
entity generic_fixed_delay is
    generic(
        NUM_DELAY_CYCLES : natural;     --! Number of delay cycles or registers. Setting this to 0 short-circuits the input with the output.
        type Data_Type_t                  --! The data type of the input and output.
    );
    port(
        clk    : in  std_logic;         --! Clock input
        input  : in  Data_Type_t;         --! Data input
        output : out Data_Type_t          --! Data output
    );
end entity generic_fixed_delay;

architecture rtl of generic_fixed_delay is
begin
    LOGIC_BLOCK_GEN_INST : if NUM_DELAY_CYCLES = 0 generate
        --! Act as short-circuit
        output <= input;
    elsif NUM_DELAY_CYCLES = 1 generate
        name : process is
        begin
            wait until rising_edge(clk);
            --! Output is registered
            output <= input;
        end process name;
    elsif NUM_DELAY_CYCLES > 1 generate
        type Data_Tpe_Arr_t is array (0 to NUM_DELAY_CYCLES - 2) of Data_Type_t;
        signal reg_arr : Data_Tpe_Arr_t;
    begin
        process is
        begin
            --! Delay of NUM_DELAY_CYCLES
            wait until rising_edge(clk);
            reg_arr   <= input & reg_arr(reg_arr'low to reg_arr'high - 1);
            output <= reg_arr(reg_arr'high);
        end process;
    end generate LOGIC_BLOCK_GEN_INST;
end architecture rtl;
```

**Context**
- OS: Win10
- Origin: Built from sources: `main`

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
/work/hsi640-fpga-design/src/evk-vhdl-utility-library/build/evk__vhdl-utility-generic-fixed-delay_0.0.1/sim-vunit
            Data_Type_t        => std_logic
                                           ^
/work/hsi640-fpga-design/src/evk-vhdl-utility-library/build/evk__vhdl-utility-generic-fixed-delay_0.0.1/sim-vunit
******************** GHDL Bug occurred ***************************
GHDL release: 4.0.0-dev (3.0.0.r696.g8c0683a5f) [Dunoon edition]
```

tgingold, 2023-11-23:
Yes, this is a known issue.   That's also the next feature to be implemented.

creiter64, 2023-11-24:
@tgingold I'm also exictedly waiting for generic data types to be supported by GHDL, do you have a rough estimate how long this will take? Will it be weeks or months?

tgingold, 2023-11-25:
Hopefully for release 4.0 (beginning of 2024).

commit 095aa64a (2023-12-17): elab-vhdl_insts: remove incorrect assertions

(Not working for instantiated units).

Fix #2544
closed 2023-12-17 by tgingold