ghdl/ghdl #2545: To_integer in port maps is not implemented for synthesis

yikkrubz, 2023-11-26:
**Description**
When the file top.vhd is synthesised, the output file top.v 
contains an assert, always @* if (line 118).
Therefore, nextpnr does not run successfully.

It seems that to_integer in port maps is not implemented for synthesis
when an unsigned is to be converted to an integer with a range, 
see line 38 in file file top.vhd.

I would appreciate it very much if this facility could be implemented
for synthesis.
**Expected behaviour**
Either ghdl should warn or a netlist without the assert should be created.

**How to reproduce?**
ghdl synth --out=verilog top.vhd -e > top.v
yosys -m ghdl -p 'ghdl top.vhd -e top; write_verilog top_yosys.v'

Info: Device utilisation:
Info:            ICESTORM_LC:     1/ 1280     0%
Info:           ICESTORM_RAM:     0/   16     0%
Info:                  SB_IO:     1/  112     0%
Info:                  SB_GB:     0/    8     0%
Info:           ICESTORM_PLL:     0/    1     0%
Info:            SB_WARMBOOT:     0/    1     0%

ERROR: cell type '$assert' is unsupported (instantiated as '\14')
9 warnings, 1 error

**Context**
Please, provide the following information:

- OS:macos 12.7.1
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `commit SHA`
GHDL 4.0.0-dev (3.0.0.r700.gc958d490e) [Dunoon edition]
 Compiled with GNAT Version: 13.1.0
 llvm 16.0.6 code generator
Written by Tristan Gingold.

Yosys 0.35+36 (git sha1 5e603c224, clang 14.0.0 -fPIC -Os)

```
[top.zip](https://github.com/ghdl/ghdl/files/13466876/top.zip)

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem

tgingold, 2023-11-26:
Can you file your top.vhd ?  I think there is a yosys pass to remove all the assertions.  Maybe this should also be an option within ghdl.

yikkrubz, 2023-11-26:
Here is the file top.vhd

[top.txt](https://github.com/ghdl/ghdl/files/13467372/top.txt)

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity A is
  generic (
    c : positive );
  port (
    clk : in std_logic;
    b : in integer range 0 to c - 1
  );
end A;

architecture rtl of A is
begin
end architecture;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity top is
  port (
    clk : in std_logic
  );
end top;

architecture str of top is
  constant d : positive := 8;
  signal e : unsigned(d - 1 downto 0);
begin
  A_INST : entity work.A(rtl)
  generic map (
    c => 2**d
  )
  port map (
    clk => clk,
    b => to_integer(e) -- does not work in synthesis
  );
end architecture;
[top.zip](https://github.com/ghdl/ghdl/files/13467374/top.zip)

closed 2023-11-26 by tgingold

tgingold, 2023-11-26:
Should now be fixed, thank you for the issue.

commit e0ce2c6a (2023-11-26): synth-vhdl_insts: handle known function conversions

Some pre-defined functions (like to_integer) can be
used as conversion functions.  They must be handled directly.

Fix #2545