ghdl/ghdl #2552: lack of port information in warning message when unable to associate signal

ams, 2023-12-05:
It would be really nice if the following warning message also contained which port, the message is good for Emacs' VHDL-mode, but for human consumption it requires a bit eye exercises to find the circumflex.

``
cannot associate out signal interface "y2" with actual port of mode in
``

E.g,

```
cannot associate out signal interface "y2" with actual port (\-destpdl(x)\) of mode in 
```

tgingold, 2023-12-07:
I think I have slightly improved the error message.  I am testing the change.

(A reproducer is always welcome, it helps to know which message is referred).

ams, 2023-12-09:
Apologies for that, would you still want a test case?  I noticed that there are several more messages that (in my opinion at least) could warrant from a similar addition of which port.

tgingold, 2023-12-10:
I have rewritten a testcase for this issue, so not needed.
If you have other messages to improve, please don't hesitate to create issues.

commit 9d75948e (2023-12-10): improve error message for incorrect associations.

Fix #2552
closed 2023-12-10 by tgingold